vanvught/rpidmx512 #2: Errors on building (RPi3 with jessie)

mansj, 2016-07-13:
I can't build this project using the included makeall-lib.sh. 

Output is as follows:

`$ ./makeall-lib.sh 
[lib-artnet]
arm-none-eabi-g++ -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-s -DRPI1 -I"./include" -I"../lib-esp8266/include" -I"../lib-utils/include" -I"../lib-hal/include" -Wall -Werror -fno-exceptions -fno-unwind-tables -fno-rtti -O3 -nostartfiles -ffreestanding -mhard-float -mfloat-abi=hard   src/artnetnode.cpp -c -o lib/artnetnode.o	
/bin/sh: 1: arm-none-eabi-g++: not found
Makefile:60: recipe for target 'lib/artnetnode.o' failed
make: *** [lib/artnetnode.o] Error 127`

rewolff, 2016-07-13:
sudo apt-get install gcc-arm-none-eabi binutils-arm-none-eabi

mansj, 2016-07-13:
Thank you, that fixed that error!

However, still getting error:

`Makefile:78: recipe for target 'lib8/artnetnode.o' failed`

Should I close this issue and start another?

mansj, 2016-07-13:
Full output:

`sh makeall-lib.sh 
[lib-artnet]
arm-none-eabi-g++ -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-s -DRPI1 -I"./include" -I"../lib-esp8266/include" -I"../lib-utils/include" -I"../lib-hal/include" -Wall -Werror -fno-exceptions -fno-unwind-tables -fno-rtti -O3 -nostartfiles -ffreestanding -mhard-float -mfloat-abi=hard   src/artnetnode.cpp -c -o lib/artnetnode.o	
arm-none-eabi-g++ -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-s -DRPI1 -I"./include" -I"../lib-esp8266/include" -I"../lib-utils/include" -I"../lib-hal/include" -Wall -Werror -fno-exceptions -fno-unwind-tables -fno-rtti -O3 -nostartfiles -ffreestanding -mhard-float -mfloat-abi=hard   src/blinktask.cpp -c -o lib/blinktask.o	
arm-none-eabi-g++ -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-s -DRPI1 -I"./include" -I"../lib-esp8266/include" -I"../lib-utils/include" -I"../lib-hal/include" -Wall -Werror -fno-exceptions -fno-unwind-tables -fno-rtti -O3 -nostartfiles -ffreestanding -mhard-float -mfloat-abi=hard   src/lightset.cpp -c -o lib/lightset.o	
arm-none-eabi-ar -r lib/libartnet.a  lib/artnetnode.o lib/blinktask.o lib/lightset.o  
arm-none-eabi-ar: creating lib/libartnet.a
arm-none-eabi-objdump -D lib/libartnet.a  | arm-none-eabi-c++filt > lib.list
arm-none-eabi-g++ -mfpu=vfpv4 -march=armv7-a -mtune=cortex-a7  -DRPI2 -I"./include" -I"../lib-esp8266/include" -I"../lib-utils/include" -I"../lib-hal/include" -Wall -Werror -fno-exceptions -fno-unwind-tables -fno-rtti -O3 -nostartfiles -ffreestanding -mhard-float -mfloat-abi=hard   src/artnetnode.cpp -c -o lib7/artnetnode.o	
arm-none-eabi-g++ -mfpu=vfpv4 -march=armv7-a -mtune=cortex-a7  -DRPI2 -I"./include" -I"../lib-esp8266/include" -I"../lib-utils/include" -I"../lib-hal/include" -Wall -Werror -fno-exceptions -fno-unwind-tables -fno-rtti -O3 -nostartfiles -ffreestanding -mhard-float -mfloat-abi=hard   src/blinktask.cpp -c -o lib7/blinktask.o	
arm-none-eabi-g++ -mfpu=vfpv4 -march=armv7-a -mtune=cortex-a7  -DRPI2 -I"./include" -I"../lib-esp8266/include" -I"../lib-utils/include" -I"../lib-hal/include" -Wall -Werror -fno-exceptions -fno-unwind-tables -fno-rtti -O3 -nostartfiles -ffreestanding -mhard-float -mfloat-abi=hard   src/lightset.cpp -c -o lib7/lightset.o	
arm-none-eabi-ar -r lib7/libartnet.a lib7/artnetnode.o lib7/blinktask.o lib7/lightset.o 
arm-none-eabi-ar: creating lib7/libartnet.a
arm-none-eabi-objdump -D lib7/libartnet.a | arm-none-eabi-c++filt > lib7.list
arm-none-eabi-g++ -mfpu=fp-armv8 -march=armv8-a -mtune=cortex-a53  -DRPI3 -I"./include" -I"../lib-esp8266/include" -I"../lib-utils/include" -I"../lib-hal/include" -Wall -Werror -fno-exceptions -fno-unwind-tables -fno-rtti -O3 -nostartfiles -ffreestanding -mhard-float -mfloat-abi=hard   src/artnetnode.cpp -c -o lib8/artnetnode.o	
arm-none-eabi-g++: error: unrecognized argument in option '-mtune=cortex-a53'
arm-none-eabi-g++: note: valid arguments to '-mtune=' are: arm1020e arm1020t arm1022e arm1026ej-s arm10e arm10tdmi arm1136j-s arm1136jf-s arm1156t2-s arm1156t2f-s arm1176jz-s arm1176jzf-s arm2 arm250 arm3 arm6 arm60 arm600 arm610 arm620 arm7 arm70 arm700 arm700i arm710 arm7100 arm710c arm710t arm720 arm720t arm740t arm7500 arm7500fe arm7d arm7di arm7dm arm7dmi arm7m arm7tdmi arm7tdmi-s arm8 arm810 arm9 arm920 arm920t arm922t arm926ej-s arm940t arm946e-s arm966e-s arm968e-s arm9e arm9tdmi cortex-a15 cortex-a5 cortex-a7 cortex-a8 cortex-a9 cortex-m0 cortex-m0plus cortex-m1 cortex-m3 cortex-m4 cortex-r4 cortex-r4f cortex-r5 cortex-r7 ep9312 fa526 fa606te fa626 fa626te fa726te fmp626 generic-armv7-a iwmmxt iwmmxt2 marvell-pj4 mpcore mpcorenovfp native strongarm strongarm110 strongarm1100 strongarm1110 xscale
Makefile:78: recipe for target 'lib8/artnetnode.o' failed
make: *** [lib8/artnetnode.o] Error 1
`

rewolff, 2016-07-13:
That "full output" is important. The "recipe failed"  is just make telling you the previous command didn't work. For fixing, we would want to know WHAT to fix.... 

It is saying that optimizing for "cortex-A53" is not recoginized. Try finding that string in your sources and change it for something like cortex-a7 instead. The differences are unlikely to matter.

vanvught, 2016-09-24:
The Raspberry Pi 3 is not supported for the baremetal DMX512 / RDM / MIDI / OSC / Art-Net solutions

closed 2016-09-24 by vanvught

rewolff, 2016-09-26:
You are saying raspberry pi 3 as a TARGET is not supported. This is about someone using it as a HOST to compile the sources into a binary....

vanvught, 2016-09-26:
The compile errors seen are related for building the libraries with target Raspberry Pi 3 (arm-none-eabi-g++: error: unrecognized argument in option '-mtune=cortex-a53'). 
As the Raspberry Pi 3 is not supported for the firmware, I have removed the option for building the libraries as target RPI3.
The firmware can now be build on a Raspberry Pi 3.

[putty.TXT](https://github.com/vanvught/rpidmx512/files/492802/putty.TXT)