vanvught/rpidmx512 #3: How to use this

Xaver2k, 2016-11-15:
Hello, nice software!
But is there any tutorial how to use these images    shields? any board or forum?
i mean do i only need to write the images to an sdcard (like Win32DiskImager) and then look for the configuration file on the sd card?
which image? (kernel.img or kernel7.img)
which pins are used to connect the leds?
where do i choose the which led protocol i need?

Thanks in advance

rewolff, 2016-11-15:
The procedure that works for me (but is slightly inefficient, but I haven't gotten around to improving it)...

download raspbian, put it on an SD card. Then copy all the files that you see from windows into a directory on your PC. Overwrite the existing kernel.img and kernel7.img there with the files from here. 

Next you can copy those few files (only about 10 or 20 or so) onto a fresh SD card and it will boot into Arjan's software....

Xaver2k, 2016-11-15:
ah ok thanks, so for network the configuration etc the files need to be created just in the root of the sdcard.

what about:
which pins are used to connect the leds Data?
where do i choose which led protocol i need?

hippyau, 2019-06-22:
Hey Arjan, 

    Newbie here :)   I have burnt the uboot image to sd, mounted and put the desired  uImage on there, but I simply get a red blinking light and no obvious signs of life.  

I have a Orange Pi One v1.1.   

Is the red light blinking a good sign?  Do I need a UART for console (where all the printf's go?) or should see something on the HDMI?   

I'd like to help put a little by writing a how-to (or is there already one i can't find?).

Even though I can't get anywhere yet, I can still see the great work that is being done on this platform!   

Cheers,
  Hippy