vanvught/rpidmx512 #4: Compile the project

folny, 2017-02-05:
Hi

I would like to ask whether it is possible to compile your projects through Windows or compilation is possible only on Linux respectively Paspberry PI.

vanvught, 2017-02-05:
Yes, you can compile the source code in a Windows environment. An example for installing the IDE / tool-chain can be found here -> http://gnuarmeclipse.github.io/toolchain/install/

vanvught, 2017-02-05:
For finding another solutions, the  Google search string could be : "windows arm toolchain"

closed 2017-02-05 by vanvught

folny, 2017-02-06:
Thanks for your response, but I have a problem with creating a kernel files please can you write deployment of files in which directories should be copied and used as Circle under Windows if possible use the software to compile Eclipse Neon thank you.