vanvught/rpidmx512 #5: Compilation Problems

folny, 2017-02-08:
Hi

For some time I try to compile your projects but still no result, please can you tell me where the problem may be ?. I use Cygwin to compile these files https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q3-update http://gnutoolchains.com/raspberry/ 

and there is an error message
$ make all
sh ./generate_sofware_version_id.sh
arm-none-eabi-as --warn --fatal-warnings  -mcpu=arm1176jzf-s -march=armv6 -mfpu=
vfp firmware/vectors.s -o build/vectors.o
arm-none-eabi-gcc -Wall -Werror -O3 -nostartfiles -ffreestanding -mcpu=arm1176jz
f-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_FRAMEBUFFER -DRDM_CO
NTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -I"../fb/include" -I
"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -I"../hal-baremetal/
include" common/lib/monitor.c -c -o build/common/monitor.o
arm-none-eabi-gcc -Wall -Werror -O3 -nostartfiles -ffreestanding -mcpu=arm1176jz
f-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_FRAMEBUFFER -DRDM_CO
NTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -I"../fb/include" -I
"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -I"../hal-baremetal/
include" common/lib/rdm_send.c -c -o build/common/rdm_send.o
arm-none-eabi-gcc -Wall -Werror -O3 -nostartfiles -ffreestanding -mcpu=arm1176jz
f-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_FRAMEBUFFER -DRDM_CO
NTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -I"../fb/include" -I
"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -I"../hal-baremetal/
include" common/lib/sscan.c -c -o build/common/sscan.o
arm-none-eabi-gcc -Wall -Werror -O3 -nostartfiles -ffreestanding -mcpu=arm1176jz
f-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_FRAMEBUFFER -DRDM_CO
NTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -I"../fb/include" -I
"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -I"../hal-baremetal/
include" common/lib/rdm_device_info.c -c -o build/common/rdm_device_info.o
arm-none-eabi-gcc -Wall -Werror -O3 -nostartfiles -ffreestanding -mcpu=arm1176jz
f-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_FRAMEBUFFER -DRDM_CO
NTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -I"../fb/include" -I
"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -I"../hal-baremetal/
include" common/lib/dmx.c -c -o build/common/dmx.o
arm-none-eabi-gcc -Wall -Werror -O3 -nostartfiles -ffreestanding -mcpu=arm1176jz
f-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_FRAMEBUFFER -DRDM_CO
NTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -I"../fb/include" -I
"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -I"../hal-baremetal/
include" lib/widget.c -c -o build/widget.o
arm-none-eabi-gcc -Wall -Werror -O3 -nostartfiles -ffreestanding -mcpu=arm1176jz
f-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_FRAMEBUFFER -DRDM_CO
NTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -I"../fb/include" -I
"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -I"../hal-baremetal/
include" lib/widget_sniffer.c -c -o build/widget_sniffer.o
arm-none-eabi-gcc -Wall -Werror -O3 -nostartfiles -ffreestanding -mcpu=arm1176jz
f-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_FRAMEBUFFER -DRDM_CO
NTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -I"../fb/include" -I
"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -I"../hal-baremetal/
include" lib/widget_params.c -c -o build/widget_params.o
arm-none-eabi-gcc -Wall -Werror -O3 -nostartfiles -ffreestanding -mcpu=arm1176jz
f-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_FRAMEBUFFER -DRDM_CO
NTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -I"../fb/include" -I
"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -I"../hal-baremetal/
include" lib/widget_monitor.c -c -o build/widget_monitor.o
arm-none-eabi-gcc -Wall -Werror -O3 -nostartfiles -ffreestanding -mcpu=arm1176jz
f-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_FRAMEBUFFER -DRDM_CO
NTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -I"../fb/include" -I
"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -I"../hal-baremetal/
include" lib/widget_usb.c -c -o build/widget_usb.o
arm-none-eabi-gcc -std=c99 -pedantic -Wall -Werror -O3 -nostartfiles -ffreestand
ing -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_F
RAMEBUFFER -DRDM_CONTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -
I"../fb/include" -I"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -
I"../hal-baremetal/include" firmware/main.c -c -o build/main.o
arm-none-eabi-gcc -std=c99 -pedantic -Wall -Werror -O3 -nostartfiles -ffreestand
ing -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -mhard-float  -DBARE_METAL -DENABLE_F
RAMEBUFFER -DRDM_CONTROLLER -I"./common/include" -I"./include" -I"../ff9b/src" -
I"../fb/include" -I"../bob-baremetal/include" -I"../bcm2835-baremetal/include" -
I"../hal-baremetal/include" firmware/lib/syscalls.c -c -o build/syscalls.o
arm-none-eabi-ld build/vectors.o  build/common/monitor.o build/common/rdm_send.o
 build/common/sscan.o build/common/rdm_device_info.o build/common/dmx.o build/wi
dget.o build/widget_sniffer.o build/widget_params.o build/widget_monitor.o build
/widget_usb.o build/main.o build/syscalls.o  -Map kernel.map -T firmware/memmap
-o build/main.elf -L /opt/gnuarm-hardfp/arm-none-eabi/lib/ -L/opt/gnuarm-hardfp/
lib/gcc/arm-none-eabi/4.9.2  -L"../fb/lib" -L"../bob-baremetal/lib" -L"../bcm283
5-baremetal/lib" -L"../hal-baremetal/lib" -L"../emmc/lib" -L"../ff9b/lib" -L/usr
/lib/arm-none-eabi/newlib/fpu -L/usr/lib/gcc/arm-none-eabi/4.8/fpu/ -lfb -lff9b
-lemmc -lhal -lbob -lbcm2835 -lc -lgcc
C:\Program Files\GNU Tools ARM Embedded\6.2 2016q4\bin\arm-none-eabi-ld.exe: can
not find -lc
C:\Program Files\GNU Tools ARM Embedded\6.2 2016q4\bin\arm-none-eabi-ld.exe: can
not find -lgcc
Makefile:83: recipe for target 'build/main.elf' failed
make: *** [build/main.elf] Error 1

rewolff, 2017-02-09:
Hi, 
Your compiler environment is not correctly set up. 

C programs may use functions from the "standard library". The library is called "c", and you can tell the linker to link with that library by passing -lc on the commandline. If your compiler cannot find the C library, that is a problem in the compiler installation. 

On some processors, some things are not implemented in hardware. Complex processors may have a division instruction or even a floating point multiplication insturction, but other processors require a software function for that. The compiler could inline the code for that each time it is needed, but calling a function is more efficient. Those functions live in a library called "gcc". If your compiler cannot find its own library called "gcc" then there is a problem with your compiler installation. 

The file that is being searched when searching for the library called "gcc", is "libgcc.a". Similarly, the C library should live in a file called "libc.a".

folny, 2017-02-09:
Thanks for responding and what you're suggesting I have therefore compiled files on my Raspberry Pi in Linux? Can you write what version of the compiler can I use ?.

rewolff, 2017-02-09:
Arjan has something setup that "works for him". I have Ubuntu Linux on my system, and something in my compiling environment is wrong too. I have not been sufficiently motivated to get things to work. Arjan distributes the "kernel.img" and "kernel7.img" compiled images in this repository. So if you just want to run the code, you do not need to compile it yourself.

folny, 2017-02-09:
I wanted to make some changes in the code and therefore I want to compile your code, so when I install Ubuntu on my Raspberry should be possible to compile the code correctly ?.

rewolff, 2017-02-09:
On Raspbian I did: 
`sudo apt-get install binutils-arm-none-eabi gcc-arm-none-eabi gdb-arm-none-eabi libstdc++-arm-none-eabi-newlib libnewlib-arm-none-eabi
`
Then I searched for cortex-a53: 
`grep -r cortex-a53
`
and changed all occurrances of a53 into a8. 
Next building with 
`sh makeall-lib.sh
`seemed to work....

folny, 2017-02-09:
Thank you for your advice i hope that this guide helps. I still want to ask in which files should I look for cortex-a53 is text or file ?.

vanvught, 2017-02-09:
There is no need to change anything with respect to cortex-a53
These targets are not built. See :
https://github.com/vanvught/rpidmx512/blob/master/firmware-template/Rules.mk#L126
https://github.com/vanvught/rpidmx512/blob/master/firmware-template/lib/Rules.mk#L45

vanvught, 2017-02-09:
> C:\Program Files\GNU Tools ARM Embedded\6.2 2016q4\bin\arm-none-eabi-ld.exe: can
not find -lc

Make  sure your GCC library path is set before starting Eclipse. You can do this in a .cmd file before starting Eclipse. There is no dependency on Newlib.

folny, 2017-02-09:
I've compiled all libraries properly but I was still unable to compile the final Kernel files what else I should do to properly compile files for the code rpi_circle_artnet_dmx ?.

folny, 2017-02-10:
Finally I managed to compile code correctly on Windows rpi_circle_artnet_dmx :-) but with others I still have a problem, for example rpi_dmx_usb_pro I still throws this error message.

PC@PC-PC /cygdrive/c/rpidmx/rpi_dmx_usb_pro
$ make
arm-none-eabi-gcc -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-
s -DRPI1 -DBARE_METAL -DRDM_CONTROLLER -DUPDATE_CONFIG_FILE  -I./include -I../li
b-dmx/include -I../lib-utils/include -I../lib-monitor/include -I../lib-hal/inclu
de -I../lib-uuid/include -I../lib-ff11/include -I../lib-emmc/include -I../lib-fb
/include -I../lib-bob/include -I../lib-utils/include -I../lib-bcm2835/include -I
../lib-ff11/src -Wall -Werror -O3 -nostartfiles -ffreestanding -mhard-float -mfl
oat-abi=hard -c common/lib/rdm_device_info.c -o build/common/lib/rdm_device_info
.o
arm-none-eabi-gcc -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-
s -DRPI1 -DBARE_METAL -DRDM_CONTROLLER -DUPDATE_CONFIG_FILE  -I./include -I../li
b-dmx/include -I../lib-utils/include -I../lib-monitor/include -I../lib-hal/inclu
de -I../lib-uuid/include -I../lib-ff11/include -I../lib-emmc/include -I../lib-fb
/include -I../lib-bob/include -I../lib-utils/include -I../lib-bcm2835/include -I
../lib-ff11/src -Wall -Werror -O3 -nostartfiles -ffreestanding -mhard-float -mfl
oat-abi=hard -c firmware/main.c -o build/firmware/main.o
arm-none-eabi-gcc -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-
s -DRPI1 -DBARE_METAL -DRDM_CONTROLLER -DUPDATE_CONFIG_FILE  -I./include -I../li
b-dmx/include -I../lib-utils/include -I../lib-monitor/include -I../lib-hal/inclu
de -I../lib-uuid/include -I../lib-ff11/include -I../lib-emmc/include -I../lib-fb
/include -I../lib-bob/include -I../lib-utils/include -I../lib-bcm2835/include -I
../lib-ff11/src -Wall -Werror -O3 -nostartfiles -ffreestanding -mhard-float -mfl
oat-abi=hard -c lib/widget.c -o build/lib/widget.o
arm-none-eabi-gcc -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-
s -DRPI1 -DBARE_METAL -DRDM_CONTROLLER -DUPDATE_CONFIG_FILE  -I./include -I../li
b-dmx/include -I../lib-utils/include -I../lib-monitor/include -I../lib-hal/inclu
de -I../lib-uuid/include -I../lib-ff11/include -I../lib-emmc/include -I../lib-fb
/include -I../lib-bob/include -I../lib-utils/include -I../lib-bcm2835/include -I
../lib-ff11/src -Wall -Werror -O3 -nostartfiles -ffreestanding -mhard-float -mfl
oat-abi=hard -c lib/widget_sniffer.c -o build/lib/widget_sniffer.o
arm-none-eabi-gcc -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-
s -DRPI1 -DBARE_METAL -DRDM_CONTROLLER -DUPDATE_CONFIG_FILE  -I./include -I../li
b-dmx/include -I../lib-utils/include -I../lib-monitor/include -I../lib-hal/inclu
de -I../lib-uuid/include -I../lib-ff11/include -I../lib-emmc/include -I../lib-fb
/include -I../lib-bob/include -I../lib-utils/include -I../lib-bcm2835/include -I
../lib-ff11/src -Wall -Werror -O3 -nostartfiles -ffreestanding -mhard-float -mfl
oat-abi=hard -c lib/widget_params.c -o build/lib/widget_params.o
arm-none-eabi-gcc -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-
s -DRPI1 -DBARE_METAL -DRDM_CONTROLLER -DUPDATE_CONFIG_FILE  -I./include -I../li
b-dmx/include -I../lib-utils/include -I../lib-monitor/include -I../lib-hal/inclu
de -I../lib-uuid/include -I../lib-ff11/include -I../lib-emmc/include -I../lib-fb
/include -I../lib-bob/include -I../lib-utils/include -I../lib-bcm2835/include -I
../lib-ff11/src -Wall -Werror -O3 -nostartfiles -ffreestanding -mhard-float -mfl
oat-abi=hard -c lib/widget_monitor.c -o build/lib/widget_monitor.o
arm-none-eabi-gcc -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-
s -DRPI1 -DBARE_METAL -DRDM_CONTROLLER -DUPDATE_CONFIG_FILE  -I./include -I../li
b-dmx/include -I../lib-utils/include -I../lib-monitor/include -I../lib-hal/inclu
de -I../lib-uuid/include -I../lib-ff11/include -I../lib-emmc/include -I../lib-fb
/include -I../lib-bob/include -I../lib-utils/include -I../lib-bcm2835/include -I
../lib-ff11/src -Wall -Werror -O3 -nostartfiles -ffreestanding -mhard-float -mfl
oat-abi=hard -c lib/widget_usb.c -o build/lib/widget_usb.o
arm-none-eabi-ld build/vectors.o  build/common/lib/rdm_device_info.o build/firmw
are/main.o build/lib/widget.o build/lib/widget_sniffer.o build/lib/widget_params
.o build/lib/widget_monitor.o build/lib/widget_usb.o    -Map kernel.map -T firmw
are/memmap -o build/main.elf -L../lib-dmx/lib -L../lib-utils/lib -L../lib-monito
r/lib -L../lib-hal/lib -L../lib-uuid/lib -L../lib-ff11/lib -L../lib-emmc/lib -L.
./lib-fb/lib -L../lib-bob/lib -L../lib-utils/lib -L../lib-bcm2835/lib -L/usr/lib
/gcc/arm-none-eabi/4.9.3/fpu -L/opt/gnuarm-hardfp/lib/gcc/arm-none-eabi/4.9.3/ar
mv6zk/arm1176jzf-s/hardfp/vfp   -ldmx -lutils -lmonitor -lhal -luuid -lff11 -lem
mc -lfb -lbob -lutils -lbcm2835 -lgcc
C:\Program Files\GNU Tools ARM Embedded\4.9 2015q3\bin\arm-none-eabi-ld.exe: can
not find -lgcc
../firmware-template/Rules.mk:149: recipe for target 'build/main.elf' failed
make: *** [build/main.elf] Error 1

vanvught, 2017-02-10:
> C:\Program Files\GNU Tools ARM Embedded\4.9 2015q3\bin\arm-none-eabi-ld.exe: can
not find -lgcc
Your development environment is missing the gcc library.

folny, 2017-02-11:
Yes, I already figured it out that the problem is.

closed 2017-05-13 by vanvught