vanvught/rpidmx512 #11: unable to use this outside of examples folder

prismspecs, 2018-10-02:
I wonder if you can help me get this working in an arbitrary cpp project--when I try to compile I get pca9685servo.h: No such file or directory. Do you happen to know how to link this?

...Perhaps with PROJECT_LDFLAGS?

prismspecs, 2018-10-03:
I'm working from the command line on Raspbian.

I can get it to compile just fine on its own, from the example folder using the steps you provided, but I would like to use this library in another c++ project (more specifically, an OpenFrameworks project) and can't seem to get it to link correctly.

I've tried for example -lpca9685 (much like I already do for -lwiringPi) but it throws "pca9685servo.h: No such file or directory"

I could set it as an external source path, but I don't really know which folder to point it to (though I think I've tried them all).

Is it possible to compile this as a library that functions like wiringPi so I can just simply link it?

Thank you for your time

vanvught, 2018-10-03:
pca968servo.h: No such file or directory" has to do with with the include path settings for your compiler. Make sure that you have provided the path to pca968servo.h in the -I option.

The header file is in here https://github.com/vanvught/rpidmx512/tree/master/lib-pca9685/include

Please note that I am not using wiringPi at all, but the best GPIO library -> http://www.airspayce.com/mikem/bcm2835/
So you should link with -lpca9685  -lbcm2835  

But first you would need to solve your include path.

prismspecs, 2018-10-05:
Does this go into the config.make file?

vanvught, 2018-10-05:
> Does this go into the config.make file?

I cannot tell. I am not familiar with your development environment.

"pca968servo.h: No such file or directory" is typically a missing include directive setting within the compile step.

prismspecs, 2018-10-15:
Hmm I've tried every which way to no avail. I've determined that it will ultimately be much easier to use Arduino for wiring and send messages via serial. I wonder if there's a way to package your library for use by people unfamiliar with the intricacies of C programming and development environments? If you're ever looking for some free +rep you might consider turning this into an ofx library

vanvught, 2018-10-15:
All lib-* projects are cross-platform C/C++ implementations. I've tested the compilation and build on: MacOS, Linux Ubuntu. Linux Debian, Linux Raspbian and even Cygwin. 

The "pca968servo.h: No such file or directory" is something that needs to be fixed in your development tool.

closed 2018-10-15 by vanvught

prismspecs, 2018-10-15:
I mean so that it can be used as easily as something like wiringPi which is sort of plug and play

vanvught, 2018-10-15:
I do not know wiringPi. But all libraries here can be linked with -l
And the include path must be known in the development environment.

rewolff, 2018-10-15:
Wiringpi is "included in the distribution". This means that it can be installed in places where the compiler looks by default. 

Because you do not provide any details (like compiler commandline) but that you're getting a not-found on the header file, we cannot give you more specific help than to tell you to instruct the compiler where to find the header file.

prismspecs, 2018-10-15:
I understand that I'm not providing enough information, I'm just a bit too lost to even begin to do so--

I'm working with openFrameworks, which is a set of libs for C++ and it's fairly idiosyncratic in terms of how it generates config.make files, and it feels a bit at odds (to someone without a lot of knowledge on this topic) with the suggested implementation of the linking flag.

Obviously you're under no obligation to spend time helping me with something not specifically related to your project, but in case this makes things clear with haste, here is the config.make file  for oF projects and how I am currently linking wiringPi (though it did not come on my distro and was installed by me)

```
################################################################################
# PROJECT LINKER FLAGS
#	These flags will be sent to the linker when compiling the executable.
#
#		(default) PROJECT_LDFLAGS = -Wl,-rpath=./libs
#
#   Note: Leave a leading space when adding list items with the += operator
#
# Currently, shared libraries that are needed are copied to the
# $(PROJECT_ROOT)/bin/libs directory.  The following LDFLAGS tell the linker to
# add a runtime path to search for those shared libraries, since they aren't
# incorporated directly into the final executable application binary.
################################################################################
# PROJECT_LDFLAGS=-Wl,-rpath=./libs
PROJECT_LDFLAGS=-Wl,-rpath=./libs
PROJECT_LDFLAGS += -lwiringPi
```

Does this seem like the correct place to also attempt to link your pca9685 lib? Mine is located at
~/workbench/rpidmx512/lib-pca9685/include

vanvught, 2018-10-15:
This is the linker section configuration. It must also include -lbcm2835

I am using http://www.airspayce.com/mikem/bcm2835/ 
With reference to the remark from @rewolff , this library can be installed in the distribution. Hence the include and library path is known for the system compiler.

Do you also have a compile configuration section ?

rewolff, 2018-10-15:
Just for your understanding: There is compilation and linking. Compilation makes machine code from your (hopefully) readable sourcecode, linking then combines those machinecodes from different files into one executable. 

Your "...h not found" happens in the compilation phase. The section you quoted is for the linking phase. 

You need to add -I<dir> where for <dir> you fill in the directory where that header lives. 
 -I~/workbench/rpidmx512/lib-pca9685/include
however the "~" probably won't work there, so  you should end up with: 
 -I/home/you/workbench/rpidmx512/lib-pca9685/include