vanvught/rpidmx512 #15: Trouble building l6470 sparkfun examples

TaylorHokanson, 2018-12-28:
Hello,

First of all, thanks for releasing this project! I'm working with a Raspberry Pi 3, and am only interested in the Sparkfun l6470 example code. I successfully installed the BCM 2835 prerequisite. I then downloaded the repo, cd to the sparkfun folder, and attempt to make per the README. I get the following error:

```
pi@raspberrypi:/development/workspace/lib-l6470/examples/sparkfun $ make
g++ SparkFunGetSetParamTest.cpp -I./../../../lib-l6470/include -Wall -Werror -O3 -fno-rtti -std=c++11 -DNDEBUG -o SparkFunGetSetParamTest -L./../../../lib-l6470/lib_linux -ll6470  -lbcm2835
SparkFunGetSetParamTest.cpp: In function ‘void test(float, float)’:
SparkFunGetSetParamTest.cpp:237:17: error: call of overloaded ‘abs(float)’ is ambiguous
  if (abs(v1 - v2) > 0.1) {
                 ^
In file included from /usr/include/c++/6/cstdlib:75:0,
                 from /usr/include/c++/6/stdlib.h:36,
                 from SparkFunGetSetParamTest.cpp:6:
/usr/include/stdlib.h:735:12: note: candidate: int abs(int)
 extern int abs (int __x) __THROW __attribute__ ((__const__)) __wur;
            ^~~
In file included from /usr/include/c++/6/stdlib.h:36:0,
                 from SparkFunGetSetParamTest.cpp:6:
/usr/include/c++/6/cstdlib:180:3: note: candidate: long long int std::abs(long long int)
   abs(long long __x) { return __builtin_llabs (__x); }
   ^~~
/usr/include/c++/6/cstdlib:172:3: note: candidate: long int std::abs(long int)
   abs(long __i) { return __builtin_labs(__i); }
   ^~~
Makefile:41: recipe for target 'SparkFunGetSetParamTest' failed
make: *** [SparkFunGetSetParamTest] Error 1  
```
Any suggestions you can offer are appreciated.

vanvught, 2018-12-28:
I am able to reproduce the error. I need to investigate what happened with this compiler version.

> pi@raspberrypi-3:/development/workspace/lib-l6470/examples/sparkfun $ g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 6.3.0-18+rpi1+deb9u1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1) 

> pi@raspberrypi-3:/development/workspace/lib-l6470/examples/sparkfun$ make      
cd ./../../../lib-l6470 && make -f Makefile.Linux "DEF=-DRASPPI"
make[1]: Entering directory '/development/workspace/lib-l6470'
mkdir -p build_linux/linux 
mkdir -p lib_linux
g++ -DRASPPI -DNDEBUG -I./include -I../lib-debug/include  -I../lib-bcm2835_raspbian/include -Wall -Werror -O2 -fno-rtti -std=c++11 src/slushboard.cpp -c -o build_linux/slushboard.o
g++ -DRASPPI -DNDEBUG -I./include -I../lib-debug/include  -I../lib-bcm2835_raspbian/include -Wall -Werror -O2 -fno-rtti -std=c++11 src/l6470support.cpp -c -o build_linux/l6470support.o
g++ -DRASPPI -DNDEBUG -I./include -I../lib-debug/include  -I../lib-bcm2835_raspbian/include -Wall -Werror -O2 -fno-rtti -std=c++11 src/autodriver.cpp -c -o build_linux/autodriver.o
g++ -DRASPPI -DNDEBUG -I./include -I../lib-debug/include  -I../lib-bcm2835_raspbian/include -Wall -Werror -O2 -fno-rtti -std=c++11 src/l6470config.cpp -c -o build_linux/l6470config.o
g++ -DRASPPI -DNDEBUG -I./include -I../lib-debug/include  -I../lib-bcm2835_raspbian/include -Wall -Werror -O2 -fno-rtti -std=c++11 src/slushtemp.cpp -c -o build_linux/slushtemp.o
g++ -DRASPPI -DNDEBUG -I./include -I../lib-debug/include  -I../lib-bcm2835_raspbian/include -Wall -Werror -O2 -fno-rtti -std=c++11 src/l6470dump.cpp -c -o build_linux/l6470dump.o
g++ -DRASPPI -DNDEBUG -I./include -I../lib-debug/include  -I../lib-bcm2835_raspbian/include -Wall -Werror -O2 -fno-rtti -std=c++11 src/l6470commands.cpp -c -o build_linux/l6470commands.o
g++ -DRASPPI -DNDEBUG -I./include -I../lib-debug/include  -I../lib-bcm2835_raspbian/include -Wall -Werror -O2 -fno-rtti -std=c++11 src/l6470.cpp -c -o build_linux/l6470.o
g++ -DRASPPI -DNDEBUG -I./include -I../lib-debug/include  -I../lib-bcm2835_raspbian/include -Wall -Werror -O2 -fno-rtti -std=c++11 src/slushmotor.cpp -c -o build_linux/slushmotor.o
ar -r lib_linux/libl6470.a   build_linux/slushboard.o build_linux/l6470support.o build_linux/autodriver.o build_linux/l6470config.o build_linux/slushtemp.o build_linux/l6470dump.o build_linux/l6470commands.o build_linux/l6470.o build_linux/slushmotor.o  
ar: creating lib_linux/libl6470.a
objdump -D lib_linux/libl6470.a  | c++filt > lib_linux/lib.list
make[1]: Leaving directory '/development/workspace/lib-l6470'
g++ SparkFunGetSetParamTest.cpp -I./../../../lib-l6470/include -I./../../../lib-bcm2835_raspbian/include -Wall -Werror -O3 -fno-rtti -std=c++11 -DNDEBUG -o SparkFunGetSetParamTest -L./../../../lib-l6470/lib_linux -L./../../../lib-bcm2835_raspbian/lib_linux -ll6470  -lbcm2835_raspbian
SparkFunGetSetParamTest.cpp: In function 'void test(float, float)':
SparkFunGetSetParamTest.cpp:237:17: error: call of overloaded 'abs(float)' is ambiguous
  if (abs(v1 - v2) > 0.1) {
                 ^
In file included from /usr/include/c++/6/cstdlib:75:0,
                 from /usr/include/c++/6/stdlib.h:36,
                 from SparkFunGetSetParamTest.cpp:6:
/usr/include/stdlib.h:735:12: note: candidate: int abs(int)
 extern int abs (int __x) __THROW __attribute__ ((__const__)) __wur;
            ^~~
In file included from /usr/include/c++/6/stdlib.h:36:0,
                 from SparkFunGetSetParamTest.cpp:6:
/usr/include/c++/6/cstdlib:180:3: note: candidate: long long int std::abs(long long int)
   abs(long long __x) { return __builtin_llabs (__x); }
   ^~~
/usr/include/c++/6/cstdlib:172:3: note: candidate: long int std::abs(long int)
   abs(long __i) { return __builtin_labs(__i); }
   ^~~
Makefile:41: recipe for target 'SparkFunGetSetParamTest' failed`

vanvught, 2018-12-28:
@TaylorHokanson There is commit 0e010e5 with the fix.

closed 2018-12-28 by vanvught