vanvught/rpidmx512 #16: pca9685 lib not reading pwm channel value correctly?

MrGrimod, 2019-01-01:
I'm trying to get the read function of [this](https://github.com/vanvught/rpidmx512/tree/master/lib-pca9685) c++ lib for the pca9685 pwm driver on my raspberry to work. I did everything as shown in the examples, it has to be something obvious but I just can't find it. The wiring is pretty simple a RC receiver is connected via GND, 5v, PWM (signal) to channel 7 of the pca9685.
But the output of the program(node) is just `Channel PWM: 0000h-1000h` and is not changing at all. If I write the PWM value before (via.`pca9685.Write(CHANNEL(5), VALUE(PCA9685_VALUE_MAX/2));`) and unplug the wire the output changes to: `Channel PWM: 0000h-0800h`.

      int fd;

 

      ros::init(argc, argv, "talker_bat");
    
      ros::NodeHandle n;
    
      battery_node::bat_stat bat_data;
    
      ros::Publisher chatter_pub = n.advertise<battery_node::bat_stat>("bat_stat", 100);
    
      ros::Rate loop_rate(10);
    
      PCA9685 pca9685;
      pca9685.SetFrequency(50);
      
      uint16_t OnValue, OffValue;
    
      int val = 0;
      while (ros::ok())
      {
      	pca9685.Read(CHANNEL(5), &OnValue, &OffValue);
        printf("\tChannel PWM: %04xh-%04xh \n", OnValue, OffValue);
        bat_data.bat_val = 10;
        chatter_pub.publish(bat_data);
      }
      return 0;

vanvught, 2019-01-01:
Are the examples working fine for you?
What do you exactly mean with "unplug the wire"?. Wires should be in place all the time.
You write "PWM (signal) to channel 7" and then "pca9685.Write(CHANNEL(5), VALUE(PCA9685_VALUE_MAX/2));". That are different channels.

MrGrimod, 2019-01-01:
1. Yeah, the example is working fine, altough the ouput is kinda weird (0-0800h), I mean when the dutycycle is 50% shouldn't it be 0-0500h then?

2. I'removing the wire from channel 5 of the pca9685, to test the example. Because the pwm signal has already been set before, at least that's how I interpret the example.

3. Sry I corrected it, I ment channel 5.

A dumb question, is it even possible, to read incoming pwm signals, when they are not set by the board? I mean are the values in regs which the read function is reading always the same as the input signal, too if they have not been set? Because what I'm trying to do is to read an **incoming** pwm value that has not been created by the board.

vanvught, 2019-01-01:
1. See the datasheet for the explanation. 

The PCA9685 is PWM output only.

closed 2019-01-01 by vanvught

MrGrimod, 2019-01-01:
Oh, ups ok sry for wasting your time did not want to be outrageous, sry.
Thanks for your work and fast replies!

vanvught, 2019-01-01:
No problem

rewolff, 2019-01-01:
> 1. Yeah, the example is working fine, altough the ouput is kinda weird (0-0800h), I mean when the dutycycle is 50% shouldn't it be 0-0500h then?
Half of 0x1000 is 0x0800. (or "half of 1000h is 0800h" as you seem to be writing it).