vanvught/rpidmx512 #18: Binary packages for Linux and DMX Input

EmerickH, 2019-02-27:
Hello,
(I'm using DMX USB Pro firmware on a RPI with BitWizard DMX shield)
This project is very great but I think it's a shame to have to use a complete Raspberry Pi with a powerful CPU, a GPU with a HDMI, an Ethernet, USB ports,... that we can't use at the same time of the firmware!
Do you think it would be possible to transform at least the DMX library of this repo into a library that can be embedded in a C++ Raspbian software for example? DMX UART output libraries already exists but I've never found any for DMX input! Or if it's not possible at least to add the possibility to Artnet firmware to use also DMX input to Artnet out. Actually the only solution to make a DMX to Artnet node is to have 2 Raspberries, 1 for DMX>USB and another with QLC+/OLA for USB>Artnet...

Thank you,
Emerick

vanvught, 2019-02-27:
Hi Emerick,

Thanks for the compliment.

I am just using the Raspberry Pi board as a simple ARM board. Please also be aware that I am moving away from the Raspberry Pi boards. The boards I am going to support are the Orange Pi Zero and the Orange Pi One. Both boards are cheaper and have a far better Ethernet implementation. Moreover, the Orange Pi Zero has SPI flash. Hence it can boot from SPI flash.

It is impossible to combine Linux Raspbian with bare metal firmware on the same board. DMX input and RDM is also impossible on Linux. 

The best option is to run the Art-Net node on an Orange Pi Zero.

closed 2019-02-27 by vanvught

EmerickH, 2019-02-27:
Ok thank you for quick answer, I suspected a little this...
But it seems impossible to convert DMX to Artnet with the current Art-Net 3 Node firmware, am I wrong?
If it's possible it should be more documented in the Wiki.
I already considered buying a Orange Pi but it is very hard to find them in France without 20€ of shipping...

Emerick

vanvught, 2019-02-27:
Correct, the current Art-Net 3 firmware is Art-Net to DMX. As soon as I have migrated all Raspberry Pi firmware to Orange Pi (and upgraded to Art-Net 4), then I will work on new solutions. DMX to Art-Net 4 is on my radar.

EmerickH, 2019-02-27:
Ok, I'll be waiting!
I finally found an OPi Zero for 12€95 on eBay, if you release this feature I'll probably buy one!
I'm going to keep my 2 RPI setup at this time.

Thank you again!