vanvught/rpidmx512 #22: Enhancements: Remote configuration framework

vanvught, 2019-05-09:
The current framework is based on human readable UDP messages. 
In order to develop cross-platform QT applications, it would great to have binary data (well defined structs) exchanged.

New commands:

?uptime#bin

Returning uint64 (8 bytes)

?list#bin

Returning:

```
struct TRemoteConfigListBin {
	uint8_t aMacAddress[6];
	uint8_t nType;				// TRemoteConfig
	uint8_t nMode;				// TRemoteConfigMode
	uint8_t nActiveUniverses;
	uint8_t aDisplayName[REMOTE_CONFIG_DISPLAY_NAME_LENGTH];
}__attribute__((packed));
```

?store# and !store#

Where the parameters can be 
https://github.com/vanvught/rpidmx512/blob/master/lib-remoteconfig/src/remoteconfig.cpp#L97

Proposed by @rbarreiros

rewolff, 2019-05-11:
IMHO, something that is capable of IP should also be capable of parsing the strings. 

So.. IMHO using "modern" human readable UDP packets is preferable over the binary method. 

Stuff like compatiblity issues become easier with such a protocol. If necessary you can implement "what parameters do you support?" Clients can talk to all versions of all servers because they send whatever they know about and get "don't know how to do that" back when a parameter is specified that is only available on other versions. Much better than having to negotiate: I know protocol version 2, 3 and 4, well I only know 1 and two, so lets use two. (that's what you end up doing after getting an incompatible change in the protocol a few times).

vanvught, 2019-05-11:
The design is a simple lightweight UDP fire-and-forget protocol. There will be no negotiation. It is for the client to be up-to-date with the node implementation. 

http://www.orangepi-dmx.org/remote-configuration

rbarreiros, 2019-05-11:
rewolff you're complicating, the 'human readable' UDP packets still exists, the 'binary' comms are an extension and make the parsing none existant, these are extensions to the current protocol, which is not being replaced.

Between something like:

struct reply = parse_text_string(datagram);

function parse_text_string(...)
{
  // huge complicated state text parsing stuff
}

and something like:

struct *reply = reinterpret_cast<reply*>(datagram.data);

I'd say the binary method is preferable for an applicationm, I guess I don't need to enumerate all the advantages.

rbarreiros, 2019-05-11:
There's an issue with !display#, at application start, there is no way to figure out the current state of display, the node might have display on, but we have no idea of knowing. 

Now there's 2 possible scenarios:

a) the application will eventually sync to the state of the node, the first time we press the button to enable the display, the display on cmd is sent, and, the state of display for that node in the app will be the same as the node itself, drawbacks, we don't know remotely the state of the display untill we turnt it on/off in the app at least once for each node. This could be done automatically, but it'll mean everytime we start the app, all the nodes will display on/of once to sync with the app.

b) we reply the state of the display by calling !display with not args, or ?display# ? this would be the simplest, and would show the app all node current state without interfering with their state.

vanvught, 2019-05-11:
The design is fire-and-forget. There is no negation in this protocol. It is expected that the user has a visible view on the display. Hence, we just send "ON", "OFF" messages to the node.

vanvught, 2019-05-11:
@rbarreiros 

```
pi@nuc-i5:~$ echo '?display#' | nc -u -p 10501 -w 1 192.168.2.137 10501
display:On
pi@nuc-i5:~$ echo '?display#bin' | nc -u -p 10501 -w 1 192.168.2.137 10501 | xxd -l 4
00000000: 01                                       .
pi@nuc-i5:~$ echo '!display#0' | nc -u -p 10501 -w 1 192.168.2.137 10501
pi@nuc-i5:~$ echo '?display#' | nc -u -p 10501 -w 1 192.168.2.137 10501
display:Off
pi@nuc-i5:~$ echo '?display#bin' | nc -u -p 10501 -w 1 192.168.2.137 10501 | xxd -l 4
00000000: 00                                       .
pi@nuc-i5:~$ 
```

rbarreiros, 2019-05-11:
Perfect!

commit ec0ccc98 (2019-06-21): Fixed #22
closed 2019-06-21 by vanvught