vanvught/rpidmx512 #24: Orange pi version support.

mrv96, 2019-05-20:
Hi, thank you for your amazing work. I have 2 questions about hardware:

1. The number of Artnet universes is limited by the hardware or by the pinout of the Orange Pi Zero?

2. If I chose to work with the Zero version is the total amount of RAM a constraint? I mean, should I chose the 256MB or the 512MB version?

vanvught, 2019-05-20:
Thanks for the compliments.

1. There is no software limitation for Art-Net 4 universes. There are software limitations in the case of:

- DMX/RDM; There are 2 UARTs on the OPi Zero and 4 UARTs on the OPi One.
- Pixel; There is one SPI output on the OPi Zero and therefore there is the standard limitation of 4 Universes.

However, there is 4x 4 Universes pixel prototype board -> http://www.orangepi-dmx.org/orange-pi-dmx512-rdm/ws28xx-configuration/prototype-boards/multi/4-output 
The firmware will be released soon.

2. The 256MB is good enough. The firmware is just using a couple of 1MB pages.

mrv96, 2019-05-20:
Thank you for your fast reply. If I well understand, since USART peripherals are in the core, I should be able to handle 4 DMX/RDM universes using the Orange Pi One firmware on Orange Pi Zero Plus or Plus2. Do you agree?

I was not looking WS2812 NeoPixel, but I don't understand your answer... since SPI is a BUS, why have you limitations? The only limitation of SPI is the number of CS pins available.

vanvught, 2019-05-20:
> If I well understand, since USART peripherals are in the core, I should be able to handle 4 DMX/RDM universes using the Orange Pi One firmware on Orange Pi Zero Plus or Plus2. Do you agree?

The only supported boards are the Orange Pi Zero and Orange Pi One. Both have similar Allwinner SoC's (H2+/H3). The Orange Pi Zero Plus has a Allwinner H5 SoC; the Allwinner H5 SoC is not supported.

> I was not looking about WS2812 NeoPixel, but I didn't understand your answer... since SPI is a BUS, why have you limitations? 
The limitation with Allwinner SoC is just having a single SPI master. The CS is not used in an addressable led (pixels or NeoPixel {Adafruit}) environment.

There is a 4x DMX/RDM OPi One board here -> http://www.bitwizard.nl/shop/4x-dmx-interface-for-orange-pi?search=orange%20
Note: The picture is showing old firmware. The current firmware is a full Art-Net 4 implementation.
[image]

mrv96, 2019-05-20:
Asap I'll see also the WS2812 application. Thank you for the link.

According official OPi web site (http://www.orangepi.org/OrangePiZeroPlus2) I think that there is a version of Zero Plus2 with H3. I don't have any OPi at home, so if you are interested and if you think that Zero Plus2 H3 could work, I could buy it and try...

vanvught, 2019-05-20:
I do not support the Zero Plus2 H3. Moreover, that board is 23$ while the Orange Pi Zero 256MB is just 9$.

mrv96, 2019-05-20:
Ok. It was only a matter of space due to the smaller size of Zero. I agree completely with you about the price; I was only curious 🤣

closed 2019-05-20 by vanvught