vanvught/rpidmx512 #35: Will you add an HTTP config interface?

mrv96, 2019-06-14:
Hi, usually ArtNet-E1.31 devices have an HTML page where is possible to configure the device easily. Are you thinking to develop something like that?

Thank you, Mario.

vanvught, 2019-06-14:
Hi Mario,

There is a remote configuration framework -> http://www.orangepi-dmx.org/orange-pi-dmx512-rdm/remote-configuration
The major advantage for this framework is that there is no need to login into each device individually. 
An example Java UI is provided. See also #22

With the above and the fact that there no TCP/IP stack available, a HTTP server will never be implemented.

mrv96, 2019-06-14:
I saw your java UI example just before to open the issue. I made the question because I think that an HTTP config page would be preferable for common users. I'm an electronic engineer, so I understand and love your approach, but if I think to a final possible generic user that buy your product, I'm quite sure that he would be happier with a simpler and user friendly HTTP interface. Usually users don't love to see hard coded text to change in order to configure a device.

Do you understand my point of view?

vanvught, 2019-06-14:
Please note that I am delivering open source (and open hardware). I am not selling any products. 

Personally I believe that having a central remote configuration is far better than having a HTTP server on each node for logging into. Moreover, there is no TCP/IP stack available.

mrv96, 2019-06-14:
Ok. Thank you very much for your answer. I understand you very well. It was only a suggestion in order to have a possible improvement of the project visibility and diffusion, because it is very great!

Thank you again for your work :)

vanvught, 2019-06-14:
Thanks for the compliments Mario.

closed 2019-06-14 by vanvught