vanvught/rpidmx512 #38: Remote configuration framework: Add ?version# and ?version#bin

vanvught, 2019-06-22:
This would make it possible to show the installed firmware details (date and time).

vanvught, 2019-06-22:
@rbarreiros For the bin, are you happy with just dumping the U-Boot header?

rbarreiros, 2019-06-22:
I'd guess so, yes.

vanvught, 2019-06-23:
@rbarreiros I've decided to create a much more simpler struct : 

```
#define SOFTWARE_VERSION_LENGTH	3
#define GCC_DATE_LENGTH			11
#define GCC_TIME_LENGTH			8

struct TFirmwareVersion {
	char SoftwareVersion[SOFTWARE_VERSION_LENGTH];
	char BuildDate[GCC_DATE_LENGTH];
	char BuildTime[GCC_TIME_LENGTH];
};
```

```
MacBook-Air:~ arjanvanvught$ echo '?version#' | nc -u -p 10501 -w 1 192.168.2.251 10501
version:[V2.2] Orange Pi Zero Compiled on Jun 23 2019 at 19:22:20

MacBook-Air:~ arjanvanvught$ echo '?version#bin' | nc -u -p 10501 -w 1 192.168.2.251 10501 | xxd -l 22
00000000: 322e 324a 756e 2032 3320 3230 3139 3139  2.2Jun 23 201919
00000010: 3a32 323a 3230                           :22:20
```

```
...
MX25L1605D, sector size 4096, 2097152 bytes
Detected MX25L1605D with sector size 4096 total 2097152 bytes
[V2.2] Orange Pi Zero Compiled on Jun 23 2019 at 19:22:20
Ethernet sACN E1.31 DMX Output / Pixel controller {4 Universes}
Network init
Network configuration
 Hostname   : allwinner_270351
 Interface  : 192.168.2.251
 Netmask    : 255.255.255.0
 MacAddress : 02:42:1c:27:03:51
 DHCP       : No
...
```

commit 161fa90f (2019-06-23): Fixed #38
closed 2019-06-23 by vanvught