vanvught/rpidmx512 #39: sACN reporting as ArtNet

hippyau, 2019-06-23:
Using the the latest uBoot and sACN uImage,  it's working great as far as DMX, but if I request...

`echo '?list#' | nc -u -p 10501 -w 1 10.0.0.3 10501`

I get back...
`10.0.0.3,Art-Net,RDM,4`

but should return:
`10.0.0.3,sACN E1.31,RDM,4` ?

vanvught, 2019-06-23:
```
MacBook-Air:~ arjanvanvught$ echo '?list#' | nc -u -p 10501 -w 1 192.168.2.251 10501
192.168.2.251,sACN E1.31,DMX,4
```

I am not able to re-create the issue reported.

hippyau, 2019-06-23:
Sorry, that's weird.... I am sending sACN from MagicQ and it's working
perfectly(!)... but reporting "10.0.0.3,Art-Net,RDM,4" from
"opi_emac_e131_dmx_multi.zip" downloaded today.

vanvught, 2019-06-23:
Are you using the BitWizard.nl 4x DMX/RDM board?
If not, then which SPI flash do you have attached to the One?

hippyau, 2019-06-23:
Hey, oh, i'm using the microSD card on a oPi One V1.1

... and a max485.  DMX works great :)

vanvught, 2019-06-23:
The remote configuration framework works on the the data in the SPI flash.
When there is no SPI flash, then the data is unknown. I will open an enhancement request for disabling the remote configuration framework when there is no SPI flash (attached to the One).

closed 2019-06-23 by vanvught

hippyau, 2019-06-23:
Ahh... sorry.....   thanks!!