vanvught/rpidmx512 #40: Orange Pi One: Multi DMX/RDM: Disable remote configuration framework when there is no SPI flash attached

vanvught, 2019-06-23:
Reference #39

hippyau, 2019-06-23:
I don't suppose it's possible to "Enable SD card configuration when no SPI flash attached"?
Maybe a baremetal limitation, but could pick a block on the SD read/write that?

vanvught, 2019-06-23:
> I don't suppose it's possible to "Enable SD card configuration when no SPI flash attached"?

It is doable, but will not be implemented per design -> http://www.orangepi-dmx.org/orange-pi-dmx512-rdm/remote-configuration/spi-flash-boot-process

> Maybe a baremetal limitation, but could pick a block on the SD read/write that?

There is no baremetal limitation. Per design, the SDCard is used for the initial installation/configuration only.

For less then 1$ you can attach a SPI flash to the One. For example W25Q32 -> https://nl.aliexpress.com/item/32918070894.html

rewolff, 2019-06-23:
> for less than $1 you can attach a SPI flash to the One.
For a bit more you can buy the 4x DMX board from me with the W25Q32 soldered on. :-) 
Besides that you get 4x isolation and 3 more channels. :-)

hippyau, 2019-06-23:
Thanks Gents,
  That's what i'll be doing.    Thanks for providing the understanding.
  SD not required.
Awesome!
Cheers,

commit 92b450a8 (2019-06-27): Fixed #40
closed 2019-06-27 by vanvught