vanvught/rpidmx512 #45: U-Boot: Verifying Checksum ... Bad Data CRC

hippyau, 2019-07-17:
Hi there, any ideas as to what is happening here?   

Using a fresh "PI Zero - 256" I have compiled and uploaded U-Boot following the [Instructions for setting up SPI flash on Zero](http://www.orangepi-dmx.org/allwinner-h3/spiflash-write)

I then uploaded orangepi_zero.uImage - from freshly downloaded opi_emac_artnet_dmx.zip 
```
$ sudo ../sunxi-tools/sunxi-fel -p spiflash-write 0x180000 ./orangepi_zero.uImage
100% [================================================]   134 kB,   64.5 kB/s
```
And on the UART console I get...
```

[21:14:07:671] U-Boot SPL 2017.03-rc2-00252-gc987c17120-dirty (Jul 17 2019 - 14:12:05)
[21:14:07:684] DRAM: 256 MiB
[21:14:07:690] Trying to boot from sunxi SPI
[21:14:08:408] 
[21:14:08:414] 
[21:14:08:414] U-Boot 2017.03-rc2-00252-gc987c17120-dirty (Jul 17 2019 - 14:12:05 +1000) Allwinner Technology
[21:14:08:420] 
[21:14:08:420] CPU:   Allwinner H3 (SUN8I 1680)
[21:14:08:430] Model: Xunlong Orange Pi Zero
[21:14:08:430] DRAM:  256 MiB
[21:14:08:441] MMC:   SUNXI SD/MMC: 0
[21:14:08:446] MMC: no card present
[21:14:08:446] mmc_init: -123, time 2
[21:14:08:446] *** Warning - MMC init failed, using default environment
[21:14:08:452] 
[21:14:08:456] In:    serial
[21:14:08:456] Out:   serial
[21:14:08:456] Err:   serial
[21:14:08:463] Net:   phy interface0
[21:14:08:463] eth0: ethernet@1c30000
[21:14:08:463] starting USB...
[21:14:08:470] USB0:   USB EHCI 1.00
[21:14:08:478] USB1:   USB OHCI 1.0
[21:14:08:478] scanning bus 0 for devices... 1 USB Device(s) found
[21:14:09:611]        scanning usb for storage devices... 0 Storage Device(s) found
[21:14:09:619] Hit any key to stop autoboot:  2 <0x08><0x08><0x08> 1 <0x08><0x08><0x08> 0 
[21:14:11:623] SF: Detected mx25l1605d with page size 256 Bytes, erase size 64 KiB, total 2 MiB
[21:14:11:633] device 0 offset 0x180000, size 0x20000
[21:14:11:691] SF: 131072 bytes @ 0x180000 Read: OK
[21:14:11:694] ## Booting kernel from Legacy Image at 40000000 ...
[21:14:11:702]    Image Name:   http://www.orangepi-dmx.org
[21:14:11:707]    Image Type:   ARM U-Boot Standalone Program (uncompressed)
[21:14:11:712]    Data Size:    133888 Bytes = 130.8 KiB
[21:14:11:712]    Load Address: 40000000
[21:14:11:719]    Entry Point:  40000000
[21:14:11:719]    Verifying Checksum ... Bad Data CRC
[21:14:11:726] ERROR: can't get kernel image!
[21:14:11:726] => 
```

hippyau, 2019-07-17:
Quick update,  I have now compiled my own uImage and uploaded.

The "Data Size" reported by U-Boot shows as 136032 bytes, though the file is showing in `ls -la` as 136096 bytes.  Maybe this matters, or I'm doing something stupid...

`-rw-rw-r--  1 hip hip  136096 Jul 17 21:28 orangepi_zero.uImage`

U-Boot shows...

```
[21:31:46:222] ## Booting kernel from Legacy Image at 40000000 ...
[21:31:46:230]    Image Name:   http://www.orangepi-dmx.org
[21:31:46:234]    Image Type:   ARM U-Boot Standalone Program (uncompressed)
[21:31:46:236]    Data Size:    136032 Bytes = 132.8 KiB
[21:31:46:242]    Load Address: 40000000
[21:31:46:242]    Entry Point:  40000000
[21:31:46:245]    Verifying Checksum ... Bad Data CRC
[21:31:46:251] ERROR: can't get kernel image!
[21:31:46:252] => 
```

vanvught, 2019-07-17:
As the firmware is now writing the supported SPI boot and firmware to SPI flash, the mentioned page is not supported anymore. The page has been deleted. 
See instructions on https://github.com/vanvught/h3dmx512-zip for how to setup the SDCard for initial setup.

closed 2019-07-17 by vanvught

hippyau, 2019-07-17:
Ahh, thanks :+1: