vanvught/rpidmx512 #54: bcm2835.h not found

Breidenbach, 2019-08-30:
I cloned the library to my Raspberry Pi, entered the lib-tlc59711/examples directory, and tried to remake pwmled.  When I execute make, I get bcm2835.h not found.
[image]

vanvught, 2019-08-30:
This indicates that the prerequisite library is missing -> https://github.com/vanvught/rpidmx512/tree/master/lib-tlc59711

> Linux Raspbian
> Prerequisite: C library for Broadcom BCM 2835 as used in Raspberry Pi

Breidenbach, 2019-08-30:
And when I installed the library, it worked perfectly.

Thanks for the help.

closed 2019-08-30 by Breidenbach