vanvught/rpidmx512 #55: SMPTE Error in arrange. LTC to ArtNet & MIDI to ArtNET

YANG-Mingcong, 2019-08-31:
I just made some test and find out some bugs.
I input in many kind of signals and out put into ArtNET.

Input type   |         input        |         display      |     output ArtNet  |  status
          LTC    |   hh:mm:ss.ff   |    hh:mm:ss.ff   |      ff:ss:mm.hh    |   error
          MIDI   |   hh:mm:ss.ff   |    hh:mm:ss.ff   |      ff:ss:mm.hh    |   error
      rtpMIDI  |   hh:mm:ss.ff   |    hh:mm:ss.ff   |      hh:mm:ss.ff    |   correct
       ArtNet  |   hh:mm:ss.ff   |    hh:mm:ss.ff   |      hh:mm:ss.ff    |   correct

I wish I could find out why it wrong. ...

vanvught, 2019-08-31:
Thank you for using my firmware. I will try to replicate your issue seen.

vanvught, 2019-08-31:
Issue confirmed.

The wrong type of struct is used here:

- https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/h3/ltcreader.cpp#L283
- https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/h3/ltcreader.cpp#L287
- https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/h3/midireader.cpp#L244
- https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/h3/midireader.cpp#L248

vanvught, 2019-08-31:
@YANG-Mingcong I have attached the updated firmware. Please will you give it a test and let me know the outcome. Thanks, Arjan

Note: There is also a new input: LTC generator

- http://www.orangepi-dmx.org/orange-pi-smpte-timecode-ltc-reader-converter/udp-messages
- https://github.com/vanvught/h3dmx512-zip/tree/master/MA2/OPi%20GEN%20LUA

[opi_emac_ltc_smpte.zip](https://github.com/vanvught/rpidmx512/files/3562460/opi_emac_ltc_smpte.zip)

YANG-Mingcong, 2019-09-01:
Thanks a lot! I will test this Monday(GMT+8)! And feedback!
Thanks again!

YANG-Mingcong, 2019-09-02:
@vanvught Thank you very much!
It works in today's test.
LTC input hh:mm:ss.ff, Display hh:mm:ss.ff and ArtNET output hh:mm:ss.ff.

It's correct!

THANKS AGAIN!

closed 2019-09-02 by YANG-Mingcong