vanvught/rpidmx512 #57: Flashing Data

YANG-Mingcong, 2019-09-10:
Hello
Me again. 
I just find out another little situation.

I'm using one OPi as SMPTE conventer to receiving LTC and output to ArtNet in LAN. 
Using another OPi as SMPTE receiver to receiving the ArtNet TC and display.

I find out when I plug in the SMPTE receiver and receiving ArtNet TC, the displayer will shortly flashing once or twice in first 30sec. After it runs well.

I wish to find out what cause that ,so I connect a LTC output from the SMPTE Receiver which receiving ArtNET TC. And the LTC output is "missing" for one frame or two(not very sure the time) at the same time with displayer flashing.

I'm trying to figure out it caused by what.

maybe bug?

vanvught, 2019-09-10:
Hi @YANG-Mingcong 

> I find out when I plug in the SMPTE receiver and receiving ArtNet TC, the displayer will shortly flashing once or twice in first 30sec. After it runs well.

I do not understand this situation. When reading Art-Net TC, then the display will go blank (or shows the system time, _show_systime=1_ ), when the updates are less then 24 frames per second. I cannot correlate the 30 seconds. 

> I wish to find out what cause that ,so I connect a LTC output from the SMPTE Receiver which receiving ArtNET TC. And the LTC output is "missing" for one frame or two(not very sure the time) at the same time with displayer flashing.

Yes, this looks like that the network packets are receiving with less then 24 times per second.

I will try to reproduce the situation.

vanvught, 2019-09-10:
@YANG-Mingcong 

I have created a setup with the configuration shown below.  I am using the firmware from #55 
At this moment, I am not able to reproduce your situation. I will do some more debugging in the weekend. - Arjan 

[image]
[image]
.

vanvught, 2019-09-22:
@YANG-Mingcong  I am not reproduce your situation. Are you able to analyze the arrival of the Art-Net Timecode packets?

vanvught, 2019-10-20:
@YANG-Mingcong Are you able to analyze the arrival of the Art-Net Timecode packets? Please will you gave the latest firmware a test. Thanks, Arjan

@hippyau Hip, maybe you have some ideas for what is happening here?

vanvught, 2019-12-12:
No activity

closed 2019-12-12 by vanvught