vanvught/rpidmx512 #62: error: conflicting types for 'widget_get_received_dmx_packet_count'

hippyau, 2019-10-09:
Just pulled last night, and I can't ./makeall_firmware_h3-lib.sh
I end up with the building of lib-widget failing and the compilation stops.

```
[../lib-widget]
rm -rf build_h3
rm -rf lib_h3
mkdir -p build_h3/src build_h3/src/h3
mkdir -p lib_h3
arm-none-eabi-gcc -DBARE_METAL -DH3 -DORANGE_PI  -DNDEBUG  -I./include -I../include -I../lib-hal/include -I../lib-debug/include -I../lib-h3/include -I../lib-arm/include  -I../lib-hal/include -I../lib-monitor/include -I../lib-dmx/include -I../lib-rdm/include -I../lib-properties/include -I../lib-usb/include -I../lib-console/include -I../lib-ff12c/src -mfpu=neon-vfpv4 -mcpu=cortex-a7 -mhard-float -mfloat-abi=hard -Wall -Werror -O2 -nostartfiles -ffreestanding -nostdinc -nostdlib -fno-exceptions -fno-unwind-tables -fprefetch-loop-arrays -c src/widget_usb.c -o build_h3/src/widget_usb.o
arm-none-eabi-gcc -DBARE_METAL -DH3 -DORANGE_PI  -DNDEBUG  -I./include -I../include -I../lib-hal/include -I../lib-debug/include -I../lib-h3/include -I../lib-arm/include  -I../lib-hal/include -I../lib-monitor/include -I../lib-dmx/include -I../lib-rdm/include -I../lib-properties/include -I../lib-usb/include -I../lib-console/include -I../lib-ff12c/src -mfpu=neon-vfpv4 -mcpu=cortex-a7 -mhard-float -mfloat-abi=hard -Wall -Werror -O2 -nostartfiles -ffreestanding -nostdinc -nostdlib -fno-exceptions -fno-unwind-tables -fprefetch-loop-arrays -c src/monitor_line.c -o build_h3/src/monitor_line.o
arm-none-eabi-gcc -DBARE_METAL -DH3 -DORANGE_PI  -DNDEBUG  -I./include -I../include -I../lib-hal/include -I../lib-debug/include -I../lib-h3/include -I../lib-arm/include  -I../lib-hal/include -I../lib-monitor/include -I../lib-dmx/include -I../lib-rdm/include -I../lib-properties/include -I../lib-usb/include -I../lib-console/include -I../lib-ff12c/src -mfpu=neon-vfpv4 -mcpu=cortex-a7 -mhard-float -mfloat-abi=hard -Wall -Werror -O2 -nostartfiles -ffreestanding -nostdinc -nostdlib -fno-exceptions -fno-unwind-tables -fprefetch-loop-arrays -c src/monitor_time_uptime.c -o build_h3/src/monitor_time_uptime.o
arm-none-eabi-gcc -DBARE_METAL -DH3 -DORANGE_PI  -DNDEBUG  -I./include -I../include -I../lib-hal/include -I../lib-debug/include -I../lib-h3/include -I../lib-arm/include  -I../lib-hal/include -I../lib-monitor/include -I../lib-dmx/include -I../lib-rdm/include -I../lib-properties/include -I../lib-usb/include -I../lib-console/include -I../lib-ff12c/src -mfpu=neon-vfpv4 -mcpu=cortex-a7 -mhard-float -mfloat-abi=hard -Wall -Werror -O2 -nostartfiles -ffreestanding -nostdinc -nostdlib -fno-exceptions -fno-unwind-tables -fprefetch-loop-arrays -c src/widget_sniffer.c -o build_h3/src/widget_sniffer.o
arm-none-eabi-gcc -DBARE_METAL -DH3 -DORANGE_PI  -DNDEBUG  -I./include -I../include -I../lib-hal/include -I../lib-debug/include -I../lib-h3/include -I../lib-arm/include  -I../lib-hal/include -I../lib-monitor/include -I../lib-dmx/include -I../lib-rdm/include -I../lib-properties/include -I../lib-usb/include -I../lib-console/include -I../lib-ff12c/src -mfpu=neon-vfpv4 -mcpu=cortex-a7 -mhard-float -mfloat-abi=hard -Wall -Werror -O2 -nostartfiles -ffreestanding -nostdinc -nostdlib -fno-exceptions -fno-unwind-tables -fprefetch-loop-arrays -c src/widget.c -o build_h3/src/widget.o
src/widget.c:87:10: error: conflicting types for 'widget_get_received_dmx_packet_count'
 **uint32_t widget_get_received_dmx_packet_count(void) {
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~**
In file included from src/widget.c:37:0:
./include/widget.h:79:23: note: previous declaration of 'widget_get_received_dmx_packet_count' was here
 **extern const uint32_t widget_get_received_dmx_packet_count(void);
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~**
../h3-firmware-template/lib/Rules.mk:85: recipe for target 'build_h3/src/widget.o' failed
make: *** [build_h3/src/widget.o] Error 1
```

In `widget.h` I removed the const
`extern const uint32_t widget_get_received_dmx_packet_count(void);`
to
`extern uint32_t widget_get_received_dmx_packet_count(void);`

and it built.   

Cheers,
 hippy

vanvught, 2019-10-09:
Hi @hippyau , I am not able to reproduce your issue. Which gcc version are you using ? - Arjan

vanvught, 2019-10-09:
The issue is here -> https://github.com/vanvught/rpidmx512/blob/master/lib-widget/include/widget.h#L79
There should be no `const`. Some compilers may complain as shown above.
I will remove the const with a next commit.

hippyau, 2019-10-09:
Thanks Arjan,
```
hip@hip-Alienware-13:~/OrangePi/hippy/rpidmx512/scripts$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/6.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../src/configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:6.3.1+svn253039-1build1 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=rmprofile CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip
Thread model: single
gcc version 6.3.1 20170620 (15:6.3.1+svn253039-1build1) 
```

I've just discovered a couple more, 

```
[../lib-dmx]
mkdir -p build/src build/src/rpi build7/src build7/src/rpi
mkdir -p lib lib7
arm-none-eabi-gcc -mfpu=vfp -march=armv6zk -mtune=arm1176jzf-s -mcpu=arm1176jzf-s -DRPI1 -DBARE_METAL -DNDEBUG -I./include -I../include -I../lib-hal/include -I../lib-bcm2835/include -I../lib-arm/include -I../lib-debug/include  -I../lib-rdm/include -I../lib-properties/include -I../lib-console/include -Wall -Werror -O2 -nostartfiles -ffreestanding -nostdinc -nostdlib -mhard-float -mfloat-abi=hard -fno-exceptions -fno-unwind-tables -fprefetch-loop-arrays  -c src/rpi/dmx.c -o build/src/rpi/dmx.o
src/rpi/dmx.c:118:25: error: conflicting types for 'dmx_get_updates_per_seconde'
 const volatile uint32_t dmx_get_updates_per_seconde(void) {
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from src/rpi/dmx.c:44:0:
./include/dmx.h:98:26: note: previous declaration of 'dmx_get_updates_per_seconde' was here
 extern volatile uint32_t dmx_get_updates_per_seconde(void);
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/rpi/dmx.c:261:16: error: conflicting types for 'dmx_get_output_break_time'
 const uint32_t dmx_get_output_break_time(void) {
                ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from src/rpi/dmx.c:44:0:
./include/dmx.h:92:17: note: previous declaration of 'dmx_get_output_break_time' was here
 extern uint32_t dmx_get_output_break_time(void);
                 ^~~~~~~~~~~~~~~~~~~~~~~~~
../firmware-template/lib/Rules.mk:99: recipe for target 'build/src/rpi/dmx.o' failed
make: *** [build/src/rpi/dmx.o] Error 1
```

 I'm going to try and fix, but I don't know the best way to send you the changes,  I don't know git very well, I think make a push to my fork and then make a pull request?  I have not done that before :)

hippyau, 2019-10-09:
Hang on, I think i'm confused...

I just noticed this is /src/rpi/dmx.*,  but I am targeting Opi Zero,  but that is H2 processor,

do I use ./makeall_firmware_h3.sh  or ./makeall_firmware-lib.sh?  Sorry, numpty here :)

Cheers, 
  Hippy

vanvught, 2019-10-09:
./makeall_firmware_h3.sh  -> Allwinner SoC's H2+/H3 
./makeall_firmware.sh -> Broadcom SoC's (minimum support)

hippyau, 2019-10-09:
Awesome, thanks :)

vanvught, 2019-10-09:
Keeping this open for the commit with the fix.

commit ad407434 (2019-10-09): Fixed #62
closed 2019-10-09 by vanvught