vanvught/rpidmx512 #63: arm-none-eabi-ld: cannot find -luuid

hippyau, 2019-10-09:
Sorry, I'm stumped again :)

I saw in the commit history that lib-uuid was removed, but ./makeall_firmware_h3.sh gives me

```
hip@hip-Alienware-13:~/OrangePi/hippy/rpidmx512/scripts$ ./makeall_firmware_h3.sh

[../opi_dmx_monitor]

/home/hip/OrangePi/hippy/rpidmx512/scripts
[../opi_dmx_usb_pro]

/home/hip/OrangePi/hippy/rpidmx512/scripts
[../opi_emac_artnet_dmx]

../opi_emac_artnet_dmx
../opi_emac_artnet_dmx
make -f Makefile.H3 clean --directory=../lib-console
make[1]: Entering directory '/home/hip/OrangePi/hippy/rpidmx512/lib-console'
rm -rf build_h3
rm -rf lib_h3
make[1]: Leaving directory '/home/hip/OrangePi/hippy/rpidmx512/lib-console'
make -f Makefile.H3 clean --directory=../lib-display

.../\/\/\/\/\/\/...
  Much later...
.../\/\/\/\/\/\/...

make[1]: Leaving directory '/home/hip/OrangePi/hippy/rpidmx512/lib-arm'
arm-none-eabi-ld build_h3/vectors.o     build_h3/firmware/main.o  -Map orangepi_zero.map -T ./../h3-firmware-template/memmap -o build_h3/main.elf -L../lib-remoteconfig/lib_h3 -L../lib-tftpdeamon/lib_h3 -L../lib-rdmdiscovery/lib_h3 -L../lib-rdm/lib_h3 -L../lib-displayudf/lib_h3 -L../lib-artnet4/lib_h3 -L../lib-artnet/lib_h3 -L../lib-artnethandlers/lib_h3 -L../lib-e131/lib_h3 -L../lib-uuid/lib_h3 -L../lib-dmxsend/lib_h3 -L../lib-dmx/lib_h3 -L../lib-ws28xxdmx/lib_h3 -L../lib-ws28xx/lib_h3 -L../lib-tlc59711dmx/lib_h3 -L../lib-tlc59711/lib_h3 -L../lib-spiflashinstall/lib_h3 -L../lib-spiflashstore/lib_h3 -L../lib-spiflash/lib_h3 -L../lib-network/lib_h3 -L../lib-lightset/lib_h3 -L../lib-ledblink/lib_h3 -L../lib-properties/lib_h3 -L../lib-display/lib_h3 -L../lib-c++/lib_h3 -L../lib-hal/lib_h3 -L../lib-bob/lib_h3 -L../lib-i2c/lib_h3 -L../lib-utils/lib_h3 -L../lib-console/lib_h3 -L../lib-ff12c/lib_h3 -L../lib-h3/lib_h3 -L../lib-debug/lib_h3 -L../lib-arm/lib_h3 -lremoteconfig -ltftpdeamon -lrdmdiscovery -lrdm -ldisplayudf -lartnet4 -lartnet -lartnethandlers -le131 -luuid -ldmxsend -ldmx -lws28xxdmx -lws28xx -ltlc59711dmx -ltlc59711 -lspiflashinstall -lspiflashstore -lspiflash -lnetwork -llightset -lledblink -lproperties -ldisplay -lc++ -lhal -lbob -li2c -lutils -lconsole -lff12c -lh3 -ldebug -larm
arm-none-eabi-ld: cannot find -luuid
../h3-firmware-template/Rules.mk:179: recipe for target 'build_h3/main.elf' failed
make: *** [build_h3/main.elf] Error 1

```

hippyau, 2019-10-09:
FYI, full output of ./makeall_firmware_h3.sh
```
hip@hip-Alienware-13:~/OrangePi/hippy/rpidmx512/scripts$ ./makeall_firmware_h3.sh

[../opi_dmx_monitor]

/home/hip/OrangePi/hippy/rpidmx512/scripts
[../opi_dmx_usb_pro]

/home/hip/OrangePi/hippy/rpidmx512/scripts
[../opi_emac_artnet_dmx]

../opi_emac_artnet_dmx
../opi_emac_artnet_dmx
make -f Makefile.H3 clean --directory=../lib-console
make[1]: Entering directory '/home/hip/OrangePi/hippy/rpidmx512/lib-console'
rm -rf build_h3
rm -rf lib_h3
make[1]: Leaving directory '/home/hip/OrangePi/hippy/rpidmx512/lib-console'
make -f Makefile.H3 clean --directory=../lib-display
make[1]: Entering directory '/home/hip/OrangePi/hippy/rpidmx512/lib-display'
rm -rf build_h3
rm -rf lib_h3
make[1]: Leaving directory '/home/hip/OrangePi/hippy/rpidmx512/lib-display'
make -f Makefile.H3 clean --directory=../lib-h3
make[1]: Entering directory '/home/hip/OrangePi/hippy/rpidmx512/lib-h3'
rm -rf build_h3
rm -rf lib_h3
make[1]: Leaving directory '/home/hip/OrangePi/hippy/rpidmx512/lib-h3'
make -f Makefile.H3 clean --directory=../lib-hal
make[1]: Entering directory '/home/hip/OrangePi/hippy/rpidmx512/lib-hal'
rm -rf build_h3
rm -rf lib_h3
make[1]: Leaving directory '/home/hip/OrangePi/hippy/rpidmx512/lib-hal'
make -f Makefile.H3 clean --directory=../lib-remoteconfig
make[1]: Entering directory '/home/hip/OrangePi/hippy/rpidmx512/lib-remoteconfig'
rm -rf build_h3
rm -rf lib_h3
make[1]: Leaving directory '/home/hip/OrangePi/hippy/rpidmx512/lib-remoteconfig'
mkdir -p build_h3/firmware build_h3/lib
for d in ../lib-remoteconfig ../lib-tftpdeamon ../lib-rdmdiscovery ../lib-rdm ../lib-displayudf ../lib-artnet4 ../lib-artnet ../lib-artnethandlers ../lib-e131 ../lib-uuid ../lib-dmxsend ../lib-dmx ../lib-ws28xxdmx ../lib-ws28xx ../lib-tlc59711dmx ../lib-tlc59711 ../lib-spiflashinstall ../lib-spiflashstore ../lib-spiflash ../lib-network ../lib-lightset ../lib-ledblink ../lib-properties ../lib-display ../lib-c++ ../lib-hal ../lib-bob ../lib-i2c ../lib-utils ../lib-console ../lib-ff12c ../lib-h3 ../lib-debug ../lib-arm; \
	do                               \
		make -f Makefile.H3 'PLATFORM=ORANGE_PI' 'MAKE_FLAGS=-DARTNET_NODE -DDMXSEND -DDISPLAY_UDF -DNDEBUG' --directory=$d;       \
	done
make[1]: Entering directory '/home/hip/OrangePi/hippy/rpidmx512/lib-remoteconfig'
mkdir -p build_h3/src build_h3/src/h3
mkdir -p lib_h3
arm-none-eabi-g++ -DBARE_METAL -DH3 -DORANGE_PI  -DNDEBUG -DARTNET_NODE -DDMXSEND -DDISPLAY_UDF -DNDEBUG -I./include -I../include -I../lib-hal/include -I../lib-debug/include -I../lib-h3/include -I../lib-arm/include  -I../lib-display/include -I../lib-properties/include -I../lib-h3/include -I../lib-hal/include -I../lib-network/include -I../lib-spiflashinstall/include -I../lib-spiflashstore/include -I../lib-tftpdeamon/include -I../lib-rdm/include -I../lib-ws28xxdmx/include -I../lib-artnet/include -I../lib-artnet4/include -I../lib-e131/include -I../lib-network/include -I../lib-displayudf/include -I../lib-display/include -I../lib-dmxsend/include -I../lib-dmx/include -I../lib-ws28xxdmx/include -I../lib-ws28xx/include -I../lib-tlc59711dmx/include -I../lib-tlc59711/include -I../lib-ltc/include -I../lib-tcnet/include -I../lib-midi/include -I../lib-oscserver/include -I../lib-oscclient/include -I../lib-widget/include -I../lib-lightset/include -I../lib-ledblink/include -mfpu=neon-vfpv4 -mcpu=cortex-a7 -mhard-float -mfloat-abi=hard -Wall -Werror -O2 -nostartfiles -ffreestanding -nostdinc -nostdlib -fno-exceptions -fno-unwind-tables -fprefetch-loop-arrays -fno-rtti -std=c++11 -nostdinc++ -c src/tftpfileserver.cpp -o build_h3/src/tftpfileserver.o
arm-none-eabi-g++ -DBARE_METAL -DH3 -DORANGE_PI  -DNDEBUG -DARTNET_NODE -DDMXSEND -DDISPLAY_UDF -DNDEBUG -I./include -I../include -I../lib-hal/include -I../lib-debug/include -I../lib-h3/include -I../lib-arm/include  -I../lib-display/include -I../lib-properties/include -I../lib-h3/include -I../lib-hal/include -I../lib-network/include -I../lib-spiflashinstall/include -I../lib-spiflashstore/include -I../lib-tftpdeamon/include -I../lib-rdm/include -I../lib-ws28xxdmx/include -I../lib-artnet/include -I../lib-artnet4/include -I../lib-e131/include -I../lib-network/include -I../lib-displayudf/include -I../lib-display/include -I../lib-dmxsend/include -I../lib-dmx/include -I../lib-ws28xxdmx/include -I../lib-ws28xx/include -I../lib-tlc59711dmx/include -I../lib-tlc59711/include -I../lib-ltc/include -I../lib-tcnet/include -I../lib-midi/include -I../lib-oscserver/include -I../lib-oscclient/include -I../lib-widget/include -I../lib-lightset/include -I../lib-ledblink/include -mfpu=neon-vfpv4 -mcpu=cortex-a7 -mhard-float -mfloat-abi=hard -Wall -Werror -O2 -nostartfiles -ffreestanding -nostdinc -nostdlib -fno-exceptions -fno-unwind-tables -fprefetch-loop-arrays -fno-rtti -std=c++11 -nostdinc++ -c src/ubootheader.cpp -o build_h3/src/ubootheader.o
arm-none-eabi-g++ -DBARE_METAL -DH3 -DORANGE_PI  -DNDEBUG -DARTNET_NODE -DDMXSEND -DDISPLAY_UDF -DNDEBUG -I./include -I../include -I../lib-hal/include -I../lib-debug/include -I../lib-h3/include -I../lib-arm/include  -I../lib-display/include -I../lib-properties/include -I../lib-h3/include -I../lib-hal/include -I../lib-network/include -I../lib-spiflashinstall/include -I../lib-spiflashstore/include -I../lib-tftpdeamon/include -I../lib-rdm/include -I../lib-ws28xxdmx/include -I../lib-artnet/include -I../lib-artnet4/include -I../lib-e131/include -I../lib-network/include -I../lib-displayudf/include -I../lib-display/include -I../l [... cut ...] -lartnet4 -lartnet -lartnethandlers -le131 -luuid -ldmxsend -ldmx -lws28xxdmx -lws28xx -ltlc59711dmx -ltlc59711 -lspiflashinstall -lspiflashstore -lspiflash -lnetwork -llightset -lledblink -lproperties -ldisplay -lc++ -lhal -lbob -li2c -lutils -lconsole -lff12c -lh3 -ldebug -larm
arm-none-eabi-ld: cannot find -luuid
../h3-firmware-template/Rules.mk:179: recipe for target 'build_h3/main.elf' failed
make: *** [build_h3/main.elf] Error 1
hip@hip-Alienware-13:~/OrangePi/hippy/rpidmx512/scripts$

```

vanvught, 2019-10-09:
I cannot replicate this issue. Please make sure you run `./makeall_firmware_h3.sh clean `first.
The better option is to use `./build_h3-firmware.sh` only

vanvught, 2019-10-09:
Solution: Use  `./build_h3-firmware.sh `only
Or alternativily run `./makeall_firmware_h3.sh clean `first.

hippyau, 2019-10-09:
Sorry to bug you, but it's still a thing...  

I did a fresh clone, fixed the const issue in widget.h, ran `chmod +x ../opi_emac_artnet_rdm_l6470/generate_sofware_version_id.sh` and I still get jammed up not finding -luuid

```
...
make[1]: Leaving directory '/home/hip/OrangePi/test/rpidmx512/lib-arm'
arm-none-eabi-ld build_h3/vectors.o     build_h3/firmware/main.o build_h3/lib/identify.o build_h3/lib/rdmsoftwareversion.o -Map orangepi_one.map -T ./../h3-firmware-template/memmap -o build_h3/main.elf -L../lib-remoteconfig/lib_h3 -L../lib-tftpdeamon/lib_h3 -L../lib-rdmresponder/lib_h3 -L../lib-rdm/lib_h3 -L../lib-rdmsensor/lib_h3 -L../lib-rdmsubdevice/lib_h3 -L../lib-l6470dmx/lib_h3 -L../lib-l6470/lib_h3 -L../lib-displayudf/lib_h3 -L../lib-artnet4/lib_h3 -L../lib-artnet/lib_h3 -L../lib-artnethandlers/lib_h3 -L../lib-e131/lib_h3 -L../lib-uuid/lib_h3 -L../lib-ws28xxdmx/lib_h3 -L../lib-ws28xx/lib_h3 -L../lib-tlc59711dmx/lib_h3 -L../lib-tlc59711/lib_h3 -L../lib-spiflashinstall/lib_h3 -L../lib-spiflashstore/lib_h3 -L../lib-spiflash/lib_h3 -L../lib-network/lib_h3 -L../lib-lightset/lib_h3 -L../lib-ledblink/lib_h3 -L../lib-properties/lib_h3 -L../lib-display/lib_h3 -L../lib-c++/lib_h3 -L../lib-hal/lib_h3 -L../lib-bob/lib_h3 -L../lib-i2c/lib_h3 -L../lib-utils/lib_h3 -L../lib-console/lib_h3 -L../lib-ff12c/lib_h3 -L../lib-h3/lib_h3 -L../lib-debug/lib_h3 -L../lib-arm/lib_h3 -lremoteconfig -ltftpdeamon -lrdmresponder -lrdm -lrdmsensor -lrdmsubdevice -ll6470dmx -ll6470 -ldisplayudf -lartnet4 -lartnet -lartnethandlers -le131 -luuid -lws28xxdmx -lws28xx -ltlc59711dmx -ltlc59711 -lspiflashinstall -lspiflashstore -lspiflash -lnetwork -llightset -lledblink -lproperties -ldisplay -lc++ -lhal -lbob -li2c -lutils -lconsole -lff12c -lh3 -ldebug -larm
arm-none-eabi-ld: cannot find -luuid
../h3-firmware-template/Rules.mk:179: recipe for target 'build_h3/main.elf' failed
make: *** [build_h3/main.elf] Error 1
total 400
drwxrwxr-x 97 hip hip 4096 Oct  9 21:43 .
drwxrwxr-x  3 hip hip 4096 Oct  9 21:40 ..
drwxrwxr-x  4 hip hip 4096 Oct  9 21:43 E131
drwxrwxr-x  5 hip hip 4096 Oct  9 21:43 esp8266-ota
drwxrwxr-x  8 hip hip 4096 Oct  9 21:43 esp8266_rtos_sdk_rpi
drwxrwxr-x  3 hip hip 4096 Oct  9 21:43 firmware-template
drwxrwxr-x  8 hip hip 4096 Oct  9 21:43 .git
-rw-rw-r--  1 hip hip  984 Oct  9 21:43 .gitignore
drwxrwxr-x  3 hip hip 4096 Oct  9 21:43 h3-firmware-template
...\/\/\/\/\/\/\/...
drwxrwxr-x  6 hip hip 4096 Oct  9 21:43 rpi_wifi_osc_dmx
drwxrwxr-x  2 hip hip 4096 Oct  9 21:43 scripts
100
```

it's weird, but I think last time I did a full build was after you removed lib-uuid, and I did get there eventually.   I am having trouble working out who is calling for it.   

Love your work, i'm working on adding to it, i'm building a huge 3d-printed open-source 7-segment display TC clock with ws2812 leds, i've started hacking at code, but jammed up in compilation :)

Cheers, 
  Hippy

vanvught, 2019-10-09:
Hippy, I just noticed that lib-uuid is not on GitHub. My apologize for the troubles. I will commit lib-uuid as soon as possible. Arjan 

Verstuurd vanaf mijn iPhone

hippyau, 2019-10-09:
No worries Arjan, thank you :)

I removed uuid from from line 41 of Rules.mk in the h3-firmware-template, but weird because then compilation worked but only 5 firmwares built...

No stress :)  Thanks again!

vanvught, 2019-10-09:
Missing lib-uuid on GitHub

closed 2019-10-09 by vanvught

hippyau, 2019-10-09:
thank you, fixed in your commit :+1:

commit b843db3e (2019-10-09): Fixed #63