vanvught/rpidmx512 #64: generate_sofware_version_id.sh

hippyau, 2019-10-09:
This might be a git thing, with all the build scripts after a pull, I have to chmod +x *.sh

In order to get past 
```
mkdir -p build_h3/firmware build_h3/lib
./generate_sofware_version_id.sh
make: execvp: ./generate_sofware_version_id.sh: Permission denied
Makefile.H3:11: recipe for target 'prerequisites' failed
make: *** [prerequisites] Error 127
```
I had to... 
`rpidmx512/opi_emac_artnet_rdm_l6470$ chmod +x generate_sofware_version_id.sh`

vanvught, 2019-10-09:
It is fine on my local system. It might be a GitHub thing.

```
pi@nuc-i5:~/workspace$ find . -name generate_sofware_version_id.sh | xargs ls -al
-rwxr--r-- 1 pi pi 158 sep  8 14:45 ./lib-rdmnet/example/generate_sofware_version_id.sh
-rwxr--r-- 1 pi pi 178 jan  9  2017 ./linux_artnet/generate_sofware_version_id.sh
-rwxr--r-- 1 pi pi 178 jan  9  2017 ./linux_rpi_artnet_l6470/generate_sofware_version_id.sh
-rwxr--r-- 1 pi pi 178 jan  9  2017 ./linux_rpi_artnet_pixel_multi/generate_sofware_version_id.sh
-rwxr--r-- 1 pi pi 178 jan  9  2017 ./opi_emac_artnet_rdm_l6470/generate_sofware_version_id.sh
-rwxr--r-- 1 pi pi 178 jan  9  2017 ./opi_emac_llrp_tftpd/generate_sofware_version_id.sh
-rwxr--r-- 1 pi pi 178 okt  5 12:07 ./opi_rdm_responder/generate_sofware_version_id.sh
-rwxr--r-- 1 pi pi 178 jan  9  2017 ./opi_rdm_responder_l6470/generate_sofware_version_id.sh
-rwxr--r-- 1 pi pi 178 jan  9  2017 ./scratch-linux-artnet/generate_sofware_version_id.sh
-rwxr--r-- 1 pi pi 178 jan  9  2017 ./scratch-linux-e133/generate_sofware_version_id.sh
```

closed 2019-10-09 by vanvught

hippyau, 2019-10-09:
Further, i had to `chmod +x ../opi_rdm_responder_l6470/generate_sofware_version_id.sh`

Maybe these can be executed by the build system as `bash ./my_script.sh`?

Cheers,
 hippy