vanvught/rpidmx512 #65: build_h3-firmware.sh not building all firmwares

hippyau, 2019-10-09:
Sorry,  i'm running `./build_h3-firmware.sh` on a fresh pull,  
 
compilation is completing...

```
[../rpi_wifi_osc_dmx]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
Skipping...
/home/hip/OrangePi/test/rpidmx512/scripts
-rw-rw-r-- 1 hip hip  48888 Oct  9 22:31 ./opi_dmx_monitor/orangepi_one.uImage
-rw-rw-r-- 1 hip hip  65544 Oct  9 22:30 ./opi_dmx_usb_pro/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 131904 Oct  9 22:31 ./opi_emac_artnet_dmx_multi/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 126352 Oct  9 22:29 ./opi_emac_artnet_dmx_multi/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 125776 Oct  9 22:29 ./opi_emac_artnet_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  96492 Oct  9 22:31 ./opi_emac_artnet_monitor/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 114216 Oct  9 22:30 ./opi_emac_artnet_pixel_multi/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 120364 Oct  9 22:29 ./opi_emac_artnet_pixel/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 135316 Oct  9 22:31 ./opi_emac_artnet_rdm_l6470/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 133036 Oct  9 22:30 ./opi_emac_artnet_rdm_l6470/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 107860 Oct  9 22:31 ./opi_emac_e131_dmx_multi/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 102244 Oct  9 22:30 ./opi_emac_e131_dmx_multi/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 102132 Oct  9 22:30 ./opi_emac_e131_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  73376 Oct  9 22:31 ./opi_emac_e131_monitor/orangepi_one.uImage
-rw-rw-r-- 1 hip hip  97580 Oct  9 22:30 ./opi_emac_e131_pixel_multi/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 103760 Oct  9 22:30 ./opi_emac_e131_pixel/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  72876 Oct  9 22:31 ./opi_emac_osc_monitor/orangepi_one.uImage
-rw-rw-r-- 1 hip hip  49696 Oct  9 22:31 ./opi_midi_monitor/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 106824 Oct  9 22:31 ./opi_rdm_responder_l6470/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 110544 Oct  9 22:30 ./rpi_wifi_artnet_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  99476 Oct  9 22:30 ./rpi_wifi_e131_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  97824 Oct  9 22:30 ./rpi_wifi_osc_dmx/orangepi_zero.uImage
22
```
but the firmware i'm interested (../opi_emac_ltc_smpte) and a bunch of others are not getting built...
```
....\/\/\/\/\/\/....
[../opi_emac_e131_dmx]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_emac_e131_dmx_multi]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_emac_e131_monitor]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_emac_e131_pixel]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_emac_e131_pixel_multi]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_emac_llrp_tftpd]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_emac_ltc_smpte]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_emac_osc_client]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_emac_osc_dmx]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_emac_osc_monitor]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_emac_osc_pixel]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_midi_monitor]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_rdm_responder]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
/home/hip/OrangePi/test/rpidmx512/scripts
[../opi_rdm_responder_l6470]
PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
../opi_rdm_responder_l6470 PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
../opi_rdm_responder_l6470 PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1
make -f Makefile.H3 clean --directory=../lib-console
make[1]: Entering directory '/home/hip/OrangePi/test/rpidmx512/lib-console'
rm -rf build_h3
rm -rf lib_h3
[ build continues, opi_rdm_responder_l6470 gets built]
....\/\/\/\/\/\....
```

I note the line `PLATFORM=ORANGE_PI_ONE NO_EXT_LED=1` repeated...
but not sure if relevant, I'm targeting Opi Zero.

Cheers, 
  Hip

vanvught, 2019-10-09:
I cannot replicate the issue.

```
-rw-rw-r-- 1 pi pi  48232 okt  9 14:56 ./opi_dmx_monitor/orangepi_one.uImage
-rw-rw-r-- 1 pi pi  65028 okt  9 14:56 ./opi_dmx_usb_pro/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi 125180 okt  9 14:55 ./opi_emac_artnet_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi 130988 okt  9 14:56 ./opi_emac_artnet_dmx_multi/orangepi_one.uImage
-rw-rw-r-- 1 pi pi 125724 okt  9 14:55 ./opi_emac_artnet_dmx_multi/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi  95784 okt  9 14:56 ./opi_emac_artnet_monitor/orangepi_one.uImage
-rw-rw-r-- 1 pi pi 119816 okt  9 14:55 ./opi_emac_artnet_pixel/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi 113844 okt  9 14:55 ./opi_emac_artnet_pixel_multi/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi 134016 okt  9 14:57 ./opi_emac_artnet_rdm_l6470/orangepi_one.uImage
-rw-rw-r-- 1 pi pi 132184 okt  9 14:55 ./opi_emac_artnet_rdm_l6470/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi 101376 okt  9 14:55 ./opi_emac_e131_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi 106800 okt  9 14:56 ./opi_emac_e131_dmx_multi/orangepi_one.uImage
-rw-rw-r-- 1 pi pi 101440 okt  9 14:55 ./opi_emac_e131_dmx_multi/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi  72496 okt  9 14:56 ./opi_emac_e131_monitor/orangepi_one.uImage
-rw-rw-r-- 1 pi pi 103116 okt  9 14:55 ./opi_emac_e131_pixel/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi  97112 okt  9 14:55 ./opi_emac_e131_pixel_multi/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi 110224 okt  9 14:55 ./opi_emac_llrp_tftpd/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi 137476 okt  9 14:55 ./opi_emac_ltc_smpte/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi  92828 okt  9 14:55 ./opi_emac_osc_client/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi 104216 okt  9 14:55 ./opi_emac_osc_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi  71292 okt  9 14:57 ./opi_emac_osc_monitor/orangepi_one.uImage
-rw-rw-r-- 1 pi pi 103984 okt  9 14:56 ./opi_emac_osc_pixel/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi  49104 okt  9 14:57 ./opi_midi_monitor/orangepi_one.uImage
-rw-rw-r-- 1 pi pi  84276 okt  9 14:56 ./opi_rdm_responder/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi 105624 okt  9 14:57 ./opi_rdm_responder_l6470/orangepi_one.uImage
-rw-rw-r-- 1 pi pi 103776 okt  9 14:56 ./opi_rdm_responder_l6470/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi 109420 okt  9 14:56 ./rpi_wifi_artnet_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi  98352 okt  9 14:56 ./rpi_wifi_e131_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 pi pi  96300 okt  9 14:56 ./rpi_wifi_osc_dmx/orangepi_zero.uImage
29
/home/pi/workspace/scripts
```

vanvught, 2019-10-09:
@hippyau Please try to find an error in the output. And let me know. Thanks.

vanvught, 2019-10-09:
Hi Hip,

When you just want to compile the LTC SMPTE firmware, then you can do:

```
pi@nuc-i5:~$ cd /development/workspace/opi_emac_ltc_smpte/
pi@nuc-i5:/development/workspace/opi_emac_ltc_smpte$ make -f Makefile.H3 clean && make -f Makefile.H3 && ./do-tftp.sh yourZeroIp
```

https://github.com/vanvught/h3dmx512-zip/blob/master/do-tftp.sh

hippyau, 2019-10-09:
Gawd damn it, it seems the generate_software_version_id.sh is screwing everything over for me :)

where the other ones would stop compilation, I got to the point that things were building - but not all of them.  so in the middle of a 2MB   build.txt dump...
```
arm-none-eabi-ar: creating lib_h3/libh3.a
../opi_emac_llrp_tftpd PLATFORM=ORANGE_PI
make: execvp: ./generate_sofware_version_id.sh: Permission denied
make: *** [prerequisites] Error 127
../opi_dmx_usb_pro PLATFORM=ORANGE_PI NO_EXT_LED=1 clean
../h3-firmware-template/Rules.mk:164: target '../lib-uuid/lib_h3/libuuid.a' given more than once in the same rule
../opi_dmx_usb_pro PLATFORM=ORANGE_PI NO_EXT_LED=1
arm-none-eabi-ar: creating lib_h3/libwidget.a
```

so I chmod every script in the directory...

`find ../ -type f -iname "*.sh" -exec chmod +x {} \;`

and now I can build :)

```
/home/hip/OrangePi/test/rpidmx512/scripts
-rw-rw-r-- 1 hip hip  48888 Oct  9 23:22 ./opi_dmx_monitor/orangepi_one.uImage
-rw-rw-r-- 1 hip hip  65544 Oct  9 23:21 ./opi_dmx_usb_pro/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 131904 Oct  9 23:22 ./opi_emac_artnet_dmx_multi/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 126352 Oct  9 23:20 ./opi_emac_artnet_dmx_multi/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 125776 Oct  9 23:20 ./opi_emac_artnet_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  96492 Oct  9 23:22 ./opi_emac_artnet_monitor/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 114216 Oct  9 23:20 ./opi_emac_artnet_pixel_multi/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 120364 Oct  9 23:20 ./opi_emac_artnet_pixel/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 135316 Oct  9 23:22 ./opi_emac_artnet_rdm_l6470/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 133036 Oct  9 23:20 ./opi_emac_artnet_rdm_l6470/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 107860 Oct  9 23:22 ./opi_emac_e131_dmx_multi/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 102244 Oct  9 23:20 ./opi_emac_e131_dmx_multi/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 102132 Oct  9 23:20 ./opi_emac_e131_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  73376 Oct  9 23:22 ./opi_emac_e131_monitor/orangepi_one.uImage
-rw-rw-r-- 1 hip hip  97580 Oct  9 23:20 ./opi_emac_e131_pixel_multi/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 103760 Oct  9 23:20 ./opi_emac_e131_pixel/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 111508 Oct  9 23:21 ./opi_emac_llrp_tftpd/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 138412 Oct  9 23:21 ./opi_emac_ltc_smpte/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  94032 Oct  9 23:21 ./opi_emac_osc_client/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 105644 Oct  9 23:21 ./opi_emac_osc_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  72876 Oct  9 23:22 ./opi_emac_osc_monitor/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 105316 Oct  9 23:21 ./opi_emac_osc_pixel/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  49696 Oct  9 23:22 ./opi_midi_monitor/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 106824 Oct  9 23:22 ./opi_rdm_responder_l6470/orangepi_one.uImage
-rw-rw-r-- 1 hip hip 104512 Oct  9 23:21 ./opi_rdm_responder_l6470/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  85076 Oct  9 23:21 ./opi_rdm_responder/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 110544 Oct  9 23:21 ./rpi_wifi_artnet_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  99476 Oct  9 23:21 ./rpi_wifi_e131_dmx/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip  97824 Oct  9 23:21 ./rpi_wifi_osc_dmx/orangepi_zero.uImage
29
```

I did notice this, but I don't think it's an show stopper...
```
arm-none-eabi-ar: creating lib_h3/libconsole.a
arm-none-eabi-ar: creating lib_h3/libh3.a
../opi_dmx_usb_pro PLATFORM=ORANGE_PI NO_EXT_LED=1 clean
../h3-firmware-template/Rules.mk:164: target '../lib-uuid/lib_h3/libuuid.a' given more than once in the same rule
../opi_dmx_usb_pro PLATFORM=ORANGE_PI NO_EXT_LED=1
arm-none-eabi-ar: creating lib_h3/libwidget.a
arm-none-eabi-ar: creating lib_h3/librdm.a
```

Thanks for your help Arjan!

vanvught, 2019-10-09:
> Thanks for your help Arjan!

You're welcome! Please show some photo's here -> [https://www.facebook.com/groups/584412254990472/](https://www.facebook.com/groups/584412254990472/) when your project is finished.

I will implement a solution for the missing +x. Keeping this issue open until resolved.

vanvught, 2019-10-09:
Adding the third line in Rules.mk

```
builddirs:
	mkdir -p $(BUILD_DIRS)
	[ -f generate_sofware_version_id.sh ] && chmod u+x generate_sofware_version_id.sh || true
```

commit 705e3b3a (2019-10-09): Fixed #65
closed 2019-10-09 by vanvught