vanvught/rpidmx512 #66: Add support for Nextion HMI Display's.

vanvught, 2019-10-11:
See [https://github.com/vanvught/opi-nextion](https://github.com/vanvught/opi-nextion)

- [ ] Move shared functions from lib-remoteconfig to lib-spiflashstore

hippyau, 2019-12-09:
Hi Arjan,

 Fresh clone of staging and build is failing, several times because of missing file...
```
In file included from src/storenextion.cpp:28:0:
./include/storenextion.h:29:27: fatal error: nextionparams.h: No such file or directory
 #include "nextionparams.h"
                ^
```

Cheers :)

vanvught, 2019-12-09:
Hi Hippy,

Oh shoot ;-) The idea of the staging branch is not having the Nextion dependency. I will investigate what went wrong with my commits. And of course...fix it.

Greets, Arjan

hippyau, 2019-12-09:
All good...  I am doing fresh clone compiles so to test "out of the box" experience for users for V2.2

I will use my experience here to help with writing a compile how-to.  there are still things like have to `chmod +x *.sh` in the scripts directory,  these are easy to do if I write up a how-to for newbies.

Legend!
Cheers, 
  Hippy

vanvught, 2019-12-09:
Hi Hippy, please will you check if you have cloned the staging branch? When looking at GitHub, there is no  storenextion.cpp in the staging branch. Cheers! Arjan

hippyau, 2019-12-09:
I am soooo sorry to waste your time Arjan...

I did not clone the repo properly!   
 `git clone --branch staging https://github.com/vanvught/rpidmx512.git`
builds clean!!  Thank you!    Doh!  

I'll begin testing :)

I can see the compressed firmware will be useful also!
```
**[./opi_emac_artnet_rdm_l6470/orangepi_one.uImage is too big for SPI Flash -> 151964]**
-rw-rw-r-- 1 hip hip 107376 Dec  9 20:58 ./opi_emac_e131_pixel/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 114716 Dec  9 20:58 ./opi_emac_llrp_tftpd/orangepi_zero.uImage
**[./opi_emac_ltc_smpte/orangepi_zero.uImage is too big for SPI Flash -> 154288]**
-rw-rw-r-- 1 hip hip 82983 Dec  9 20:58 ./opi_emac_ltc_smpte/orangepi_zero.uImage.gz
-rw-rw-r-- 1 hip hip 97424 Dec  9 20:58 ./opi_emac_osc_client/orangepi_zero.uImage
-rw-rw-r-- 1 hip hip 109252 Dec  9 20:58 ./opi_emac_osc_dmx/orangepi_zero.uImage

```

Cheers,
  Hip

hippyau, 2019-12-09:
PS.  When I pulled staging, the scripts were still executable :)

vanvught, 2019-12-09:
> Op 9 dec. 2019, om 12:04 heeft Hippy <notifications@github.com> het volgende geschreven:
> 
> I am soooo sorry to waste your time Arjan...
> 
> 
No worries!

vanvught, 2020-12-24:
No user feedback.

closed 2020-12-24 by vanvught