vanvught/rpidmx512 #67: TFTP Server: Can't make it work :(

hippyau, 2019-10-13:
Hi Arjan!

 Making good progress, except I can't upload via tftp, and SD card swapping is driving me crazy :)

Initially, I was getting no action and just timeouts.  Turns out was blocked by the Ubuntu firewall, I fixed by:

`sudo ufw allow from 192.168.1.82 to any`

Now I am actually sending it something, but jammed up here...

```
hip@hip-Alienware-13:~/OrangePi/hippy/rpidmx512/opi_emac_ltc_smpte$ ../scripts/do-tftp.sh 192.168.1.82
tftp:On
tftp> Verbose mode on.
tftp> mode set to octet
tftp> putting orangepi_zero.uImage to 192.168.1.82:orangepi_zero.uImage [octet]
Error code 3: Write failed
Sent 139264 bytes in 0.1 seconds [11141120 bits/sec]
tftp> tftp:Off
hip@hip-Alienware-13:~/OrangePi/hippy/rpidmx512/opi_emac_ltc_smpte$ 
```
I can see the `Error code 3: Write failed` is  `ERROR_CODE_DISK_FULL = 3,	///< Disk full or allocation exceeded.`

My orangepi_zero.uImage is **141804** bytes, bigger than the one I'm replacing, which is **141740** bytes:
 
```
   Image Name:   http://www.orangepi-dmx.org
   Image Type:   ARM U-Boot Standalone Program (uncompressed)
   Data Size:    141740 Bytes = 138.4 KiB
```

How to get around this?

Cheers,
  Hip

vanvught, 2019-10-13:
Hi Hip,

Please see the comment here -> https://github.com/vanvught/rpidmx512/blob/master/lib-remoteconfig/src/remoteconfig.cpp#L99
There is the maximum size defined -> 0x22000 which is 136 kB

- Arjan

closed 2019-10-13 by vanvught

hippyau, 2019-10-13:
Oh wow, okay,  explains why my images from SD are not booting from SPI flash also.  

If I build a new u-boot, how big do you think I can grow before running out of SPI flash space?  I mean, is this limit 136kb for a reason?

Thanks again!