vanvught/rpidmx512 #68: U-Boot SPI: Increase firmware size

vanvught, 2019-10-13:
Follow-up for #67 
cc: @hippyau

vanvught, 2019-10-13:
Currently the maximum size is 0x22000 bytes. (138kB)
In #67 there is a firmware with size: 138.4 KiB
With #66 there is also mostly likely a need for an increased maximum size.

hippyau, 2019-10-13:
I was able to 
```
$ gzip orangepi_zero.img
$ mkimage -n 'http://www.orangepi-dmx.org' -A arm -O u-boot -T standalone -C gzip -a 0x40000000 -d orangepi_zero.img.gz opi.uImage
```
...and the output was  76334 bytes, and it booted from the SD, wrote itself to the SPI flash, and worked fine, until the SD card was removed.

```
...\/\/\/\/\/\/...
76398 bytes read in 7 ms (10.4 MiB/s)
## Booting kernel from Legacy Image at 48000000 ...
   Image Name:   http://www.orangepi-dmx.org
   Image Type:   ARM U-Boot Standalone Program (gzip compressed)
   Data Size:    76334 Bytes = 74.5 KiB
   Load Address: 40000000
   Entry Point:  40000000
   Verifying Checksum ... OK
EHCI failed to shut down host controller.
EHCI failed to shut down host controller.
   Uncompressing Standalone Program ... OK
MX25L1605D, sector size 4096, 2097152 bytes
uboot.spi
Check difference
No difference
Done
uImage
Check difference
Writing
76398 bytes written
Done
...\/\/\/\/\/\/...
```

However, removing the SD and trying to boot from SPI flash failed?

```
U-Boot SPL 2017.03-rc2-00252-gc987c17-dirty (Mar 10 2019 - 18:20:54)
DRAM: 256 MiB
Trying to boot from sunxi SPI

U-Boot 2017.03-rc2-00252-gc987c17-dirty (Mar 10 2019 - 18:20:54 +0100) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Xunlong Orange Pi Zero
DRAM:  256 MiB
MMC:   SUNXI SD/MMC: 0
spi_flash_decode_fdt: Cannot decode address
SF: Detected mx25l1605d with page size 256 Bytes, erase size 64 KiB, total 2 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   phy interface0
eth0: ethernet@1c30000
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
spi_flash_decode_fdt: Cannot decode address
SF: Detected mx25l1605d with page size 256 Bytes, erase size 64 KiB, total 2 MiB
device 0 offset 0x180000, size 0x22000
SF: 139264 bytes @ 0x180000 Read: OK
## Booting kernel from Legacy Image at 40000000 ...
   Image Name:   http://www.orangepi-dmx.org
   Image Type:   ARM U-Boot Standalone Program (gzip compressed)
   Data Size:    76334 Bytes = 74.5 KiB
   Load Address: 40000000
   Entry Point:  40000000
   Verifying Checksum ... OK
   Uncompressing Standalone Program ... Error: inflate() returned -3
gzip compressed: uncompress error -1
Must RESET board to recover
resetting ...
[ boot loop]
```

Can we install the `U-Boot 2018.09 (Mar 02 2019 - 08:59:42 +0100) Allwinner Technology` to the SPI flash?

vanvught, 2019-10-13:
> However, removing the SD and trying to boot from SPI flash failed?

I do not support compressed firmware. The compressed firmware is written to SPI flash, hence the boot from SPI splash will fail.

> Can we install the U-Boot 2018.09 (Mar 02 2019 - 08:59:42 +0100) Allwinner Technology to the SPI flash?

U-Boot 2017.03-rc2-00252-gc987c17-dirty is known that SPI flash is working. Switching to an unknown release would take a lot of testing and debugging. 

```
pi@nuc-i5:~/uboot-spi/u-boot$ cat include/generated/version_autogenerated.h
#define PLAIN_VERSION "2017.03-rc2-00252-gc987c17-dirty"
#define U_BOOT_VERSION "U-Boot " PLAIN_VERSION
#define CC_VERSION_STRING "arm-none-eabi-gcc (GCC) 7.3.0"
#define LD_VERSION_STRING "GNU ld (GNU Binutils) 2.30"
pi@nuc-i5:~/uboot-spi/u-boot$
```

hippyau, 2019-10-13:
oh okay, I see.   I don't really know what i'm doing with u-boot...  It seemed like it was going to work :) 
Cheers :)

vanvught, 2019-10-13:
@hippyau  I will invest if supporting compressed firmware won't break any current SDCArds in use.
As a quick fix you, I will attach a new uboot.spi here which supports increased firmware size.

I am going for +16Kb. Ok?

There is no technical limitation; but with keeping the installed firmware small, I `force` myself creating highly re-usable code with small footprint.

hippyau, 2019-10-13:
@vanvught that would be awesome, but no stress, I can keep swapping in/out SD card for testing my code.  Compressed firmware for the SPI flash would be good for the long run I think. 

You are doing great work here, I enjoy reading through your code!

vanvught, 2019-10-13:
@hippyau 

Update remoteconfig.cpp with:
```
// nuc-i5:~/uboot-spi/u-boot$ grep CONFIG_BOOTCOMMAND include/configs/sunxi-common.h
// #define CONFIG_BOOTCOMMAND "sf probe; sf read 40000000 180000 26000; bootm 40000000"
#define FIRMWARE_MAX_SIZE	0x26000
```

[uboot-spi.zip](https://github.com/vanvught/rpidmx512/files/3721635/uboot-spi.zip)

hippyau, 2019-10-13:
Works great, thank you @vanvught, I am tftp-ing!

hippyau, 2019-10-13:
@vanvught I got gzip compressed uImage to boot from SPI flash working in U-Boot SPL 2017.03-rc2-00252-gc987c17120-dirty,  just by loading it to a higher memory address.  I noticed the U-Boot 2018.09 from the SD card did this.  Makes sense now that gunzip was trying to decompress straight over the top of the compressed image in memory, destroying it.  

I booted gziped uImage from SD card, which wrote it to SPI flash, removed the SD card, rebooted and interrupted the autoboot, and executed:

`sf probe; sf read 48000000 180000 26000; bootm 48000000`

but I don't know if there are other implications, or how to make this persist (I guess is built into uboot.spi)

```
U-Boot SPL 2017.03-rc2-00252-gc987c17120-dirty (Oct 13 2019 - 12:27:10)
DRAM: 256 MiB
Trying to boot from sunxi SPI

U-Boot 2017.03-rc2-00252-gc987c17120-dirty (Oct 13 2019 - 12:27:10 +0200) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Xunlong Orange Pi Zero
DRAM:  256 MiB
MMC:   SUNXI SD/MMC: 0
spi_flash_decode_fdt: Cannot decode address
SF: Detected mx25l1605d with page size 256 Bytes, erase size 64 KiB, total 2 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   phy interface0
eth0: ethernet@1c30000
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
```
` =>` `sf probe; sf read 48000000 180000 26000; bootm 48000000`

 ```
spi_flash_decode_fdt: Cannot decode address
SF: Detected mx25l1605d with page size 256 Bytes, erase size 64 KiB, total 2 MiB
device 0 offset 0x180000, size 0x26000
SF: 155648 bytes @ 0x180000 Read: OK
## Booting kernel from Legacy Image at 48000000 ...
   Image Name:   http://www.orangepi-dmx.org
   Image Type:   ARM U-Boot Standalone Program (gzip compressed)
   Data Size:    77573 Bytes = 75.8 KiB
   Load Address: 40000000
   Entry Point:  40000000
   Verifying Checksum ... OK
   Uncompressing Standalone Program ... OK
MX25L1605D, sector size 4096, 2097152 bytes
Detected MX25L1605D with sector size 4096 total 2097152 bytes
[V2.0] Orange Pi Zero Compiled on Oct 13 2019 at 21:08:10
Network init
Network
 Hostname : allwinner_1F0A53
 If       : 1: eth0
 Inet     : 192.168.1.82 /24
 Netmask  : 255.255.255.0
 Gateway  : 192.168.1.1
 Mac      : 02:42:18:1f:0a:53
 DHCP     : Yes
Setting Node parameters
Starting the Node
Node 3
 Firmware   : 1.38
 Short name : LTC Node
 Long name  : Orange Pi Zero Art-Net 3 www.orangepi-dmx.org
Node started
MIDI
 Direction    : Output
 Active sense : Disabled
 Baudrate     : 31250 (Default)
 Interface    : UART2
mDNS
 Name : allwinner_1F0A53.local
 _apple-midi._udp.local 5004 
 _config._udp.local 10501 
AppleMIDI
 SSRC    : c0a80152 (3232235858)
 Session : allwinner_1F0A53
TCNet
 Node : AvV-OPi
 Using SMTPE fields
 30 FPS
Internal clock
 EBU 25fps  
 Start : 00.00.00:00
 Stop  : 23.29.59:24
Source : Internal
 Max7219 output is disabled
 NTP output is disabled
Display : -1 (0,4)
DisplayWS28xx: 1 Digit(s), 7 LED(S)

```

vanvught, 2019-10-13:
> I got gzip compressed uImage to boot from SPI flash working in U-Boot SPL 2017.03-rc2-00252-gc987c17120-dirty, just by loading it to a higher memory address. I noticed the U-Boot 2018.09 from the SD [...]

This is great! Thanks Hip @hippyau 

I am not sure about `bootm 48000000'
Reading the manual now, it looks like that just `bootm` should work fine.
Maybe I revert back to 0x22000 and use compressed firmware. 

I will attach a new zip with loading the compressed uImage to a higher address.

hippyau, 2019-10-13:
Hey @vanvught, no worries!  As a quick hack, I used a hex editor on the uboot.spi you sent me to change the 40000000 to 48000000 and got that written to SPI, and now it autoboots from SPI.  :)

vanvught, 2019-10-13:
> I am not sure about bootm 48000000' Reading the manual now, it looks like that just bootm` should work fine.

There must be an argument,

hippyau, 2019-10-13:
I figure the bootm triggers the decompression at the loaded location, and the uImage header tells it to decompress and load at 4000000?

vanvught, 2019-10-13:
> I figure the bootm triggers the decompression at the loaded location 48000000, and the uImage header tells it to decompress and load at 4000000?

Yes, that looks the process. 

The load location 0x48000000 is at offset 128Mb in the RAM. So that is fine for the 256MB version.

One disadvantage of moving to compressed image's is is that with a next firmware update, that all  users must also upgrade uboot.spi . However, I've done this before, and I did not see any complains/issues. One *major* advantage although, no worries when adding new functionalities; for example : RDMNet LLRP and Nextion HMI.   I will give it all some thought's. 

Thanks Hip for all this great feedback. Much appreciated!

vanvught, 2019-10-13:
> As a quick hack, I used a hex editor on the uboot.spi you sent me to change the 40000000 to 48000000 and got that written to SPI

Cool! I will leave the uboot.spi as it is. I have #69 for the compressed option.

hippyau, 2019-10-13:
You are so welcome!  Thank you for your cool project!   

> One disadvantage of moving to compressed image's is is that with a next firmware update, that all users must also upgrade uboot.spi

If we move to 128Mb and compress the uImage, it will not have to happen again for a while.

I'm counting on you...  hehehe ;)

[image]

vanvught, 2019-10-20:
Continue working with #69

closed 2019-10-20 by vanvught