vanvught/rpidmx512 #69: U-Boot SPI:  Add support for compressed firmware

vanvught, 2019-10-13:
See conversation in #68 
As this change will imply some _major_ changes, it would need its own tracking.

vanvught, 2019-10-13:
In order to make sure that the user is uploading `gzip` only, there must be a check added here https://github.com/vanvught/rpidmx512/blob/master/lib-remoteconfig/src/ubootheader.cpp#L33 for `ih_comp`

hippyau, 2019-10-13:
Maybe for backwards compatibility, could just check filesize <= max allowed, as u-boot will allow uncompressed or compressed, as the type is described in the uImage header.

hippyau, 2019-10-13:
Rules.mk that compresses target and cleans .gz files

1:24am... Goodnight :)

```
PREFIX ?= arm-none-eabi-

CC	= $(PREFIX)gcc
CPP	= $(PREFIX)g++
AS	= $(CC)
LD	= $(PREFIX)ld
AR	= $(PREFIX)ar
GZIP	= gzip

PLATFORM?=ORANGE_PI
CONSOLE?=

SUFFIX=orangepi_zero

ifeq ($(findstring ORANGE_PI_ONE,$(PLATFORM)),ORANGE_PI_ONE)
	SUFFIX=orangepi_one
endif

COND=
ifeq ($(findstring ORANGE_PI,$(PLATFORM)),ORANGE_PI)
	COND=1
endif

ifeq ($(findstring ENABLE_SPIFLASH,$(DEFINES)),ENABLE_SPIFLASH)
	COND=1
endif

ifeq ($(findstring NO_EMAC,$(DEFINES)),NO_EMAC)
else
	ifdef COND
		LIBS:=remoteconfig tftpdeamon $(LIBS)
	endif
endif

ifeq ($(findstring DISPLAY_UDF,$(DEFINES)),DISPLAY_UDF)
	ifdef COND
		LIBS+=displayudf
	endif
endif

ifdef COND
	LIBS+=artnet4 artnet artnethandlers e131 uuid
endif

ifeq ($(findstring RDMNET_LLRP_ONLY,$(DEFINES)),RDMNET_LLRP_ONLY)
	ifneq ($(findstring uuid,$(LIBS)),uuid)
		LIBS+=uuid
	endif
endif

ifeq ($(findstring DMXSEND,$(DEFINES)),DMXSEND)
	LIBS+=dmxsend dmx
endif

#ifeq ($(findstring PIXEL,$(DEFINES)),PIXEL)
#	LIBS+=ws28xxdmx ws28xx tlc59711dmx tlc59711
#endif

ifdef COND
	LIBS+=ws28xxdmx ws28xx tlc59711dmx tlc59711 spiflashinstall spiflashstore spiflash
endif

LIBS+=network

ifeq ($(findstring ESP8266,$(DEFINES)),ESP8266)
	LIBS+=esp8266
endif

ifeq ($(findstring LTC_READER,$(DEFINES)),LTC_READER)
	DEFINES+=ENABLE_TC1602
endif

# Output 
TARGET = $(SUFFIX).img
LIST = $(SUFFIX).list
MAP = $(SUFFIX).map
BUILD=build_h3/

# Input
SOURCE = ./
FIRMWARE_DIR = ./../h3-firmware-template/
LINKER = $(FIRMWARE_DIR)memmap

LIBS+=lightset ledblink properties display c++ hal bob i2c utils console ff12c h3 debug arm

DEFINES:=$(addprefix -D,$(DEFINES))

ifneq ($(CONSOLE),)
	DEFINES+=-D$(CONSOLE)
endif

# The variable for the firmware include directories
INCDIRS=../include $(wildcard ./include) $(wildcard ./*/include)
INCDIRS:=$(addprefix -I,$(INCDIRS))

# The variable for the libraries include directory
LIBINCDIRS=$(addprefix -I../lib-,$(LIBS))
LIBINCDIRS:=$(addsuffix /include, $(LIBINCDIRS))

# The variables for the ld -L flag
LIBH3=$(addprefix -L../lib-,$(LIBS))
LIBH3:=$(addsuffix /lib_h3, $(LIBH3))

# The variable for the ld -l flag 
LDLIBS:=$(addprefix -l,$(LIBS))

# The variables for the dependency check 
LIBDEP=$(addprefix ../lib-,$(LIBS))
LIBSDEP=$(addsuffix /lib_h3/lib, $(LIBDEP))
LIBSDEP:=$(join $(LIBSDEP), $(LIBS))
LIBSDEP:=$(addsuffix .a, $(LIBSDEP))

COPS=-DBARE_METAL -DH3 -D$(PLATFORM) $(DEFINES)
COPS+=$(INCDIRS) $(LIBINCDIRS) $(addprefix -I,$(EXTRA_INCLUDES))
COPS+=-Wall -Werror -O2 -nostartfiles -nostdinc -nostdlib -ffreestanding -mhard-float  -fno-unwind-tables -fprefetch-loop-arrays
COPS+=-mfpu=neon-vfpv4 -mcpu=cortex-a7 -mfloat-abi=hard

C_OBJECTS=$(foreach sdir,$(SRCDIR),$(patsubst $(sdir)/%.c,$(BUILD)$(sdir)/%.o,$(wildcard $(sdir)/*.c)))
C_OBJECTS+=$(foreach sdir,$(SRCDIR),$(patsubst $(sdir)/%.cpp,$(BUILD)$(sdir)/%.o,$(wildcard $(sdir)/*.cpp)))
ASM_OBJECTS=$(foreach sdir,$(SRCDIR),$(patsubst $(sdir)/%.S,$(BUILD)$(sdir)/%.o,$(wildcard $(sdir)/*.S)))

BUILD_DIRS:=$(addprefix $(BUILD),$(SRCDIR))

OBJECTS:=$(ASM_OBJECTS) $(C_OBJECTS)

define compile-objects
$(BUILD)$1/%.o: $(SOURCE)$1/%.cpp
	$(CPP) -pedantic -fno-exceptions -fno-unwind-tables -fno-rtti -std=c++11 -nostdinc++ $(COPS) -c $$< -o $$@	

$(BUILD)$1/%.o: $(SOURCE)$1/%.c
	$(CC) $(COPS) -c $$< -o $$@
	
$(BUILD)$1/%.o: $(SOURCE)$1/%.S
	$(CC) $(COPS) -D__ASSEMBLY__ -c $$< -o $$@
endef

all : clearlibs builddirs prerequisites $(TARGET)
	
.PHONY: clean builddirs

clearlibs:
	$(MAKE) -f Makefile.H3 clean --directory=../lib-console
	$(MAKE) -f Makefile.H3 clean --directory=../lib-display
	$(MAKE) -f Makefile.H3 clean --directory=../lib-h3
	$(MAKE) -f Makefile.H3 clean --directory=../lib-hal
	$(MAKE) -f Makefile.H3 clean --directory=../lib-remoteconfig

builddirs:
	mkdir -p $(BUILD_DIRS)
	[ -f generate_sofware_version_id.sh ] && chmod u+x generate_sofware_version_id.sh || true

clean:
	rm -rf $(BUILD)
	rm -f $(TARGET)
	rm -f $(MAP)
	rm -f $(LIST)
	rm -f $(SUFFIX).uImage
	rm -f $(SUFFIX).img.gz
	rm -f $(SUFFIX).uImage.gz
	rm -f build0.txt

	for d in $(LIBDEP); \
		do                               \
			$(MAKE) -f Makefile.H3 clean --directory=$$d;       \
		done

#
# Build libraries
#
$(LIBSDEP):
	for d in $(LIBDEP); \
		do                               \
			$(MAKE) -f Makefile.H3 'PLATFORM=$(PLATFORM)' 'MAKE_FLAGS=$(DEFINES)' --directory=$$d;       \
		done

# Build uImage

$(BUILD_DIRS) :
	mkdir -p $(BUILD_DIRS)

$(BUILD)vectors.o : $(FIRMWARE_DIR)/vectors.S
	$(AS) $(COPS) -D__ASSEMBLY__ -c $(FIRMWARE_DIR)/vectors.S -o $(BUILD)vectors.o
	
$(BUILD)main.elf: Makefile.H3 $(LINKER) $(BUILD)vectors.o $(OBJECTS) $(LIBSDEP)
	$(LD) $(BUILD)vectors.o $(OBJECTS) -Map $(MAP) -T $(LINKER) -o $(BUILD)main.elf $(LIBH3) $(LDLIBS)
	$(PREFIX)objdump -D $(BUILD)main.elf | $(PREFIX)c++filt > $(LIST)
	$(PREFIX)size -A $(BUILD)main.elf

$(TARGET) : $(BUILD)main.elf 
	$(PREFIX)objcopy $(BUILD)main.elf -O binary $(TARGET)
	$(GZIP) -n --best -c $(TARGET) > $(TARGET).gz 
	mkimage -n 'http://www.orangepi-dmx.org' -A arm -O u-boot -T standalone -C gzip -a 0x40000000 -d $(TARGET).gz $(SUFFIX).uImage.gz

$(foreach bdir,$(SRCDIR),$(eval $(call compile-objects,$(bdir))))

```

vanvught, 2019-11-06:
- Updated U-Boot SPI is available here -> https://github.com/vanvught/h3dmx512-zip
- All new releases can now use compressed firmware

vanvught, 2019-11-10:
https://github.com/vanvught/rpidmx512/blob/master/lib-spiflashinstall/src/spiflashinstall.cpp#L53

The file uImage.gz must be used.

vanvught, 2019-11-27:
Available in staging branch

closed 2019-11-27 by vanvught

hippyau, 2020-09-08:
Hey Arjan,

>     * Updated U-Boot SPI is available here -> https://github.com/vanvught/h3dmx512-zip

Where does your uboot come from?  do you have available source?

I'm trying to enable the PAL composite TV encoder on the zero in uboot, before the firmware boots,  I have downloaded and compiled a uboot from [here](https://github.com/jernejsk/u-boot/commit/25fa60a9ee1a92d751c0c3a4a8d4acc2495800cc) but it won't boot :(

I see main-line uboot has support for the TV encoder [here](https://github.com/u-boot/u-boot/search?q=tve&unscoped_q=tve) but you have made modifications to the u-boot when we were getting compressed firmware working?

Cheers :)

vanvught, 2020-09-08:
Hi Hippy,

> were getting compressed firmware working?

At the time working with SPI flash, I was not able to get it working with main-line U-Boot. I have used the version linked in here https://linux-sunxi.org/Bootable_SPI_flash => https://github.com/StephanvanSchaik/u-boot/tree/sunxi-spi

Then updating the `CONFIG_BOOTCOMMAND` 

```
pi@nuc-i5:~/uboot-spi/u-boot$ grep CONFIG_BOOTCOMMAND include/configs/sunxi-common.h 
#define CONFIG_BOOTCOMMAND "sf probe; sf read 48000000 180000 22000; bootm 48000000"
```

Also referenced here https://github.com/vanvught/rpidmx512/blob/master/lib-remoteconfig/src/remoteconfig.cpp#L132
and here https://github.com/vanvught/rpidmx512/blob/master/lib-spiflashinstall/src/h3/compressed.cpp#L46

And then updating `orangepi_zero_defconfig` for enabled SPI flash

```
pi@nuc-i5:~/uboot-spi/u-boot$ cat ./configs/orangepi_zero_defconfig
CONFIG_ARM=y
CONFIG_ARCH_SUNXI=y
CONFIG_FIT=y
CONFIG_MACH_SUN8I_H3=y
CONFIG_DRAM_CLK=672
CONFIG_DRAM_ZQ=3881979
CONFIG_DRAM_ODT_EN=y
CONFIG_DEFAULT_DEVICE_TREE="sun8i-h2-plus-orangepi-zero"
# CONFIG_SYS_MALLOC_CLEAR_ON_INIT is not set
CONFIG_CONSOLE_MUX=y
CONFIG_SPL=y
CONFIG_CMD_SF=y
CONFIG_CMD_SPI=y
# CONFIG_CMD_IMLS is not set
# CONFIG_CMD_FLASH is not set
# CONFIG_CMD_FPGA is not set
CONFIG_SUN8I_EMAC=y
CONFIG_USB_EHCI_HCD=y
CONFIG_SPL_SPI_SUNXI=y
CONFIG_DM_SPI_FLASH=y
CONFIG_SPI_FLASH=y
CONFIG_SPI_FLASH_MACRONIX=y
CONFIG_DM_SPI=y
CONFIG_SUNXI_SPI=y

```

Somehow odd that `videomodes.c` is part of the TV encoder and not part of HDMI output => https://github.com/u-boot/u-boot/blob/master/drivers/video/sunxi/Makefile#L6

And http://u-boot.10912.n7.nabble.com/SUNXI-CONFIG-VIDEO-SUNXI-is-never-set-td360711.html

Groet, Arjan

hippyau, 2020-09-09:
Thanks Arjan,  very cool and much appreciated.  

 I did get my mainline to output composite, but it didn't boot the firmware. 

 I'll try compiling yours and mess around for a bit, but is not really a priority just curious :+1: 

Cheers!