vanvught/rpidmx512 #72: LTC SMPTE: Add support for "ltc!rate#xx"

vanvught, 2019-10-22:
This is a follow-up for #71 
There are several commit conflicts to resolve. I hope to get in sync with @hippyau soon.

vanvught, 2019-10-23:
- [ ] I need to do some rendering here

[image]

hippyau, 2019-10-24:
Hey Arjan,
   Is that a typo for drop frame,  29.97 not 24.99?
Cheers,
 Hip

vanvught, 2019-10-24:
Hi Hip, yeah that was a typo. I will commit a new version this evening. 
Greetings, Arjan

hippyau, 2019-10-24:
;-D 

thanks for working on this, I'm pretty new to C++ and really liking it.   My git skills are amateur, I am in some kind of hell where it seems easiest for me to ditch my for and re-fork now, because you have made awesome changes and I can't work out how to merge them....  All good!   I still have work to do on the WS28xx display, and i'll try to keep my changes limited to there for a while until :)  

Thanks for the new parameters, I stuggled to understand the storing of settings, and knew I had not disabled MAX7219 very well.   I had considered a transistor to 'chip select' for the ws28xx leds, but not bothered with both max7129 and ws28xx at the same time.    I'll do more of a write-up for makers who wan't to build their own. :)

Thanks again!

vanvught, 2019-10-24:
Re C++, you might have seen that I've changed the calling for https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/ltc.cpp#L195

The `nFPS` parameter is now called by reference -> https://www.ntu.edu.sg/home/ehchua/programming/cpp/cp4_PointerReference.html

vanvught, 2019-10-24:
>  I'll do more of a write-up for makers who wan't to build their own. :)

I have given you full editor acces for my website. I have also created a sub-page -> https://sites.google.com/site/rpidmx512/orange-pi-smpte-timecode-ltc-reader-converter/ws28xx-7-segment

I am happy when you do the write-up there ;-)

vanvught, 2019-10-31:
There is an issue reported by Chip. The rate change is working for the display's, but not for other outputs. 

I need to set the nType also for m_pStartLtcTimeCode and m_pStopLtcTimeCode
-> https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/h3/ltcgenerator.cpp#L311

vanvught, 2019-11-10:
https://github.com/vanvught/h3dmx512-zip/commit/a368c6936cb7e3526afd1230c32080b55ee57601

closed 2019-11-10 by vanvught