vanvught/rpidmx512 #76: ShowSysTime() is Time Of Day?

hippyau, 2019-11-04:
Hi Arjan,

  Is ShowSysTime() the current Time Of Day?  ie. the time right now?
  Where does that come from?

  Maybe easy to run a NTP client for ToD, as an source?   
 
  Only need server and local GMT offset?

Cheers,
  Hippy

vanvught, 2019-11-04:
Hi Hippy,

Currently the ShowSysTime is: or Uptime, or RTC time (when mcp7941x is attached), or the time set by ArtTimeSync (which is also then written to RTC). This ArtTimeSync is enabled by default.
http://www.orangepi-dmx.org/orange-pi-smpte-timecode-ltc-reader-converter/configuration

For testing the embedded NTP Server, I have written some NTP Client code. So implementing a NTP Client would be 'easy'. We need to add a property in ltc.txt -> for example: enable_ntp_client
Just run it at startup? Or when ShowSysTime() is called?

Then Timezone's ;) Let's just go for GMT offset. 

Groeten, Arjan

hippyau, 2019-11-04:
> or RTC time (when mcp7941x is attached)

of course, I don't have one yet :) :+1: 

Adding ToD as a source for TC output could be useful in some circumstances, like as a master clock at a theme park or architectural installation.  

An NTP Client to update ToD which connect to an internet NTP server would be useful (just specify IP address to NTP server, so no DNS lookup required).   

Then sync ToD with NTP server every 15 minutes maybe?

vanvught, 2019-11-05:
> An NTP Client to update ToD which connect to an internet NTP server would be useful (just specify IP address to NTP server, so no DNS lookup required).

Hi Hip, please see this "Current limitation Ethernet versions: All nodes must be in the same network" (https://github.com/vanvught/h3dmx512-zip).

Currently I do not have a gateway implementation. This is because my assumption that light shows are never connected to the Internet. 

So for the time being, there must be a local NTP Server. And this local NTP Server can have Internet access for updating using public NTP Server's.

Greetings, Arjan

vanvught, 2019-11-06:
> Adding ToD as a source for TC output 

Hi Hip, adding ToD as a source would be a new enhancement item.

Current plan:

1. Deliver V2.1 -> #72 (this weekend)
1. Merge #75 (this weekend)
1. Add NTP Client
1. Do some testing 
1. Deliver V2.2 (this would include that users must upgrade U-Boot SPI as well for supporting compressed firmware)

Groet, Arjan

vanvught, 2019-11-10:
> Maybe easy to run a NTP client for ToD, as an source?

Relatively easy ;-) continuing in #78

closed 2019-11-10 by vanvught