vanvught/rpidmx512 #79: RemoteConfigUI: Add support for LTC SMPTE Display WS28xx UDP messages

vanvught, 2019-11-09:
cc @hippyau

vanvught, 2019-11-09:
[image]

cc @hippyau

vanvught, 2019-11-09:
[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/3827679/RemoteConfigUI.jar.zip)

cc @hippyau

hippyau, 2019-11-10:
Hi Arjan!

 Thanks for this!  Unfortunately, it's not working because it seems like the UI is still sending to UDP port 21571 not the new port of 10258.   The device responds to echo command lines properly.

[image]

Cheers, 
 hip

vanvught, 2019-11-10:
Hi Hip,

That's a bummer. My lesson's learned; never work off-line again - do always a proper test ;-)

Greetings, Arjan

[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/3828183/RemoteConfigUI.jar.zip)

hippyau, 2019-11-10:
Wow, thanks for the quick fix!  Now works beautifully.  :)

hippyau, 2019-11-10:
Hi Arjan, 
   Thanks so much, this one works great!
Cheers, hip

hippyau, 2019-11-10:
Thanks so much, is all good now!

vanvught, 2019-11-10:
Available in staging branch -> https://github.com/vanvught/rpidmx512/commit/72d007d264d2e013714a70c6fc6166c3192019f7

closed 2019-11-10 by vanvught