vanvught/rpidmx512 #82: LTC SMTPE: Add support for remote source select

hippyau, 2019-11-10:
I'm wanting to implement a remote source selection.  I'm going to work off your rotary encoder / button source selection codes,  but wonder where you would like the UDP message to be based?  

It feels like it should be:
`
ltc!source=rtp-midi
ltc!source=midi
ltc!source=internal
[etc]...
`
?

vanvught, 2019-11-10:
Hi Hippy,

Changing the source at run-time is almost impossible; it would need a complete rewrite of all the Reader Classes. 

The Rotary/Button implementation gives you the opportunity to set the source. But all times, it needs a reboot in order to make the selection active. 

The same, reboot,  would be needed when we implement `ltc!source=`

hippyau, 2019-11-10:
Ah, yeah I figured it would be tricky.
Since it needs a reboot anyway, the functionality is already implemented
with existing UDP get,set commands.
Closed.

closed 2019-11-10 by hippyau