vanvught/rpidmx512 #85: Investigate 'soft-reset'

vanvught, 2019-12-11:
With respect to LTC run-time source select, remark by @hippyau in #84
> I do wonder, can we 'soft-reset' by jmp to not_main()? that might speed up source change?

cc:  @rbarreiros

vanvught, 2019-12-29:
Available in branch staging and updated firmware is on Google Drive

cc: @hippyau @rbarreiros @brunchboy @drummerclint

vanvught, 2020-05-26:
The soft-reset does not work for all cases. For the time being the SoftReset won't be used, the Reboot is used instead.

cc: @hippyau @rbarreiros @brunchboy @drummerclint

hippyau, 2020-06-05:
Sorry, I hoped it would make source changes faster.  Not a big issue though, can probably close.

vanvught, 2020-06-05:
The tricky part is that:

- All C++ Destructors should be called before soft-reset
- I need to re-initiate the not 0 assigned variables which are in the .data section.  I can do this by copying the .data section from the spi flash.

hippyau, 2021-07-03:
I am just dreaming out loud here....   

We could have a goodbye flag, global signal ```SHUTDOWN```;  when ```SHUTDOWN``` goes true, a 200ms hold off timer is started in main loop, to allow cleanup of any connections, and any ```Run()```'s could handle this ```SHUTDOWN``` flag and do their disconnect protocol / clean up if required within 200ms.

When the hold off timer elapses, notmain() exits, or is called recursively, or is ```jmp``` to but then immediately starts again... or something...

Is this the boot loader starting again or is some assembly required to pull this off?

Or would it work if we just call notmain() recursively from notmain()?  ... and all new memory is allocated and old memory is 'lost'... This is pretty crap, but we have 256/512MB RAM, we could get away with a few times, then force a reboot?

Low priority

vanvught, 2022-01-10:
Quick reboot is available with MCU implementations:

- https://github.com/vanvught/GD32F103RC-DMX512-RDM
- https://github.com/vanvught/GD32F107RC-DMX512-RDM
- https://github.com/vanvught/GD32F207RG-DMX512-RDM
- https://github.com/vanvught/GD32207C-EVAL-board-DMX512-RDM

closed 2022-01-10 by vanvught