vanvught/rpidmx512 #89: LTC Generator: Implement pitch control

vanvught, 2019-12-19:
Follow-up for #86 

Received request by e-mail from deepsymmetry.org

Pitch values in % :

-  -100 -> this stops the incrementing/decrementing 
- 0 -> is normal
- +100 -> twice the speed

This is similar to the Pioneer CDJ WIDE pitch control.

- [ ]   Create bool LtcGenerator::PitchControl(void)
- [ ]   Create new UDP commands: 'ltc!pitch#x' where -100 <= x <= 100 
- [ ]   Create new OSC commands: /tc/pitch -> Int type 
- [ ]   Update website
- [ ]   Update sample Java UI

cc: @hippyau @rbarreiros

brunchboy, 2019-12-19:
Thank you! I can work with that, although I find it more natural for 0 to mean stopped (that is how I implemented this in my Max/MSP solution, and how Pioneer’s gear report their pitch when the DJ holds down the jog wheel). I can translate that into your separate commands, though, that is a simple matter of putting more smarts in my own software.

However, if the DJ puts their player into Wide pitch range, the pitch values I will have to work with go well beyond ±100%. I can’t see any way to work around that here. Is there any possibility of supporting the full integer range for pitch? Otherwise I can document this as a limitation of using rpidmx512 instead of the Max/MSP patch.

Once there is a way for Clint to get trial builds of the firmware, I can ask him to start testing them as I help him write the expressions that will control it.

brunchboy, 2019-12-19:
Thank you, I can work with that! However, is there any way to extend the range and resolution? Pioneer gear reports their pitch in units of 1/100 of a percent, and the DJ can set their pitch range to Wide, which goes well beyond double speed. Unless I can convey those to rpidmx512, the visuals are going to lose synchronization with the music. The fractions of a percent are more important than the wider range, because they are going to happen far more often, but I can’t really recommend this to my users as a viable solution without both.

brunchboy, 2019-12-19:
Once there is a way for Clint to get trial builds of the firmware, I can ask him to start testing them as I help him write the expressions that will control it.

brunchboy, 2019-12-19:
Here is the exact formula I use for determining pitch from the network packets sent by the players, from page 9 of the [packet analysis]():

The player’s current pitch adjustment4 can be found in bytes `54`– `57`, labeled _Pitch_. It represents a three-byte pitch adjustment percentage, where `0x00100000` represents no adjustment (0%), `0x00000000` represents slowing all the way to a complete stop (−100%, reachable only in Wide tempo mode), and `0x00200000` represents playing at double speed (+100%).

vanvught, 2019-12-19:
> Pioneer gear reports their pitch in units of 1/100 of a percent.

It might be reported in units of 1/100 of a percent. However, I've found the following details:

```
For CDJ-2000s/1000s/900s/850s/400s/350s/200s:

+/- 6% : 0.02%
+/- 10%: 0.05%
+/- 16%: 0.05%
+/- 100% (WIDE) : 0.5% 

CDJ-800:

+/- 10% : 0.05%
+/- 100% (WIDE) : 0.5% for cds, 0.1% for mp3
```

brunchboy, 2019-12-19:
Sorry about the flurry of incorrect versions of this, I should have looked at my packet analysis before responding. I have edited them to be correct now. The link now works as well.

brunchboy, 2019-12-19:
The displays on the players update in units of 0.5%. But the music playback (and the pitch adjustment reported on the network) are actually adjusted in finer increments than that, using the formula reported in my packet analysis, linked above.

vanvught, 2019-12-19:
> However, is there any way to improve the pitch resolution?

There are no limitations. I can implement all what is doable with TimeCode conversion.

brunchboy, 2019-12-19:
Thanks, I apologize again for posting incorrect responses earlier, I was trying to sneak in a quick reaction while at work (and on not enough sleep from last night) because I am so excited about this.

vanvught, 2019-12-19:
- There is 1% pitch accuracy for UDP messages
- There is 0.01 % pitch accuracy for OSC messages

UDP messages with Java example UI:
[image]

vanvught, 2019-12-21:
OSC examples:

```
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/start
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/pitch f 1.0
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/pitch f 0.0
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/pitch f -1.0
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/pitch f 0.0001
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/stop
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/start
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/pitch f 0.00001
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/pitch f 0.000001
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/pitch f 0.5
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/pitch f 0.00005
@nuc-i5:~$ oscsend  192.168.2.252 8000 /allwinner_26038E/tc/pitch f 0.1
```

There is no limitation, except for what is possible with Time Code frame resolution.

cc: @brunchboy  @drummerclint

brunchboy, 2019-12-22:
That looks excellent! @drummerclint, when do you think you will have time to experiment with this branch?

vanvught, 2019-12-22:
@brunchboy   @drummerclint

Latest build from staging -> https://drive.google.com/open?id=11c89W2HXd_q7canKeprAKziMrZXUYqn9

drummerclint, 2019-12-22:
Thanks @vanvught!  
I just realized a couple things and learned why some things didn't work.  I was wondering for a while why every time I tried editing a file with the java utility, the edits never retained.  It was because I left the SD card in the pi.  I'm not sure if I missed that in your documentation, but I think you should mention that.  I can make you a list of things I've figured out along the way while building this.  To be honest I had a great time and loved the challenge!  It's been a while since I got to dive this deep into electronics.  Once the midi board comes in, I'll be able to finish the enclosure.  I'll send you photos when it's done.  

I noticed the LTC generator window is missing the pitch control aspect.  I've resized the window even.  I've tried each of the versions you've posted.  

@brunchboy, I've been playing around with the build the last few days, I've had last night and next Saturday off from the club so haven't gotten to try it yet.  I might bring a CDJ home to play with however.  All the gear will be within arms reach in that case!  ahaha

vanvught, 2019-12-22:
[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/3992946/RemoteConfigUI.jar.zip)

drummerclint, 2019-12-22:
Thanks!  Excited to try it with Resolume!

-Clint

vanvught, 2019-12-22:
>  I can make you a list of things I've figured out along the way while building this. 

Yes, please. I really need to invest time in updating my website.

vanvught, 2019-12-22:
>  Once the midi board comes in,

You have designed your own? I am about to release a prototype board with MIDI In/Out/Thru and balanced LTC In/Out

drummerclint, 2019-12-22:
I ended up ordering this MIDI shield: https://www.aliexpress.com/item/4000039505908.html but I think they shipped it via sea :(   I was trying to figure out the best way to interface the LTC audio, I might do unbalanced 1/4" or RCA.  I was trying to find a way to buffer the output, but didn't spend a lot of time and couldn't find a schematic on the opi expansion board to find out the capacitor values.  I'd love to try out your prototype however.  LTC is a big thing, especially when using Resolume.

I'll start compiling a list for your website and I'll send it over.  Do you want it as an email or just a git issue?

-Clint

vanvught, 2019-12-23:
Hi Clint,

>  this MIDI shield: https://www.aliexpress.com/item/4000039505908.html

Please bear in mind that this shield is designed for use with a 5Volt Arduino. The Orange Pi Zero is 3V3 logic. You would need to protect the UART input. For example with zener/resistor -> http://www.orangepi-dmx.org/orange-pi-smpte-timecode-ltc-reader-converter/hardware
Also, the 3V3 UART output '1' should be high enough for driving the MIDI Out circuit.

Talking about the circuit, looking at the picture, the MIDI board does not have 74xx buffer drivers. Hence it does not adhere to the specification at all. When they are driving the MIDI Out directly from the GPIO, then it is pretty bad design.

- Arjan

vanvught, 2019-12-23:
> I'll start compiling a list for your website and I'll send it over. Do you want it as an email or just a git issue?

Please open a git issue. Then it is easier for everyone to check what should be done.

Thanks, Arjan

hippyau, 2019-12-23:
Hi @brunchboy @drummerclint, merry XMAS.

@brunchboy thanks so much for your work on the CDJ protocol... Synchronicity, turns out we we both were decoding the ProDJ Link at the same time... when I stumbled upon Dysentery it was great, I could confirm my work was right!  Still you had gone further by the time I'd caught up, your white paper is a great resource.  

Cheers,
 Hip

hippyau, 2019-12-23:
>  I am about to release a prototype board with MIDI In/Out/Thru and balanced LTC In/Out. The LTC out is just a buffer for the OPi Zero line-out.

Are you ordering?  I'm in, how much?  I'll pay for your time/board.

vanvught, 2019-12-23:
> I am about to release a prototype board with MIDI In/Out/Thru and balanced LTC In/Out. The LTC out is just a buffer for the OPi Zero line-out.

@brunchboy @drummerclint @hippyau 

This is a previous prototype. There was a design error ;) There are 3 boards:

1. OPI Zero add-on with the headers
2. The board with LTC balanced in/out, with protection (for the sound engineers who would to enable phantom power ;-) and the MIDI circuit.
3. The MIDI connectors board

I will make the Eagle files available on my GitHub and the BOM on my website.

[image]

hippyau, 2019-12-23:
>     2\. for the sound engineers who would to enable phantom power ;-)

Been there, done that :) :) :) :+1: 

Thanks Arjan,  are you building any?  JLC PCB have the parts? They have a manufacture service now?

Cheers,
 Hip

vanvught, 2019-12-23:
> JLC PCB have the parts? They have a manufacture service now?

I have new boards from JLC PCB. I hope to have some time in the holidays to make one and give it a good test. I have all through-hole parts, so that isn't for the manufacture service.

Groet, Arjan

drummerclint, 2019-12-23:
> Hi Clint,
>
>
> Please bear in mind that this shield is designed for use with a 5Volt Arduino. The Orange Pi Zero is 3V3 logic. You would need to protect the UART input. For example with zener/resistor [...]

Oh boy.  ahaha I just ordered two more from a different supplier but same design I believe.  I've just cancelled the order actually.   My plan was to interface it to the SV7 connector on your board.  

I'm looking forward to seeing your prototype, I'll probably send it off to JLC as soon as I can as well.  I think minimum build qty is 5 which is fine.  That was one thing I noticed when building your board for the opi, I couldn't find a BOM and it was hard to figure out what I needed and where it had to go.  I ended up going through the eagle files and figured out that way as for example, the 3v3 D1 and 330 R5 are right over top of each other.  

Buffered I/O sounds like a dream!  I was going to run it through a DI box if I needed to run LTC over a long distance or put a balancing transformer in the enclosure.

@hippyau Merry Christmas Hippy!  Great work on TimeLord!  I ended up pitching in when you did the Pozible campaign.  This is around the same time I asked @brunchboy about Beat Link Trigger.

I've tested the pitch shifting into Resolume via LTC, into Chamsys via art-net tc and RTP MIDI into the MT app on my iphone/ipad (although the iphone and ipad will only run exclusively of each other)

Merry Christmas guys!

-Clint

brunchboy, 2019-12-23:
Hello @hippyau and merry xmas to you as well! It is always nice to hear from others who are exploring this space, and it is cool to see so many interesting solutions and projects.

vanvught, 2019-12-29:
https://github.com/vanvught/h3dmx512-zip

closed 2019-12-29 by vanvught

drummerclint, 2020-01-01:
Reported working very well with James' Beat Link Trigger! 
Great work @vanvught and @brunchboy!!

I've got a CDJ with me this week.  I'm going to see if I can do an instructional video.

brunchboy, 2020-01-01:
That would be great, @drummerclint! During my swim today I came up with a simple and efficient way to add reverse play support to the sample show I sent you. I will code that up after dinner in the hopes that you can test it for me, and then it will be the basis of the example I add to the user guide.

Handling multiple players playing the track is going to be much trickier, mostly because it is going to completely depend on the use case of the show producer what they want to do. So I think I am going to leave that unimplemented until someone has an actual approach they want to try. I did think of a really cool maximal approach, where if you were using a DJM or Allen & Heath mixer, you could hook a USB cable to the mixer and turn on MIDI mode and then do something like I did for simulating channels-on-air. You would need a way to send multiple LTC streams (maybe multiple Orange Pis), but you could overlay multiple video streams, and then by tracking the positions of the mixer channel faders and cross fader (and possibly filter knobs), could adjust the transparency of each video layer based on how audible it was. An extreme case, but cool integration proof-of-concept! 😄