vanvught/rpidmx512 #93: BCM2835 SPI Slave

ropg, 2019-12-22:
Sorry to be opening issue here, but this is linked from [here](https://groups.google.com/forum/#!topic/bcm2835/chJfQQWXFMo) and I was wondering if you know is anyone ever got SPI slave to work on BCM2835 a.k.a. Raspberry Pi Zero. Googling seems to show lots of people tried with no luck, figured I'd ask if you got any further....

vanvught, 2019-12-22:
SPI Slave on the Broadcom SoC does not seem to work. I have not looked around anymore.

Note, that I have moved away from the Raspberry Pi boards. The next supported board is the Orange Pi Zero:

- Far better Ethernet implementation.
- Can boot from SPI flash; the SDCard is just needed once.
- With the SPI flash, all firmware configuration items are stored in SPI flash.
- Much cheaper.
- Easier for software maintenance
- GPIO header is compatible with the Raspberry Pi header; RPi hardware can be used

closed 2019-12-22 by vanvught

rewolff, 2019-12-22:
Hi Rop, 
Haven't heard from you in a while! 
I was going to write that the peripherals in the PI only do "master" but there is a separate peripheral that claims to be able to do SPI and I2C slave stuff. Not sure where the pins are mapped (hint: probably not the standard pins for SPI... OK looked it up GPIO18-21.). 

I would expect the peripheral to work better than mentioned in the start of the thread you referenced. Probably some problem initializing the peripheral.

Broadcom has refused to fix a hardware bug in the successors for the  BCM2835 that they have known about since 2013. So maybe a peripheral like this not working might also be delegated to the "doesn't work, but nobody uses it so ignore it" pile. On the other hand it is documented so it should work at least a bit.

vanvught, 2019-12-22:
The I2C does do slave mode -> https://github.com/rsta2/circle/blob/master/lib/i2cslave.cpp

The I2C slave in the Allwinner H2+/H3 is well documented.