vanvught/rpidmx512 #94: LCD 16x2 i2c display not displaying

drummerclint, 2019-12-22:
Hey Arjan,

I've got a 128x64 OLED display which works like a charm but if I try to use the 16x2 display instead, it just shows a blank screen.  I've tested the display with an arduino and set the contrast and that worked just fine.  I've connected only this 1602A display to the opi with no result.  The arduino detects the display at address 0x27 and it uses the PCF 8574T i2c controller.  Any ideas as to what I could do to diagnose the issue? 

Thanks,
Clint

vanvught, 2019-12-22:
Hi Clint,

Have you connected the VCC to +5V ? 

When you have an USB/UART cable, then you can check with the debug UART that the output is showing `Display : 2 (16,4)`

Type 2 => DISPLAY_PCF8574T_1602 -> https://github.com/vanvught/rpidmx512/blob/master/lib-display/include/display.h#L42

-Arjan

vanvught, 2019-12-22:
[image]

drummerclint, 2019-12-22:
Hey Arjan,
The display lights up but it's that's it.  I'll test the debug shortly.

Thanks,
Clint

vanvught, 2019-12-22:
3V3 ->

[image]

5V ->

[image]

drummerclint, 2019-12-22:
It's amazing how things start working when you give them the correct voltage... 
Thanks Arjan!
-Clint

closed 2019-12-22 by drummerclint