vanvught/rpidmx512 #95: LTC SMTPE Minor Missing Details

drummerclint, 2019-12-23:
Hey Arjan, 

Just a small list of things I've learned along the way of building this project and some suggestions as well, in no particular order.

- BOM for board construction
- Clearer labeling for component locations on board layout, maybe label pin 1 on connectors
- Purpose of SV4 - (system status?)
- MCP23017 for i2c source selection buttons require Int pin 12 to be connected to ITA
- LTC output is derived from opi expansion pin 8/9 (l/r)
- Verify your voltage for the 16x2 i2c lcd display, mine was 5v, where the i2c headers on the board are 3.3v
- When using the Java RemoteConfigUI app, the data on the SD card takes precedence so any changes you make will get overwritten by the data on the card if it exists.  You only really need the SD card for the initial image.  (took a while for the SPI flash boot process diagram to actually click ahaha)
- Do you recommend the 512MB or 256 MB version of the opi zero?
- What are your thoughts on the LTS version of the opi zero?

Thanks so much! 
-Clint

vanvught, 2019-12-24:
Questions:

- Do you recommend the 512MB or 256 MB version of the opi zero?
-> The 256MB is the recommended version. 
- What are your thoughts on the LTS version of the opi zero?
-> The LTS is the 'Long Term Support' version which is perfect for my firmware 
- If we set source to systime, is it possible to make it start automatically on bootup?
-> I will open an enhancement request for this option.
- Is it possible to add support for other RTC i2c modules such as the DS3231? I'm having a hard time finding the mcp7941x rtc outside of bitwizard, would something like https://www.aliexpress.com/item/33048577846.html work, being based off of DS3231?
-> I will open an enhancement request for supporting this DS3231 populair RTC.

vanvught, 2019-12-24:
Action list website:

- [ ] BOM for board construction
- [ ] Clearer labeling for component locations on board layout, maybe label pin 1 on connectors
- [ ] Purpose of SV4
- [ ] MCP23017 for i2c source selection buttons require Int pin 12 to be connected to ITA
- [ ] LTC output is derived from opi expansion pin 8/9 (l/r)
- [ ] Verify the voltage for your 16x2 i2c lcd display, mine was 5v, where the i2c headers on the board are 3.3v
--> the header is designed for the OLED display and the 7-segment status display. 
- [ ] When using the Java RemoteConfigUI app, the data on the SD card takes precedence so any changes you make will get overwritten by the data on the card if it exists. You only really need the SD card for the initial image.
--> this is not sample Java UI specific. It is about the UDP messages

drummerclint, 2019-12-26:
>   --> the header is designed for the OLED display and the 7-segment status display.
-- That makes a lot of sense :)
> * [ ]  When using the Java RemoteConfigUI app, the data on the SD card takes precedence so any changes you make will get overwritten by the data on the card if it exists. You only really need the SD card for the initial image.
>   --> this is not sample Java UI specific. It is about the UDP messages
-- That's true, but I haven't played with udp messages yet.

Thanks for your hard work!  Merry Christmas!

vanvught, 2019-12-29:
BOM added. Also description for each connector -> http://www.orangepi-dmx.org/orange-pi-smpte-timecode-ltc-reader-converter/hardware
MCP23017 INTA -> http://www.orangepi-dmx.org/orange-pi-smpte-timecode-ltc-reader-converter/buttons

vanvught, 2019-12-29:
->Clearer labeling for component locations on board layout, maybe label pin 1 on connectors
This will be checked for the next prototype board.

closed 2019-12-29 by vanvught