vanvught/rpidmx512 #97: lib-ltc: If we set source to systime, to make it possible to start automatically on bootup

vanvught, 2019-12-24:
Follow-up for #95

cc: @hippyau @rbarreiros @brunchboy @drummerclint

vanvught, 2019-12-26:
@hippyau @brunchboy @drummerclint
https://drive.google.com/open?id=11c89W2HXd_q7canKeprAKziMrZXUYqn9

drummerclint, 2019-12-27:
Thanks Arjan!

Parameter works great!  Although, I have the source selection buttons connected and the system waits for the user to select the source then automatically starts.  Can we allow autostart to bypass that source selection screen at bootup.  This will lead to the issue of not being able to select a source from the front panel.  Is it possible to select the source before the reboot, store the selection, reboot and then go into the selection.  

-Clint

vanvught, 2019-12-27:
Hi Clint,

>  Is it possible to select the source before the reboot, store the selection, reboot and then go into the selection.

That does not fit in my reference architecture.  The SPI flash store's are at run-time only available with the UDP messages for the remote configuration framework.

What I can look at is, for example, having 2 button's pressed on boot-up for bypassing the auto-start=1. This can be an option for a next release.

Please check my Google Drive for  commit f779a8b

-Arjan

drummerclint, 2019-12-27:
That looks great Arjan!
I'm thinking from a show perspective where an opi would be on stage and it rebooted for some reason, it could continue from right where it left off.

drummerclint, 2019-12-27:
[image]

Here's my build.  I had to mount the headers on angles to clear the height of the enclosure they're going in.

brunchboy, 2019-12-28:
I am working on the code to send OSC commands for @drummerclint and have a question about the `start/hh/mm/ss/ff` message. Will this start the timecode as well as setting the current position if it was not already running? Or do I need to send a separate `start` message as well?

vanvught, 2019-12-29:
https://github.com/vanvught/h3dmx512-zip

vanvught, 2019-12-29:
> I am working on the code to send OSC commands for @drummerclint and have a question about the start/hh/mm/ss/ff message. Will this start the timecode as well as setting the current position if it was not already running? Or do I need to send a separate start message as well?

http://www.orangepi-dmx.org/orange-pi-smpte-timecode-ltc-reader-converter/internal-generator/run-time-osc-messages

```
start/set/hh/mm/ss/ff    Sets the start TimeCode. Activated after a Stop / Start.
start/hh/mm/ss/ff          Set the running TimeCode
goto/hh/mm/ss/ff          Set and jump to the goto TimeCode, but stop the TimeCode incrementing/decrementing.
```

cc:  @drummerclint

closed 2019-12-29 by vanvught

brunchboy, 2019-12-29:
Yes, @vanvught that is the section of the documentation which is ambiguous. It does not state what happens when you send `start/hh/mm/ss/ff` when the timecode is not already running. Does it start it? Or does it remain stopped? It would be very helpful to me if, in addition to setting it, it makes sure that it is running.

vanvught, 2019-12-29:
> Does it start it?

Yes, it does. It is about quickly going to another section in the show time-line. And continue outputting LTC SMPTE.

https://github.com/vanvught/rpidmx512/blob/master/opi_emac_ltc_smpte/lib/oscserver.cpp#L156

```
	LtcGenerator::Get()->ActionSetStart((const char *)&m_pBuffer[nOffset]);
	LtcGenerator::Get()->ActionStop();
	LtcGenerator::Get()->ActionStart();
```

brunchboy, 2019-12-29:
Thanks for the confirmation! I haven’t actively written C++ code since the early 1990s, so trying to read your source is not a good use of my time available for helping Clint.