vanvught/rpidmx512 #100: Midi schematic

drummerclint, 2020-01-09:
Happy new year Arjan!

DS3231 RTC clock appears to be working nicely!  Thank you!

Was wondering if you can share the schematic for your MIDI / balanced audio board yet.  At the moment I just need the midi part as I wouldn't mind building it into this particular version.

re: interfacing with Jame's BLT, it works quite nicely, I'm going to film a quick video demonstrating it, controlling Resolume via a LTC, via a CDJ.

Thanks in advance,
Clint

vanvught, 2020-01-12:
Hi Clint, happy new year!

As I was working on finishing the WS28xx display support and implementing TCNet specification V3.3.3, I did not have the time working on the hardware. 

The prebuild LTC SMPTE V2.4 is here -> [https://drive.google.com/open?id=169nmTLxNIRca2DcmwZ_8m6YfLBVkxkkM](https://drive.google.com/open?id=169nmTLxNIRca2DcmwZ_8m6YfLBVkxkkM)

I have created a single issue #102 for all the website updates. Therefore closing this one.

Groet, Arjan

closed 2020-01-12 by vanvught

drummerclint, 2020-01-22:
That's great! Thanks Arjan!

hippyau, 2020-06-05:
Hey Arjan,
   Is there a board that has both MIDI and LTC connections so that one can take advantage of LTC>MIDI or MIDI>LTC or ARTNET>MIDI&LTC etc :)

vanvught, 2020-06-05:
Hi Hippy, yes there is. See also the item "Add schematic, PCB and BOM for the next LTC SMPTE"  in #102 :)
[image]

[image]

[image]

hippyau, 2020-06-05:
Arjan... This is epic!!

Thanks man,   I'm looking now.

drummerclint, 2020-06-06:
Arjan, is that the ltc board with the balanced io that you mentioned a while back? :o 
@hippyau little off topic but I wanted to ask you about msexlord, what's the best way to contact you?

vanvught, 2020-06-06:
Hi Clint, 

> Arjan, is that the ltc board with the balanced io that you mentioned a while back? ⭕ 

Yes, still but still empty. However another user has made this board and successfully tested with input. 
And, quoting this user:

> I also managed to get the artnet TC really easy into Onyx :)
This looks so nice, thanks!

drummerclint, 2020-06-06:
That's exciting!  I'm looking forward to building several.  I have some industry friends really interested in the project but don't have the electronics background to build so I may end up building a few extras for them.  I ended up writing a basic guide on getting started with just ltc output from James' beat link trigger project.  You can check it out [here](https://github.com/Deep-Symmetry/beat-link-trigger/wiki/Guide:-Building-the-Orange-Pi-LTC-SMPTE-Timecode-Reader,-Converter-&-Generator ) if you're interested.  Please let me know if you'd like me to make any changes to it.  I've been meaning to record a demo video for the longest time.  I'll let you know when that happens.

drummerclint, 2020-06-15:
Hi Arjan, do you have the BOM for the next LTC board? 
Thanks!
Clint