vanvught/rpidmx512 #103: DMX to ArtNet

mrv96, 2020-02-08:
Hi, is with your project possible to have a DMX input and an ArtNet output?

Thank you very much for your work!

vanvught, 2020-02-08:
Yes, it is possible to have DMX input and an ArtNet/sACN output.

- https://github.com/vanvught/rpidmx512/blob/master/lib-artnet/include/artnetparams.h#L62
- https://github.com/vanvught/rpidmx512/blob/master/lib-artnet/src/artnetparamsconst.cpp#L49
- https://github.com/vanvught/rpidmx512/blob/master/lib-artnet/src/artnetnodehandledmxin.cpp
- https://github.com/vanvught/rpidmx512/blob/master/lib-e131/include/e131params.h#L54
- https://github.com/vanvught/rpidmx512/blob/master/lib-e131/src/e131paramsconst.cpp#L36
- https://github.com/vanvught/rpidmx512/blob/master/lib-e131/src/e131bridgehandledmxin.cpp

But it is not yet implemented in the DMX Art-Net/sACN single port firmware.

I will open an enhancement request.

vanvught, 2020-02-08:
Enhancement request created #104

closed 2020-02-08 by vanvught

mrv96, 2020-02-08:
Thank you very much for your fast reply.

Here: http://www.orangepi-dmx.org/raspberry-pi-art-net-dmx-out/configuration i'm not able to see how to configure DMX input ports for the code version with two ports (OPI Zero)

vanvught, 2020-02-08:
> Here: http://www.orangepi-dmx.org/raspberry-pi-art-net-dmx-out/configuration i'm not able to see how to configure DMX input ports for the code version with two ports (OPI).
> 
> Which is the parameter to set?

See ToDo list -> #104 
Please note that #104 is initially for the single port DMX firmware.

mrv96, 2020-02-08:
Ok. Thank!