vanvught/rpidmx512 #104: DMX to Art-Net/sACN

vanvught, 2020-02-08:
This is a follow-up for #103 

- [ ] lib-artnet -> https://github.com/vanvught/rpidmx512/blob/master/lib-artnet/src/artnetparams.cpp
- [ ] lib-artnet -> https://github.com/vanvught/rpidmx512/blob/master/lib-artnet/src/artnetparamssave.cpp
- [ ] lib-e131 -> https://github.com/vanvught/rpidmx512/blob/master/lib-e131/src/e131paramssave.cpp
- [ ] Art-Net -> https://github.com/vanvught/rpidmx512/tree/master/opi_emac_artnet_dmx
- [ ] sACN -> https://github.com/vanvught/rpidmx512/tree/master/opi_emac_e131_dmx
- [ ] Update website

cc: @mrv96

vanvught, 2020-02-09:
Hi @mrv96, please find attached the sACN E1.31 firmware for DMX Input. I hope you can give it a test. Many thanks in advance, Arjan

Set `direction=input` in `e131.txt`

[opi_emac_e131_dmx.zip](https://github.com/vanvught/rpidmx512/files/4176869/opi_emac_e131_dmx.zip)

mrv96, 2020-02-10:
Thank you very much! During this week i'll the new firmware.

mrv96, 2020-02-10:
Hu @vanvught , i tried your new sACN firmware and it works well! I checked with both WireShark and sACNView.
I suggest to add an option that allows to change the node name.

I'll waiting for the ArtNet firmware and for the firmwares with more than one port.

Thank you very much!

vanvught, 2020-02-11:
Hi @mrv96 
> i tried your new sACN firmware and it works very well! I checked with both WireShark and sACNView.

That's great news - thanks!

> I suggest to add an option that allows to change the node name.

I will take this suggestion into consideration.

> 
> I'll waiting for the ArtNet firmware and for the firmwares with more than one port.
The single port Art-Net will be soon available for testing.

The multi-port versions for sACN/E1.31 and Art-Net will take some time longer. It will involve a more complex setup/configuration for testing and debugging.

-- Arjan

vanvught, 2020-02-12:
Hi @mrv96. herewith the Art-Net firmware. 
I have also updated the website -> [http://www.orangepi-dmx.org/raspberry-pi-art-net-dmx-out/configuration](http://www.orangepi-dmx.org/raspberry-pi-art-net-dmx-out/configuration)
-- Arjan

[opi_emac_artnet_dmx.zip](https://github.com/vanvught/rpidmx512/files/4194521/opi_emac_artnet_dmx.zip)

mrv96, 2020-02-12:
Hi @vanvught, i'm not very expert about sACN, while i wrote code for ArtNet devices, so i know a little bit better the standard.

The firmware works well, i tried with both Wireshark and Artnetominator, but the protocol says that an ArtNet device that send packets on an universe should send ArtPoll every 2.5/3 seconds, and in Wireshark i don't see any ArtPoll.

I know that this isn't a functional issue, but if you wanna reach a full compliancy you should send the ArtPoll packets.

As usually good work! Let me know what you think about my comments. Bye

vanvught, 2020-02-12:
Hi @mrv96 

>  but the protocol says that an ArtNet device that send packets on an universe should send ArtPoll every 2.5/3 seconds, and in Wireshark i don't see any ArtPoll.

A Node does not send the ArtPoll; it only *must* send the ArtPollReply. 

At startup there is the ArtPollReply as per specification -> " This packet is also broadcast to the Directed Broadcast address by all Art-Net devices on power up."

[image]

mrv96, 2020-02-13:
I agree with you, but if the node is DMX to ArtNet, so in input mode, it becomes the controller, and controller must send ArtPoll (as written in your picture)

hippyau, 2020-02-13:
Not really, it is still a node.  A controller is a console or software
package which configures nodes.  A node doesn't configure another node,
therefore it's not a controller.

What are you trying to achieve?

Cheers,
  Hip

vanvught, 2020-02-13:
> I agree with you, but if the node is DMX to ArtNet, so in input mode, it can become the controller, and controller must send ArtPoll (as written in your picture)

It is a Node with DMX512 input -> Art-Net

mrv96, 2020-02-13:
Ok, probably it was a my misunderstanding. Nevertheless good work! Everything works fine.

I really look forward to the firmware versions with more ports.

vanvught, 2020-02-15:
Published -> https://github.com/vanvught/h3dmx512-zip

closed 2020-02-15 by vanvught