vanvught/rpidmx512 #106: lib-l6470 multiple boards

boinappi, 2020-02-18:
Hi!
I'm working with your lib for quite some time now. Thanks for releasing! I updated to your latest version and ran into some issues with multiple Sparkfun L6470 v13 boards.
I'm on a Raspberry Pi3.

When using one board everything works fine. When connecting a second board i get a 0x19 reg value for board position 0, and 0 for the other one. Status is 0 for both .
The board farest from the RPI is set to SDO to: prev board, all others to SDO to: next board. All boards are connected to motor power. All boards are connected by ribbon cables.

Is this some chip select related issue? I was already digging into your code but could not find something i'd blame.

Thank your for your help and releasing this great lib!

vanvught, 2020-02-18:
>  I updated to your latest version and ran into some issues with multiple Sparkfun L6470 v13 boards.

Which previous version did work for you with multiple boards ?

When it did never work before; have you changed the `Data routing jumper ` for all boards except the last one?

Thanks, Arjan

boinappi, 2020-02-18:
I downloaded a dec 2018 version which worked with the described jumper changes.
These are the files i saved.
[lib-l6470.zip](https://github.com/vanvught/rpidmx512/files/4220764/lib-l6470.zip)

Unfortunately i don't have an exact version number/commit. 

Thank you so much!

vanvught, 2020-02-18:
Please note the change in commit https://github.com/vanvught/rpidmx512/commit/1caceeeee7a33cfa2056945bf60ee5682d068c63#diff-117ea56fca8325ef2c78d5829b1ef7a5 this might have impact on your code.

The multiple boards is working fine with the Orange Pi Zero (my main board). As the SPI code is shared, I am convinced that it is also working on a Raspberry Pi Model 3 board.

The earliest time I can look into it is next weekend.

vanvught, 2020-02-18:
See example here https://github.com/vanvught/rpidmx512/blob/master/lib-l6470dmx/src/sparkfundmx.cpp#L189

First all objects for the motors are (must be) created. Afterwards, the IsConnected method can be used.

boinappi, 2020-02-18:
Dear Arjan,
you are my human of the week!

I tried to create all board objects in a loop together with a configuration, which failed. I connected too early. Your last answer was the solution!
Thank you so much for your time!

closed 2020-02-18 by boinappi