vanvught/rpidmx512 #107: Time of Day on Opi LTC project drifts

drummerclint, 2020-02-20:
Hey Arjan,

I'm noticing that after a while (>12ish hours), the time of day starts to drift or becomes off.  It's currently displaying 21:50, meanwhile the time source is displaying 22:24.  This is using your ver 2.4, compiled jan 11.  If I reboot the unit, it jumps to the correct time.  I've got two opi's running like this.  One has the RTC connected, the other doesn't.  Both with the same firmware.  They both exhibit the same issue.

Thanks!

Clint

vanvught, 2020-02-20:
Hi Clint,

I am working on the issue. 

Thanks, Arjan

drummerclint, 2020-02-21:
Great!  Thanks Arjan!

Clint

vanvught, 2020-02-21:
Hi Clint, please will you test the attached firmware?
It is still V2.4 as the published version on GitHub is V2.3
When this firmware is working for you, then I will publish the V2.4

@hippyau Please will you give it a test as well? I've done some major updates to the WS28xx code.

Thanks, Arjan
[opi_emac_ltc_smpte.zip](https://github.com/vanvught/rpidmx512/files/4237331/opi_emac_ltc_smpte.zip)

drummerclint, 2020-02-21:
Sounds good!  I've updated both pi's.  I'll let you know how they fare out.  

Thanks again!
Clint

drummerclint, 2020-02-22:
Arjan, 

It worked!
You and @brunchboy James are my programming heros!

Clint

brunchboy, 2020-02-22:
And you give us excellent and worthy challenges that help us change the world!

vanvught, 2020-02-23:
Published -> https://github.com/vanvught/h3dmx512-zip/blob/master/opi_emac_ltc_smpte.zip?raw=true

closed 2020-02-23 by vanvught

drummerclint, 2020-02-24:
Sorry Arjan, I spoke too soon.  Got home last night at 4:43am to find one opi displaying 04:09 and the other displaying 03:35.
Currently it is 23:50.  One first opi is displaying 23:16 and the other is displaying 23:50.
Opi 1 is at 203425s of uptime, opi 2 is at 203498s of uptime.
Both opi's have your feb 21 compiled version of software.  opi one has an RTC.  Both have the same ntp server set.

vanvught, 2020-02-24:
I will try to reproduce the behavior seen. Running two Orange Pi Zero's; one with RTC and one without RTC.

Both are initially synchronized with an ArtTimeSync (this will set the RTC) and running the NTP Client.
The NTP Server is on the same machine as where the ArtTimeSync is generated from.

The ArtnetTimecodeViewer v 1.0.2 from s-jaekel.de is used for comparing both time of day.

cc: @brunchboy  @hippyau

vanvught, 2020-02-24:
Hi Clint,

When using the NTP protocol for synchronization then the Orange Pi Zero gets updates every 1024 seconds (https://github.com/vanvught/rpidmx512/blob/master/lib-network/src/ntpclient.cpp#L49). 

The Wireshark output shows that there is exactly 1023 seconds between every poll. And there is a successful client-server response for both Orange Pi Zero's.

 There is a minimum frame count difference between the Orange Pi Zero's, but this is due to the nature of the NTP protocol. When the ArtTimeSync (broadcast) is used, then there would be less or close to zero differences. 

It is working as expected. I am not able able to reproduce what you are observing. Please check with Wireshark if both Orange Pi Zero's are in the NTP synchronization process.

Thanks, Arjan
cc: @brunchboy @hippyau

[image]

drummerclint, 2020-02-24:
Hey Arjan, 
I'll have a look later today with wireshark.  opi one is showing correct time of 11:11 and opi two is showing 10:03 at the moment.  If I reboot them, they'll show the correct time.  Uptime is currently 38858s.  I believe it's after a few hours is when it starts to drift.  

Thanks!
Clint

vanvught, 2020-02-24:
Hi Clint,

> I believe it's after a few hours is when it starts to drift.

With both the NTP protocol or the ArtTimeSync there is no drift. Only with the NTP protocol there can be drift; less than a second.   

I have done the test for around 4 hours, but as both Orange Pi Zero's are getting synchronized for every 1024 seconds (approximately 17 mins), there is no reason for doing a longer test.

- Arjan
cc: @brunchboy @hippyau

vanvught, 2020-02-24:
Hi Clint,

> I believe it's after a few hours is when it starts to drift.

With the ArtTimeSync there is no drift. Only with the NTP protocol there can be a very small drift; less than a second between the two Orange Pi Zero's.   

I have done the test for around 4 hours, but as both Orange Pi Zero's are getting synchronized for every 1024 seconds (approximately 17 mins), there is no reason for doing a longer test.

Thanks, Arjan
cc: @brunchboy @hippyau

vanvught, 2020-02-24:
Hi Clint,

I have ran another test; an Orange Pi Zero with RTC (192.168.2.250) and the other without RTC.
Both Orange Pi Zero's are initially synchronized with the ArtTimeSync. And there is **no** NTP Client running. 

The Orange Pi Zero with RTC is nicely in sync with the host where the ArtTimeSync is originated from.

The other Orange Pi Zero is one second behind; this is just the ARM SoC internal clock. This is a one second drift in 6 hours uptime. Then 4 seconds drift per 24 hours looks reasonable for a not real-time clock.

Using the following options gives a perfect time of day:

1. Enable the NTP Client (there is an initial client request at startup).
2. Setup an Art-Net controller for sending ArtTimeSync with a specific interval. With the assumption that the Art-Net controller has a proper RTC and/or NTP Client running. 
3. Use a proper external RTC (can be initialized -one-time- by a. The NTP Client or b. the ArtTimeSync )

Any combination for above options is also possible. At any time. the NTP process or the ArtTimeSync  packet will update both internal time of day and a connected external RTC.

Thanks, Arjan
cc: @brunchboy @hippyau
 
[image]

hippyau, 2020-02-28:
Hi Arjan!

Sorry I disappeared, life got in the way, etc :)

I'm testing
[V2.4] Orange Pi Zero Compiled on Feb 23 2020 at 12:19:26
tonight your against 7 different clocks,

   - BigTC @ 30FPS (my WS2812 oPI LTC clock) sync to local network NTP
   server (shown in photo)
   - ESP8266 Clock with NTP internet pool sync every 24hr (so might see a
   difference) (shown in photo)
   - NTP Widget (Called "AtomicClock v1.7.1") on Android phone to local
   network (wifi) NTP server on manual refresh
   - Laptop on Ubuntu system clock on x86 against both local and internet
   pools on manual refresh
   - An unsynchronized analog wall clock :)

The android app makes an audible metronome (if enabled) and it's clicking
in perfect time, super helpful.

So far, all is good.  Some of the test are not going to be accurate, but a
significant difference should be detectable.

I'll check back in tomorrow :)

Hey does my WS2812 clock get me in as a github contributor? ;)

Cheers,
 Hip

hippyau, 2020-02-28:
Should I see a NTP client debug message every 17 minutes?   I'm still on
time, but haven't seen a sync yet after initial boot.

Note I can't see on wireshark at the moment, because NTP and oPI are
talking directly and not going through my laptop...

Cheers,
  Hippy

vanvught, 2020-02-28:
Hi Hip,

> Should I see a NTP client debug message every 17 minutes?

That debug message should not happen with the firmware which I release on GitHub. 

As I have, finally, received the new LTC balanced extension boards (need to build them, hopefully this weekend), I will check this debug message as well.  

Greets Arjan

hippyau, 2020-03-01:
I am seeing drift, but only when disconnected from the network/NTP server.

Im about 3 seconds behind the NTP server after 8 hours.... I don't have a
RTC on the oPI.   24hrs passed when connected to the network and all was
good (so every 17 minutes seemed to work).

I'm not bothered by this, but this should bring into question NTP servers
that loose internet connect, we will be reliant on their clock which might
eventually drift too.

Not an issue for ArtNET and other TC's, just system-time I suppose?

Might be a good idea to poll NTP server every 5 minutes?

Cheers,
  Hip

rewolff, 2020-03-01:
Polling more often doesn't help much. Once the server on your local network is cut off from the world, it will drift no matter how often you poll it. 

The full NTP package learns the local clock drift during operation and puts that in a file. That way the drift will be minimal should the connection to the internet be lost. 

After "the crystal just happens to be a few ppm too fast" the most likely suspect for the cause of the drift is temperature changes. So once ntp has synced for a couple of hours, it will know the structural drift of the local crystal is say -33.046ppm (current number from MY workstation), and it will keep adjusting the clock with that amount even when the connection  is lost. Only when the temperature changes will the clock show significant drift again. 

If you are really nervous about clock drift, buy yourself a GPS module with a PPS output and install that on your local timeserver. You'll need to arrange for GPS reception near that server.... 

Somehow my workstation wasn't running NTP when I started writing this. My clock had drifted 0.3 ms, but it is now back in sync at 71 us offset.

vanvught, 2020-03-01:
> Im about 3 seconds behind the NTP server after 8 hours

That's a little bit more that what I am observing -> 

> Then 4 seconds drift per 24 hours looks reasonable for a not real-time clock.

Rogier @rewolff  is explaining why the SoC clock can be drifting.

> Not an issue for ArtNET and other TC's, just system-time I suppose?

Yes, not an issue for the SMPTE LTC converter in general; it is about showing or using the system-time.

vanvught, 2020-03-01:
> If you are really nervous about clock drift, buy yourself a GPS module 

It is not on my list to support any GPS module. The current solution with the I2C RTC modules is perfect.

rewolff, 2020-03-01:
What I meant was: Hip was afraid his NTP server on the local network would lose connectivty to the NTP servers, so that sync-ing with that server would start to drift too. I was NOT insinuating you should start to support GPS modules for timekeeping purposes.

hippyau, 2020-03-01:
Yep I meant if you are still online, sorry I did not explain that better.
:-)

I live in Australia, so temperature is certinally a factor!

GPS is an interesting idea, but yep in the right conditions we are good :)

Cheers,
 Hip

hippyau, 2020-03-02:
Just FYI, I'm now 2.5 days running on SystemClock without network
connection and 8 seconds behind NTP time.

drummerclint, 2020-03-03:
> Just FYI, I'm now 2.5 days running on SystemClock without network
connection and 8 seconds behind NTP time.
@hippyau is there an rtc in your opi?

vanvught, 2020-03-03:
>@hippyau is there an rtc in your opi?

See earlier post https://github.com/vanvught/rpidmx512/issues/107#issuecomment-593084919

> Im about 3 seconds behind the NTP server after 8 hours.... I don't have a
RTC on the oPI.

vanvught, 2020-03-03:
> Just FYI, I'm now 2.5 days running on SystemClock without network
> connection and 8 seconds behind NTP time.
> @hippyau is there an rtc in your opi?

The 8 seconds is reasonable fine with no RTC, with no external synchronization.

When having a RTC, then the drift is with the specification of the RTC used. See the details here -> https://datasheets.maximintegrated.com/en/ds/DS3231.pdf

And some background here -> https://blog.heypete.com/2018/02/04/ds3231-drift-results-5-months/

hippyau, 2020-03-03:
Hi Arjan!

I have bought a DS3231, I'll connect it up tonight and start the tests
again.

No stress.

Cheers,
  Hip