vanvught/rpidmx512 #109: DMX Multiple Input -> sACN / Art-Net (follow-up for #103)

vanvught, 2020-03-11:
- [ ] lib-dmx create dmx_multi_input.c
- [ ] Update project opi_emac_e131_dmx_multi
- [ ] Update project opi_emac_artnet_dmx_multi
- [ ] Update website
- [ ] Update README.md's
- [ ] Publish on https://github.com/vanvught/h3dmx512-zip

cc @mrv96
#103

vanvught, 2020-03-13:
Hi @mrv96, herewith the sACN E1.31 multi port firmware. I hope you can give it a good test with both Zero (2x Input) and One (4x Input) board. Thanks, Arjan

[opi_emac_e131_dmx_multi.zip](https://github.com/vanvught/rpidmx512/files/4330811/opi_emac_e131_dmx_multi.zip)

mrv96, 2020-03-15:
Hi, as usual very good and fast work!

Now i have some difficult due to Corona-virus because i'm italian. I'm not ill but due to governament decision we cannot leave our homes so i cannot go to my lab.

I'll try to get the OPi boards asap. I hope to try the new firmwares early.

Bye

vanvught, 2020-03-15:
@mrv96 

> Now i have some difficult due to Corona-virus because i'm italian. I'm not ill but due to governament decision we cannot leave our homes so i cannot go to my lab.

Take care, stay safe!

vanvught, 2020-03-29:
Hi @mrv96, herewith the Art-Net multi port firmware. Greetings, Arjan

[opi_emac_artnet_dmx_multi.zip](https://github.com/vanvught/rpidmx512/files/4398587/opi_emac_artnet_dmx_multi.zip)

hippyau, 2020-03-29:
❤

mrv96, 2020-04-19:
Hi @vanvught , I have just finished to try your new firmwares.

Both work well for what concern the part of protocol, but I wasn't able to set net config in OPi One, neither with DHCP nor with static settings. I tried to do it with both manual txt editing and Remote Config Manager without success. In particular OPi One wasn't never seen from Remote Config Manager. The OPi Zero instead worked well. Have you any idea?

I have a suggestion for the future. I don't know sACN, but ArtNet supports 4 ports indipendently configured as input or output. It would be great to configure indipendently the single ports instead of use one modality for all.

As usual thank you for your wonderful work!

vanvught, 2020-04-19:
Hi @mrv96 

> In particular OPi One wasn't never seen from Remote Config Manager. 

https://github.com/vanvught/rpidmx512/blob/master/opi_emac_artnet_dmx_multi/firmware/main.cpp#L265

Which SPI flash chip have you added to the Orange Pi One? Please do a boot with the SDCard and connect a HDMI display for the debug information.

> It would be great to configure indipendently the single ports instead of use one modality for all.

That would not be a problem. However, it would need a major redesign for the configuration params where backwards compatibility is mostly important.

> As usual thank you for your wonderful work!

Thanks! 

Greetings, Arjan

vanvught, 2020-06-01:
Published

closed 2020-06-01 by vanvught