vanvught/rpidmx512 #112: The compilation error

sea18c, 2020-04-09:
The compilation error is caused by the file storegeneration.cpp in lib spiflashstore

vanvught, 2020-04-09:
Please provide the output of the compilation error. How did you compile?

vanvught, 2020-04-10:
There is no file named storegeneration.cpp

```
@nuc-i5:/development/workspace/lib-spiflashstore$ find . -name storegeneration.cpp
@nuc-i5:/development/workspace/lib-spiflashstore$ 
```

sea18c, 2020-04-10:
The file name I wrote wrong, should be src/storenextion.cpp.
Here are the error tips：     
In file included from src/storenextion.cpp:28:0:
./include/storenextion.h:29:27: fatal error: nextionparams.h: No such file or directory
 #include "nextionparams.h"
                           ^
compilation terminated.
../h3-firmware-template/lib/Rules.mk:101: recipe for target 'build_h3/src/storenextion.o' failed
make: *** [build_h3/src/storenextion.o] Error 1

commit e3441c4f (2020-04-10): Fixed #112

*. Deleted files for #66
*. Updated Copyright statement mailto:
closed 2020-04-10 by vanvught