vanvught/rpidmx512 #113: Multiple RDM devices were not found.

sea18c, 2020-04-10:
Multiple RDM devices have been searched and tested at the same time, and now I'm testing that it's easy to find if it's a single, but none of the rdM devices together are discovered.

Test environment: opi_emac_artnet_dmx for orangepi_zero

vanvught, 2020-04-10:
When opening an issue, then please provide detailed information. There is not much information to work with.

1. - Which DMX/RDM hardware are you using? 
2. - What about the signal termination used? 
3. - Are the fixtures RDM compliant? For example RDM Test results? 
4. - What are the UUID's for the RDM fixtures?
5. - Can you provided RDM traces?

vanvught, 2020-04-10:
I cannot replicate the issue. I am using the DMX/RDM boards from [http://bitwizard.nl/shop/dmx]( http://bitwizard.nl/shop/dmx)

```
arjanvanvught@MacBook-Air ~ % ola_rdm_discover -u 1
29aa:02001420
5000:081e0bd3
5000:1c34020e
arjanvanvught@MacBook-Air ~ % ola_rdm_discover -u 1 -f
29aa:02001420
5000:081e0bd3
5000:1c34020e
arjanvanvught@MacBook-Air ~ % ola_rdm_discover -u 1 -i
29aa:02001420
5000:081e0bd3
5000:1c34020e
arjanvanvught@MacBook-Air ~ % 

olad/plugin_api/Universe.cpp:523: Full RDM discovery triggered for universe 1
plugins/artnet/ArtNetNode.cpp:1778: Artnet RDM discovery complete
olad/plugin_api/Universe.cpp:523: Full RDM discovery triggered for universe 1
plugins/artnet/ArtNetNode.cpp:1778: Artnet RDM discovery complete
olad/plugin_api/Universe.cpp:525: Incremental RDM discovery triggered for universe 1
plugins/artnet/ArtNetNode.cpp:1778: Artnet RDM discovery complete
```

[image]

sea18c, 2020-04-12:
1.UID:111a:3c7a738f
2.I am using network cable is not wireless.
3.DMX-Workshop software search, RDM fixtures is a self-written program, may also be my RDM fixtures is not correct. 
4.Can you provide detailed instructions on the opi_dmx_usb_pro of OLA use?

vanvught, 2020-04-12:
> 1.UID:111a:3c7a738f

That is just one. You are working with multiple UUIDs

> 2.I am using network cable is not wireless.

Ok.

> 3.DMX-Workshop software search, RDM fixtures is a self-written program, may also be my RDM fixtures is not correct.

Please run the `rdm_test_server.py `
Your RDM Responder *must* have no Broken and no Failed items.
For example -> http://www.orangepi-dmx.org/raspberry-pi-rdm-responder/ola-rdm-responder-tests

> 4.Can you provide detailed instructions on the opi_dmx_usb_pro of OLA use?

http://www.orangepi-dmx.org/raspberry-pi-rdm-controller

Note : you also run the `rdm_test_server.py ` with the Art-Net firmware.

vanvught, 2020-04-12:
Open items:

1. Which DMX/RDM hardware are you using?
2. What about the signal termination used?

vanvught, 2020-04-23:
Closing as there is no activity, the issue is not related to the firmware.

closed 2020-04-23 by vanvught