vanvught/rpidmx512 #117: Network options

drummerclint, 2020-05-18:
Hey Arjan!

Hope you are well.  I was wondering if you make the following cases happen in the network configuration:

use_dhcp=0
ip_address=192.168.2.251

* If "use_dhcp=1" then
  * if dhcp server found, then use ip from dhcp (already the case)
  * if dhcp server not found, then
    * if ip_address is "ip_address=x.x.x.x" then use that IP
    * if ip_address is "#ip_address" or not found, then use APIPA IP

* If "use_dhcp=0" or "#use_dhcp" then
  * if ip_address is "ip_address=x.x.x.x", then use that IP
  * if ip_address is "#ip_address" or not found, then use default 2.x.x.x address that you have listed

I think it makes using the orangepi easier to use in many different network environments without having to always reconfigure networking

vanvught, 2020-05-18:
Hi Clint,

I will implement the IP Addressing Modes as described in ANSI E1.37-2 – 2015, section 3.5 IP Addressing Modes: 

> Three IPv4 address configuration modes are provided: static configuration, DHCP [DHCPv4] and Zeroconf [IPv4LL]. The IP Addresses used for an interface shall be determined by:
> • If a static address is configured, it shall be used.
> • If no static address is configured, and DHCP mode is enabled, the device shall attempt
> to acquire an address for the interface using DHCP.
> • If no static address is configured, and DHCP failed to provide an address (or was disabled / unsupported) a link local address shall be used if Zeroconf mode is enabled.
> • If none of the above provided an address, the interface shall remain unconfigured.

[IPv4LL] RFC 3927– Dynamic Configuration of IPv4 Link-Local Addresses. 2005. [[http://tools.ietf.org/html/rfc3927](http://tools.ietf.org/html/rfc3927)]

Then Zeroconf is always enabled. 

And when "use_dhcp=0" AND (ip_address is "#ip_address" or ip_address=0), then defaulting to the 2.x.y.z (based on MAC address as specified in Art-Net 4 "IP address configuration – Static Addressing").

drummerclint, 2020-05-19:
Hey Arjan,

That sounds great.  Thank you!

vanvught, 2020-05-21:
In order to assist in debugging Zeroconf -> https://github.com/ETCLabs/RDMnet/pull/35

vanvught, 2020-05-21:
Added new menu option 'Broadcast' to example Java UI. This will make it easier to debug this enhancement request.

[image]

[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/4662846/RemoteConfigUI.jar.zip)

vanvught, 2020-05-21:
Hi Clint, please find attached the build with full Zeroconf support. Please let me know how this new feature is working for you. Thanks, Arjan

[opi_emac_ltc_smpte.zip](https://github.com/vanvught/rpidmx512/files/4662917/opi_emac_ltc_smpte.zip)

vanvught, 2020-05-21:
Added flow chart -> http://www.orangepi-dmx.org/orange-pi-dmx512-rdm/network-configuration/ethernet

drummerclint, 2020-05-22:
Hi Arjan,

It's fantastic!  The broadcast feature is an indispensable touch!  Is it possible that when you click the Enable DHCP, it only affects the use_dhcp value?  Main reason is to not change the hostname parameter.  The flowchart is also a nice addon.

Thank you!

Clint

vanvught, 2020-05-22:
Hi Clint,

> Is it possible that when you click the Enable DHCP, it only affects the use_dhcp value? 

Such a function is possible with doing some MacOS/Linux shell scripting. First retrieving the 'network.txt' file, then doing a 'sed' for the 'use_dhcp'. And then sending back (unicast) the 'network.txt'.
I will add on my ToDo list providing some more scripting examples. I have already one here -> https://github.com/vanvught/h3dmx512-zip/blob/master/do-tftp.sh

With the sample UI, I just do broadcasting , there are no queries.
`broadcast("#network.txt\nuse_dhcp=1");`
 I needed this simple reset in my debug and testing environment. I will rename the button to 'Reset to defaults'.

When your lighting software/desk is RDMNet LLRP enabled (or using the ETC llrp_manager_example), then you can do a selected 'Factory reset'. This one just enables DHCP and does not touch any other network configuration. And it enables the remote configuration.

Greetings, Arjan

vanvught, 2020-05-22:
Update: there is now DHCP Release message when reboot is initiated with UDP or RDMNet LLRP message. 

[opi_emac_ltc_smpte.zip](https://github.com/vanvught/rpidmx512/files/4669736/opi_emac_ltc_smpte.zip)

vanvught, 2020-06-01:
Published

closed 2020-06-01 by vanvught