vanvught/rpidmx512 #118: Wireless Connection

hippyau, 2020-05-29:
Hey Arjan,

I've been thinking about wireless :)

I see the esp8266 code, for RPI, but trying to work it out for the Zero...  is this implemented yet?

I saw these pinouts.... What is the pinout for the OrangePiZero?
```
/*
 *      RP					ESP8266		Logic Analyzer
 *
 * 11	GPIO17	--      ->	GPIO5		orange
 * 13	GPIO27	<-      --	GPIO4		red
 *
 * 15	GPIO22	<- DATA ->	GPI012		purple
 * 16	GPIO23	<- DATA ->	GPI013		blue
 * 18	GPIO24	<- DATA ->	GPI014		green
 * 22	GPIO25	<- DATA ->	GPI015		yellow
 */
```
[image]

A simple UART would be great...

I had an idea of a UART interface between the OrangePi and anything with a UART interface, where serial string messages are sent prefixed with a 16-bit port number, and then either slotted into the network driver or just by a simple API where each firmware checking for UDP Ethernet packets could also quickly run a `check_uart_udp(udp_port_num);` and get a string... with a main `recieve_uart_udp();` function that acts as a buffer for messages going to each udp_port_num... so is easy to incorporate into many firmwares?

Not so useful for DMX, but great for just passing control and config strings.

I don't know, is there a UART available that you have not already used the pins for?

My goal, easy pass messages to the OrangePI over wireless.  Maybe is already working and I don't understand?

Cheers,
  Hipps

Happy Birthday by the way :)

Cheers, 
  Hippy

vanvught, 2020-05-29:
Hi Hippy,

The Orange Pi Zero is fully supported -> https://github.com/vanvught/rpidmx512/tree/master/lib-esp8266/src/h3

And all firmware is available :

- Art-Net 3 DMX Node / RDM Controller / Pixel Controller (WS28xx/SK6812/APA102/UCSx903) 
- sACN E1.31 DMX Bridge / Pixel Controller (WS28xx/SK6812/APA102/UCSx903) 
- OSC DMX Bridge / Pixel Controller (WS28xx/SK6812/APA102/UCSx903) 

However, I am not actively developing this code anymore. I do sometimes a test in order to make sure that new code won't break these old -legacy- implementations. 

All UARTs are in used and the all GPIO pins are in use -> http://www.orangepi-dmx.org/orange-pi-dmx512-rdm/gpio-header

It is easier having a Wifi-Ethernet router then developing WiFi code ;-)

The only option is adding an UART is using the I2C bus and the sc16is740. I have already code for the sc16is740 -> https://github.com/vanvught/opi-nextion/blob/master/lib-nextion/src/sc16is740.cpp for #66 

This option must have a seamless integration with remote configuration framework. But still ... it is much easier working with Wifi-Ethernet router ;-)

Greetings, Arjan

hippyau, 2020-05-29:
Thanks Arjan,

   Yep probably easier to use a Wifi Router :)

    I like the sc16is740, cool chip :)   I had though a simple esp8266 as a I2C slave,  but constant polling without a IRQ GPIO might be an issue?

Cheers, 
  Hip

closed 2020-05-29 by hippyau

vanvught, 2020-05-29:
>  I had though a simple esp8266 as a I2C slave, but constant polling without a IRQ GPIO might be an issue?

https://github.com/vanvught/opi-nextion/blob/master/lib-nextion/src/nextion.cpp#L103

The sc16is740 is interrupt based. Be aware of the cheap Chinese boards where the INT pin is not connected. You can clearly see this bug on the pictures.

Using the ESP 01 with its UART, we could simple send/receive the 'UDP messages' format used in the remote configuration framework. 

So..when we extend the remote configuration framework with receiving/sending messages using the sc16is740, then we can attach the ESP 01. However, I am not eager writing ESP 01 software ;)

vanvught, 2020-05-29:
Basically, when the ESP/UART uses the format as described here -> https://github.com/vanvught/opi-nextion then the OPi side is almost ready. And the Nextion/ESP solutions are interchangeable.

hippyau, 2020-07-26:
Hey Arjan :)

Okay, sorry to bring this up again,  regarding UART....  We can write to the debug UART, can we read from the debug UART?  I get plenty of messages out, would you have some pseudo-code of how to read the RX?

I'd like to implement a CLI.  This is something I could sink my teeth into, if I just knew how to read a char?

Cheers, 
  Hip