vanvught/rpidmx512 #119: LTC SMPTE: In source selection mode with buttons: using rotary for setting Start/Stop time

vanvught, 2020-06-12:
User comment:
> For LTC SMPTE. Using an encoder as described in " buttons" section, is it only for selecting the input source at startup ? Or you can set the start time when using internal generator?

Currently,  the buttons/rotary are only for the source select at startup.

cc: @hippyau @rbarreiros @brunchboy @drummerclint

vanvught, 2020-06-14:
Please find attached the first concept. The idea:

When we are in source select and Internal is selected then press Start for changing the Start time and/or Frame rate. With pressing start we toggle between editing the Start time and Frame rate.

You can press 'Select' for activating the selected/updated Start time and/or Frame rate.
With 'Resume' you are getting out of editing mode.

cc: @hippyau @rbarreiros @brunchboy @drummerclint

[opi_emac_ltc_smpte.zip](https://github.com/vanvught/rpidmx512/files/4776647/opi_emac_ltc_smpte.zip)

vanvught, 2020-06-19:
This is a full-functional firmware where Start, Stop and Frame-rate can be set. 

cc: @hippyau @rbarreiros @brunchboy @drummerclint

[opi_emac_ltc_smpte.zip](https://github.com/vanvught/rpidmx512/files/4805820/opi_emac_ltc_smpte.zip)

drummerclint, 2020-06-19:
Hey Arjan, the oled display has stopped working as of this build.
After I did the tftp upload and sent a reboot command, the display showed Rebooting.. and the unit rebooted, but the display didn't change.  The 7seg changed however.

vanvught, 2020-06-19:
Hi Clint, I did a quick test and I cannot replicate your issue. I will reopen this issue and will do some more investigation in the weekend.

vanvught, 2020-06-19:
Working for me on two different Orange Pi Zero's.

wyged, 2020-06-20:
Indeed. OLED Display is not working. Reverted to previous version and all ok.

vanvught, 2020-06-20:
@wyged @drummerclint This is mostly frustrating ;-) it is all working fine for me. Do you perhaps have an USB-UART cable so I can provide a debug version and hopefully it will give us more information for what is happening.

vanvught, 2020-06-20:
@wyged @drummerclint Two setups: With and without MCP23017. Both are working fine. I am so puzzled here. 

[image]

[image]
[image]

vanvught, 2020-06-20:
@wyged @drummerclint  Please find attached a debug firmware. I have added the 'i2cdetect' at startup; showing 2 MCP23017's and the SSD1306. 

[opi_emac_ltc_smpte.zip](https://github.com/vanvught/rpidmx512/files/4807695/opi_emac_ltc_smpte.zip)

[image]

cc: @hippyau @rbarreiros @brunchboy

vanvught, 2020-06-20:
@wyged @drummerclint I am going out for lunch now ;)

[opi_emac_ltc_smpte.zip](https://github.com/vanvught/rpidmx512/files/4807731/opi_emac_ltc_smpte.zip)

drummerclint, 2020-06-21:
works now :)

vanvught, 2020-06-21:
> works now :)

Thanks Clint. 

I will adjust my test scenario's; this OLED issue shouldn't have happened.

commit 6accacc4 (2020-06-19): Fixed #119
closed 2020-06-21 by vanvught