vanvught/rpidmx512 #120: LTC SMPTE: Generator: Buttons: New/additional functionalities

vanvught, 2020-06-30:
1. Would it be possible to have the "Start" function  and button acting like "Start-Resume" ( of course, resume after "stop" was pressed ). 
The "resume" button can be repurposed as "reset to initial start timecode" if you hold it pressed for let's say 2 second. For OSC and UDP messages the action will be instantly as it is a pain to code it.
2. Implementation of skip Back and Forward. After "Stop" is pressed or triggered, to have the option of skipping back or forward with increments of 1, 5, 10 seconds ( via OSC or UDP messages ) or increments of 5 seconds if the "left" or "right" buttons are pressed, and of course the display will have to show the incremented and decremented timecode that will become the new "resume" start point. 

Implementation details will follow. 

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint

vanvught, 2020-07-01:
Introducing 2 new properties.

1. alt_function= when set to 1, these new buttons functionality is in use.
2. skip_seconds= (default is 5) sets the increments when the "left" or "right" buttons are pressed.

[image]

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint

vanvught, 2020-07-02:
1. Instead of pressing 2 seconds the idea is to implement the same functionality in alignment with the reboot option.

When the `"reset to initial start timecode"` button is pressed, then `">RESET TC?<"` is shown on the OLED display.
Then with the `"select"` button, this selection can be confirmed.
When pressing the ` "reset to initial start timecode" ` button again, the OLED display will show `"Internal"` again. Canceling the reset option.

- [ ]  7-segment 

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint

vanvught, 2020-07-02:
Herewith a first build.

- New OSC messages: http://www.orangepi-dmx.org/orange-pi-smpte-timecode-ltc-reader-converter/internal-generator/run-time-osc-messages
- New UDP messages: http://www.orangepi-dmx.org/orange-pi-smpte-timecode-ltc-reader-converter/internal-generator/udp-messages 

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/4865970/orangepi_zero.uImage.gz)

Still have to work on 7-segment display messages.

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint

vanvught, 2020-07-03:
Added new property: `skip_free`

With property skip_free=1 you can now go below start and above stop time.

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/4870494/orangepi_zero.uImage.gz)
[image]

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint

vanvught, 2020-07-05:
This should be the last attachment here before release on https://github.com/vanvught/h3dmx512-zip

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/4875352/orangepi_zero.uImage.gz)

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint

vanvught, 2020-07-09:
Added Forward/Backward button to sample UI.

[image]

[RemoteConfigUI.zip](https://github.com/vanvught/rpidmx512/files/4897948/RemoteConfigUI.zip)

There will be a next append with updated firmware for the implementation:
> Would` it be possible, after the timecode has stopped automatically, to use the skip free without pressing stop again?

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint

vanvught, 2020-07-10:
Implemented the request from previous append.

Background: When the Start Time or Stop Time limit is reached, then the generator is still in `started` state. With `skip_free=1` you can use the new forward/backward functionality once until:

1. There is a `stopped` state.
2. The  Start Time or Stop Time limit is reached again.

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/4904731/orangepi_zero.uImage.gz)

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint

vanvught, 2020-07-31:
@hippyau @rbarreiros @brunchboy @wyged @drummerclint

Please give this attachment a test. I am aiming to publish the new firmware this Sunday. Thanks, Arjan

[opi_emac_ltc_smpte.zip](https://github.com/vanvught/rpidmx512/files/5007749/opi_emac_ltc_smpte.zip)

brunchboy, 2020-07-31:
It’s cool to see all the progress you’re making Arjan, but I don’t have hardware that can run your firmware myself, so I can’t help test it I’m afraid.

vanvught, 2020-08-06:
Published

https://github.com/vanvught/h3dmx512-zip/commit/293961290ddcc84fc874847e843b527025479e34

closed 2020-08-06 by vanvught