vanvught/rpidmx512 #121: LTC SMPTE - JamSync

mungewell, 2020-07-15:


mungewell, 2020-07-15:
Hi, just found the project so trying to get my head around how it might work. I am looking to build a cheap solution to providing a stable/reliable time source which can be 'jam sync'ed' to a master clock and then provide a free running clock as LTC.

ie
`External clock --LTC---> input -> internal ntp -> output ---LTC---> camera audio input`

I'd imagine that the clock stablity may need to be learned, such as a long term connection to master clock which is then 'learned' to the internal NTP server. Possibly temperature would need to be monitored to calibrate internal crystal.

Is this possible? Or at least somewhat implemented?

vanvught, 2020-07-15:
The firmware delivers a stable and reliable LTC SMPTE Reader / Converter / Generator.
There is any-to-any converter implementation -> http://www.orangepi-dmx.org/orange-pi-smpte-timecode-ltc-reader-converter
[image]

When `source=systime` then the NTP Client can be configured as part of the network settings.
For all other `source= `the NTP Server can be enabled.

closed 2020-07-15 by vanvught

mungewell, 2020-07-15:
Excellent, thanks for the reply. Looks like getting some hardware built would be worth the effort.

Just to confirm, is it correct that the input/Output mode can be changed on the fly?

So...
1. Connect external hardware and then set Input to 'LTC' and output to 'NTP'.
2. Wait for NTP (or Chrony) server to sync and adjust it's rate to match external hardware
3. Set input to 'Generator' and output to 'LTC'

Also I think that you may have a problem between 'Drop' and 'NoDrop' modes, I'll write up another bug.

vanvught, 2020-07-15:
> Just to confirm, is it correct that the input/Output mode can be changed on the fly?

No, you do the configuration at boot process. 

> 1.  Connect external hardware and then set Input to 'LTC' and output to 'NTP'.

Here you enable the NTP Server -> http://www.orangepi-dmx.org/orange-pi-smpte-timecode-ltc-reader-converter/configuration

> 2. Wait for NTP (or Chrony) server to sync and adjust it's rate to match external hardware

> 3. Set input to 'Generator' and output to 'LTC'

Here you enable the NTP Server -> http://www.orangepi-dmx.org/orange-pi-smpte-timecode-ltc-reader-converter/configuration

> Also I think that you may have a problem between 'Drop' and 'NoDrop' modes, I'll write up another bug.

See https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/h3/ltcgenerator.cpp#L651
It is not implemented.