vanvught/rpidmx512 #122: LTC SMPTE - Drop/NoDrop time <-> NTP server

mungewell, 2020-07-15:
I think that you may have a problem in the way that time is passed between LTC and NTP domains, code is:
https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/ntpserver.cpp#L115

In 'Drop Frame' the frame numbers are non-contiguous; every minute (except multiples of 10min) 2 frame numbers (00 and 01) are missing from LTC stream.
https://en.wikipedia.org/wiki/SMPTE_timecode#Drop-frame_timecode

It looks like the code (above) does not compensate for this, so if consecutive LTC timestamps were passed there would a sudden jump in time presented to NTP. Additionally the precise time from LTC frame would(/might) creep forward wrt to true time until the correction was made.

Here are some samples of Drop/NoDrop from a SyncIO unit:
[drop_no_drop.zip](https://github.com/vanvught/rpidmx512/files/4927021/drop_no_drop.zip)

Additionally (minor point), the delimiter between "sec" and "frame" is often set to ":" for non-drop and ";" for drop. I see that you are using ".". Maybe "." and "," would be suitable alternates.

vanvught, 2020-07-15:
Due to the nature of the NTP protocol, this https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/ntpserver.cpp#L115 is best the approximation we can do. There is no need to implement otherwise.

closed 2020-07-15 by vanvught

mungewell, 2020-07-15:
"_There is no need to implement otherwise._"... LOL ;-)

Since 'xx:x0:00;00' is True Time, you could extend TLtcTimeCode to include a 'nDropComp' (ie the number of frames you are ahead of true time) which would be easy to compute when decoding LTC frame.

Then when computing time to send to NTP you could subtract this number of frames, ie
```
} else if ((pLtcTimeCode->nType == ltc::type::DF) || (pLtcTimeCode->nType == ltc::type::SMPTE)) {
		m_nFraction = static_cast<uint32_t>((143165576.5 * (pLtcTimeCode->nFrames - pLtcTimeCode->nDropComp)));
```

Your magic number might have to change according to Drop/NoDrop and framerate.

vanvught, 2020-07-15:
The NTP Client/Server model is not a real-time synchronization. The polling is typically every 1024 seconds. Hence no need to implement any additional coding. When there is a real issue proven, then I will look into it. Thanks.